lbbrhzn/ocpp #461: plug and charge support

lbbrhzn, 2022-03-25:
**Is your feature request related to a problem? Please describe.**
Some chargers / cars support plug and charge, would be good to support it in this integration.

**Describe the solution you'd like**
Enabling support for ISO 15118 Plug and Charge can be done with the configuration key ISO15118PnCEnabled. This
configuration key can also be used to determine whether the Charge Point supports this functionality. If the configuration key is
present, the Central System can consider this functionality supported by the Charge Point. See https://www.openchargealliance.org/uploads/files/ocpp_1_6_ISO_15118_v10.pdf

**Describe alternatives you've considered**
There is an alternative standard called AutoCharge. This should be supported out o fthe box, since it uses the MAC address of the car as id_tag. See https://uploads-ssl.webflow.com/607417b42ba2bfea543956dd/60c33c6cc823f9bb6bd4f275_Whitepaper-Autocharge-vs-ISO15118-Plug-and-Charge.pdf for more info.

**Additional context**
None

lbbrhzn, 2022-03-25:
Is there anyone that has a charger+car that supports plug and charge?

drc38, 2022-03-28:
I see they've included this in 1.6 from the 2.0 standard. Is it worth working on 2.0 support? Are any chargers saying they're 2.0 compatible?

closed 2022-06-04 by github-actions[bot]