lbbrhzn/ocpp #466: Charger reconnects often after Firmware update

N3rdix, 2022-03-29:
My Vestel EVC04 charger updated its firmware today on its own and since that point in time the integration constantly generates persistent notification in home assistant while it is idle.

[image]

What I can see in the logs is the following below (at the same time the persistent notification pops up).

I did not start a charging cycle yet, not sure if this changes anything.

```text

2022-03-29 21:15:01 INFO (MainThread) [ocpp] Garage: receive message [2,"7106e13d-9867-417d-a711-b0023218ad22","MeterValues",{
    "connectorId": 1,
    "meterValue": [
        {
            "timestamp": "2022-03-29T19:15:00Z",
            "sampledValue": []
        }
    ]
}]
2022-03-29 21:15:01 ERROR (MainThread) [custom_components.ocpp] Unexpected exception in connection to 'Garage': 'Payload '{'connectorId': 1, 'meterValue': [{'timestamp': '2022-03-29T19:15:00Z', 'sampledValue': []}]} for action 'MeterValues' is not valid: [] is too short

Failed validating 'minItems' in schema['properties']['meterValue']['items']['properties']['sampledValue']:
    {'items': {'additionalProperties': False,
               'properties': {'context': {'additionalProperties': False,
                                          'enum': ['Interruption.Begin',
                                                   'Interruption.End',
                                                   'Sample.Clock',
                                                   'Sample.Periodic',
                                                   'Transaction.Begin',
                                                   'Transaction.End',
                                                   'Trigger',
                                                   'Other'],
                                          'type': 'string'},
                              'format': {'additionalProperties': False,
                                         'enum': ['Raw', 'SignedData'],
                                         'type': 'string'},
                              'location': {'additionalProperties': False,
                                           'enum': ['Cable',
                                                    'EV',
                                                    'Inlet',
                                                    'Outlet',
                                                    'Body'],
                                           'type': 'string'},
                              'measurand': {'additionalProperties': False,
                                            'enum': ['Energy.Active.Export.Register',
                                                     'Energy.Active.Import.Register',
                                                     'Energy.Reactive.Export.Register',
                                                     'Energy.Reactive.Import.Register',
                                                     'Energy.Active.Export.Interval',
                                                     'Energy.Active.Import.Interval',
                                                     'Energy.Reactive.Export.Interval',
                                                     'Energy.Reactive.Import.Interval',
                                                     'Power.Active.Export',
                                                     'Power.Active.Import',
                                                     'Power.Offered',
                                                     'Power.Reactive.Export',
                                                     'Power.Reactive.Import',
                                                     'Power.Factor',
                                                     'Current.Import',
                                                     'Current.Export',
                                                     'Current.Offered',
                                                     'Voltage',
                                                     'Frequency',
                                                     'Temperature',
                                                     'SoC',
                                                     'RPM'],
                                            'type': 'string'},
                              'phase': {'additionalProperties': False,
                                        'enum': ['L1',
                                                 'L2',
[... 247 lines cut ...]
    "configurationKey": [
        {
            "key": "SupportedFeatureProfiles",
            "readonly": true,
            "value": "Core,FirmwareManagement,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger"
        }
    ]
    }]
2022-03-29 21:15:39 INFO (MainThread) [ocpp] Garage: send [2,"6276a2c1-eaf2-45b8-9809-fb689d343b43","GetConfiguration",{"key":["NumberOfConnectors"]}]
2022-03-29 21:15:39 INFO (MainThread) [ocpp] Garage: receive message [3,"6276a2c1-eaf2-45b8-9809-fb689d343b43",{
    "configurationKey": [
        {
            "key": "NumberOfConnectors",
            "readonly": true,
            "value": "1"
        }
    ]
    }]
2022-03-29 21:15:39 INFO (MainThread) [ocpp] Garage: send [2,"7e3684b0-d303-4489-9658-3f845c82085e","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2022-03-29 21:15:39 INFO (MainThread) [ocpp] Garage: receive message [3,"7e3684b0-d303-4489-9658-3f845c82085e",{"status":"Accepted"}]
2022-03-29 21:15:39 INFO (MainThread) [ocpp] Garage: send [2,"ec49c10f-a36b-4be1-a6d6-77a94ff85054","GetConfiguration",{"key":["HeartbeatInterval"]}]
2022-03-29 21:15:39 INFO (MainThread) [ocpp] Garage: receive message [2,"e2a4dc8a-77e2-4156-aba3-98ddca2b0060","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2022-03-29T19:15:38Z"}]
2022-03-29 21:15:39 INFO (MainThread) [ocpp] Garage: send [3,"e2a4dc8a-77e2-4156-aba3-98ddca2b0060",{}]

```

N3rdix, 2022-03-29:
I was able to eliminate at least one persistent notification for the rejected MeterValuesSampledData by adjusting the values in the charger settings.
But the integration still reports a "charger <XXX> rebooted" every 10 minutes with the log messages from my previous post

lbbrhzn, 2022-03-29:
The MeterValues.req that is sent by the new firmare does not comply to the OCPP standard. There should be at least one SampledValue in a MeterValue record, but your chargers sends a message with no MeterValue. You can try to enable the skip ocpp validation in the configuration flow.

The charger rebooted message occurs when an untriggered boot notification is  received, presumably because the charger rebooted due to an internal error.

Looks like the new firmware is not really an improvement. I would recommend to switch back to the previous version.

drc38, 2022-03-29:
You could also try setting Idle Interval to 0 so that the charger only reports meterValues when it is charging.

N3rdix, 2022-03-30:
> The MeterValues.req that is sent by the new firmare does not comply to the OCPP standard. There should be at least one SampledValue in a MeterValue record, but your chargers sends a message with no [...]

Thanks for clarifying that, I will try to contact Vestel and tell them the story, maybe they are motivated enough to fix the bug in a future firmware update.

Unfortunately I cannot roll back to an older firmware version as this one got pushed OTA :-(

I will give @drc38 's proposal a chance and see what happens.

N3rdix, 2022-03-30:
@drc38 your proposal works as expected, no more errors in idle mode. 

I've reported this issue to the manufacturer Vestel and they will hopefully fix it with an upcoming update.

closed 2022-03-30 by N3rdix