lbbrhzn/ocpp #468: Error setting up entry central for ocpp

nilsmau, 2022-03-30:
## Version of the custom_component
v.0.4.28

## Configuration
Error occurs at the config flow, no additional YAML configuration.
It does not matter if I change any of the default values or leave them as they are during integration setup. 
Modification of manifest.json to `websockets==9.1 `did not work (found here: https://github.com/lbbrhzn/ocpp/issues/172#issuecomment-939136784)

## Describe the bug
Error setting up entry central for ocpp.
This error occurs when trying to configure the integration from the config flow, with the default configuration.

## Debug log

```text

This error originated from a custom integration.

Logger: homeassistant.config_entries
Source: custom_components/ocpp/api.py:171 
Integration: Open Charge Point Protocol (OCPP) (documentation, issues) 
First occurred: 19:33:47 (2 occurrences) 
Last logged: 19:37:06

Error setting up entry central for ocpp
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/ocpp/__init__.py", line 71, in async_setup_entry
    central_sys = await CentralSystem.create(hass, entry)
  File "/config/custom_components/ocpp/api.py", line 171, in create
    server = await websockets.server.serve(
  File "/usr/local/lib/python3.9/site-packages/websockets/legacy/server.py", line 1071, in __await_impl__
    server = await self._create_server()
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 1494, in create_server
    raise OSError(err.errno, 'error while attempting '
OSError: [Errno 98] error while attempting to bind on address ('0.0.0.0', 9000): address in use

```
## System Config

version | core-2022.3.8
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.92-v8
arch | aarch64
timezone | Europe/Berlin

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4880
Installed Version | 1.22.0
Stage | running
Available Repositories | 1007
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.5
-- | --
update_channel | stable
supervisor_version | supervisor-2022.03.5
docker_version | 20.10.9
disk_total | 457.7 GB
disk_used | 59.6 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), Check Home Assistant configuration (3.10.0), ESPHome (2022.3.2), File editor (5.3.3), Mosquitto broker (6.0.1), Terminal & SSH (9.3.0), pigpio (1.5.0), Z-Wave JS (0.1.55), deCONZ (6.12.0), Samba Backup (5.0.0), rtl_433 MQTT Auto Discovery (0.3.1), Portainer (2.11.1-3), rtl_433 (0.2.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 4
views | 5
mode | storage

</details>

lbbrhzn, 2022-03-30:
Looks like some other service is installed op that uses port 9000. Can you try a different port number?

We should improve feedback to the user if this occurs...

nilsmau, 2022-03-31:
Hello @lbbrhzn, that seem to have worked, I got the integration set-up properly by using another port (9123).
Thanks for the advice!

closed 2022-03-31 by nilsmau