lbbrhzn/ocpp #470: argument of type int not iterable

pjcarly, 2022-03-31:
I'm trying to setup a Smappee EV Wall, the setup is a little unconventional, but it used to work in the past. I used it a while a month or 2 back, but then had to reset my HA, and only recently got back to enabling OCPP again.

The  unconventional thing about this setup is the way the Smappee connects. The EV Charger is connected to the Smappee cloud through an monitoring tool that is connected to the internet. In the Smappee Cloud dashboard I can set-up a "Public charging CSMS", and fill in the endpoint of that "public" CSMS. This is basically a way of connecting your charger via the smappee cloud over OCPP to a public platform.
The way I managed to setup this, is by connecting my local running raspberry pi over an SSH tunnel to a remote server. I then expose my OCPP port over that SSH tunnel, and set a Nginx reverse proxy in front with a HTTPS certificate. This used to work.

Enabling the setup now, I can see my charger connecting for a few seconds, and then losing the connection.
I enabled debugging, and managed to capture the following logs (with errors).

```
2022-03-31 23:02:19 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to 'SMP_6200000632'
2022-03-31 23:02:19 INFO (MainThread) [custom_components.ocpp] Charger SMP_6200000632 disconnected from 0.0.0.0:9000.
2022-03-31 23:02:19 INFO (MainThread) [websockets.server] connection closed
2022-03-31 23:02:24 INFO (MainThread) [websockets.server] connection open
2022-03-31 23:02:24 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2022-03-31 23:02:24 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/ocpp/SMP_6200000632
2022-03-31 23:02:24 INFO (MainThread) [custom_components.ocpp] Charger SMP_6200000632 reconnected to 0.0.0.0:9000.
2022-03-31 23:02:24 DEBUG (MainThread) [custom_components.ocpp] Reconnect websocket to SMP_6200000632
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: send [2,"f179f1d2-d93d-42ba-b3bd-970d98c18874","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: receive message [3,"f179f1d2-d93d-42ba-b3bd-970d98c18874",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"CSLtype"}],"unknownKey":[]}]
2022-03-31 23:02:26 INFO (MainThread) [custom_components.ocpp] Supported feature profiles: CSLtype
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: send [2,"0b1139c0-2f44-4853-b6a1-520a6a2682f0","GetConfiguration",{"key":["NumberOfConnectors"]}]
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: receive message [3,"0b1139c0-2f44-4853-b6a1-520a6a2682f0",{"configurationKey":[{"key":"NumberOfConnectors","readonly":false,"value":"1"}],"unknownKey":[]}]
2022-03-31 23:02:26 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for NumberOfConnectors: 1
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: send [2,"04482569-70dc-4456-903d-2b65988b0f5d","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: receive message [3,"04482569-70dc-4456-903d-2b65988b0f5d",{"status":"Accepted"}]
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: send [2,"049dba77-f9f4-43f1-8d2b-06837d707fa9","GetConfiguration",{"key":["HeartbeatInterval"]}]
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: receive message [3,"049dba77-f9f4-43f1-8d2b-06837d707fa9",{"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"3600"}],"unknownKey":[]}]
2022-03-31 23:02:26 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for HeartbeatInterval: 3600
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: send [2,"efab0ffd-e16d-464a-8c0b-e2145be58517","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: receive message [3,"efab0ffd-e16d-464a-8c0b-e2145be58517",{"configurationKey":[{"key":"WebSocketPingInterval","readonly":true,"value":"60"}],"unknownKey":[]}]
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: send [2,"4460344d-f162-43e3-8578-f30060c38651","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: receive message [3,"4460344d-f162-43e3-8578-f30060c38651",{"configurationKey":[{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energy.Active.Import.Interval,Energy.Reactive.Import.Interval,Power.Active.Import,Power.Reactive.Import,Power.Offered,Power.Factor,Current.Import,Current.Offered,Voltage,Frequency,RPM,SoC,Temperature,Current.Export,Energy.Active.Export.Register,Energy.Reactive.Export.Register,Energy.Active.Export.Interval,Energy.Reactive.Export.Interval,Power.Active.Export,Power.Reactive.Export"}],"unknownKey":[]}]
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: send [2,"be001b19-4c8c-4444-b6de-f00de673cd5d","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: receive message [3,"be001b19-4c8c-4444-b6de-f00de673cd5d",{"configurationKey":[{"key":"MeterValueSampleInterval","readonly":false,"value":"60"}],"unknownKey":[]}]
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: send [2,"fe3837ae-ab83-4a72-aff9-0a6c1ceecc2d","GetConfiguration",{"key":["ClockAlignedDataInterval"]}]
2022-03-31 23:02:26 INFO (MainThread) [ocpp] SMP_6200000632: receive message [3,"fe3837ae-ab83-4a72-aff9-0a6c1ceecc2d",{"configurationKey":[{"key":"ClockAlignedDataInterval","readonly":false,"value":"900"}],"unknownKey":[]}]
2022-03-31 23:02:26 ERROR (MainThread) [custom_components.ocpp] Unexpected exception in connection to 'SMP_6200000632': 'argument of type 'int' is not iterable'
Traceback (most recent call last):
File "/config/custom_components/ocpp/api.py", line 898, in run
await asyncio.gather(*self.tasks)
File "/config/custom_components/ocpp/api.py", line 448, in post_connect
if prof.SMART in self._attr_supported_features:
TypeError: argument of type 'int' is not iterable
2022-03-31 23:02:26 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to 'SMP_6200000632'
2022-03-31 23:02:26 INFO (MainThread) [custom_components.ocpp] Charger SMP_6200000632 disconnected from 0.0.0.0:9000.
2022-03-31 23:02:26 INFO (MainThread) [websockets.server] connection closed
2022-03-31 23:02:32 INFO (MainThread) [websockets.server] connection open
2022-03-31 23:02:32 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2022-03-31 23:02:32 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/ocpp/SMP_6200000632
2022-03-31 23:02:32 INFO (MainThread) [custom_components.ocpp] Charger SMP_6200000632 reconnected to 0.0.0.0:9000.
2022-03-31 23:02:32 DEBUG (MainThread) [custom_components.ocpp] Reconnect websocket to SMP_6200000632
```

drc38, 2022-03-31:
> We could make our integration a bit more robust against 'unknown' FeatureProfiles. @drc38, do you agree?

What are you thinking,  default to `Core` or simply improve error handling?

lbbrhzn, 2022-03-31:
default to Core would be a good solution I think. It should always be  supported according to the standard. Unknown FeatureProfiles should be ignored.

lbbrhzn, 2022-04-07:
Please check v0.4.29

pjcarly, 2022-04-08:
Upgraded to v0.4.29. Error still seems the same (altho the line number in the trace changed)

```
Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:448
Integration: Open Charge Point Protocol (OCPP) ([documentation](https://github.com/lbbrhzn/ocpp/blob/main/README.md), [issues](https://github.com/lbbrhzn/ocpp/issues))
First occurred: 01:28:45 (23 occurrences)
Last logged: 01:31:34

Unexpected exception in connection to 'SMP_6200000632': 'argument of type 'int' is not iterable'
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 909, in run
    await asyncio.gather(*self.tasks)
  File "/config/custom_components/ocpp/api.py", line 448, in post_connect
    if prof.SMART in self._attr_supported_features:
TypeError: argument of type 'int' is not iterable
```

drc38, 2022-04-08:
v0.4.29 won't fix your issue as the profile is not supported. I assume it logged the following before throwing an error? `Unknown feature profile detected ignoring: CSLtype`

drc38, 2022-04-08:
You can replace line 481 with `feature_list=["Core","SmartCharging"]` etc based on what profiles your charger supports.

pjcarly, 2022-04-09:
Correct, my log was spammed with the message.
`Unknown feature profile detected ignoring: CSLtype`
Because the integration kept crashing and rebooting.

When you say replace the line, am I supposed to edit the integration myself to fix this? Or should I try to log an issue with the company of my charger?

drc38, 2022-04-09:
You can do both 😉

pjcarly, 2022-06-13:
Issue is no longer relevant. Smappee fixed the problem on their side. Connection seems to be relatively stable now.

closed 2022-06-13 by pjcarly