lbbrhzn/ocpp #479: [observation] 4.29 breaks ChargeControl switch on ABB terra

a42wg, 2022-04-05:
**Describe the bug**
After updating from 4.27 -> 4.29, the charge control switch started to misbehave.
It was possible to turn on the charger, but not to turn it off again

I do not have any logs and have reverted to 4.27 - Now everything works fine

lbbrhzn, 2022-04-07:
Would be good if you can provide some logs.
@drc38 you changed something to charge control, can you look into this?

drc38, 2022-04-07:
I've had a look and can't see anything in the changes that would cause the switch not to turn off.

a42wg, 2022-04-08:
I am a little short of time right now, but I will try to produce some logs asap

a42wg, 2022-04-09:
Hi, I have tried updating to 4.29 once again, with debug log enabled. This time the integration worked.
Then I removed the debug-log entry and restarted HA. Now the integration misbehave again.

I have attached logs with and without debug.

reg Anders
[Logs.zip](https://github.com/lbbrhzn/ocpp/files/8456934/Logs.zip)

a42wg, 2022-05-08:
Tried v 4.31 yesterday, still same issue.

I am a bit unsure how this works in HA, but is it possible to make changes to the source in custom_components/ocpp directly and make my HA run on the changed src?
Looking at 4.27 vs 4.28 most changes are in switch.py and I could try to debug it my self if possible

drc38, 2022-05-08:
Yes you can. Clone the repo, make your changes and use HACS to add a custom repo with your repo url.

lbbrhzn, 2022-05-08:
You can also add the visual studio code add-on to home-assistant and edit the source code right in your browser.

a42wg, 2022-05-09:
Thanks, I was planning to access the source directly via Samba, it sounds as that should be OK then. Do I need a full restart to enable my changes or can I do a partial reload?

closed 2022-07-16 by github-actions[bot]