lbbrhzn/ocpp #486: ChangeConfigurationPayload

Jonamaita, 2022-04-11:
**Describe the bug**
I am tryigin make a CharPoint client to test my Central System. When the central System send a `ChangeConfiguration`  to my CharPoint client, I route the message with `@on(Action.ChangeConfiguration)`  and when i try send the status to Central System as fallows:

```python
@on(Action.ChangeConfiguration)
  async def on_change_configuration(self, key, value):
      """
      on_change_configuration
      """

      print(f"key {key}, value {value}")

      rq = call_result.ChangeConfigurationPayload(
          ConfigurationStatus.reboot_required.value
      )
          
      await self.call(rq)
```

I get the next Exception:

```console
Traceback (most recent call last):
  File "~/envCpClient/lib/python3.8/site-packages/ocpp/messages.py", line 192, in validate_payload
    validator.validate(message.payload)
  File "~/envCpClient/lib/python3.8/site-packages/jsonschema/validators.py", line 353, in validate
    raise error
jsonschema.exceptions.ValidationError: Additional properties are not allowed ('status' was unexpected)

Failed validating 'additionalProperties' in schema:
    {'$schema': 'http://json-schema.org/draft-04/schema#',
     'additionalProperties': False,
     'properties': {'key': {'maxLength': 50, 'type': 'string'},
                    'value': {'maxLength': 500, 'type': 'string'}},
     'required': ['key', 'value'],
     'title': 'ChangeConfigurationRequest',
     'type': 'object'}

On instance:
    {'status': 'RebootRequired'}
```

**Desktop (please complete the following information):**
 - OS: ubuntu~20.04.1

drc38, 2022-04-11:
Take a look at `test_charge_point.py` for an example of handling the configuration payload on the charger side.

Jonamaita, 2022-04-11:
> Take a look at `test_charge_point.py` for an example of handling the configuration payload on the charger side.
> 
> Also the upstream repo https://github.com/mobilityhouse/ocpp as that is the code generating your error.

Thank you !! I solved it using just `return call_result.ChangeConfigurationPayload(ConfigurationStatus.accepted)`. I thought I had to wait for a response from the Central System, that was my mistake.

closed 2022-04-14 by drc38