lbbrhzn/ocpp #492: Overload, house main breaker opened after losing connexion with SmartWallbox

ClemKC, 2022-04-17:
Hello,
I was charging my car the other night when the house suddenly tripped. I thought I had consumed too much and the car was no longer charging in order to release energy at home, but in fact there was a bug. NodeRed which manages the connection with the wallbox (for the current) sent a request for 0A to the wallbox (load shedding) but the wallbox no longer communicated with HomeAssistant so HA sent but the wallbox did not know it.
There is a function called fallback (degraded mode) in the OCPP standard but it has not been implemented? Basically I would like that when the connection is lost the wallbox no longer charges until the communication is back.
Could you enlighten me?

drc38, 2022-04-18:
Where have you come across "fallback" in the Ocpp standard? If there's an issue with communication any "failsafe" action would need to happen at the charger end.

ClemKC, 2022-04-18:
https://www.oasis-open.org/committees/download.php/58944/ocpp-1.6.pdf

In this document I think page 63. Otherwise it's done in LMS (Load Management System) by Schneider Electric so I think it's possible. Let me explain, when the wallbox is connected to the LMS a setprofile is sent and as soon as the communication is dead the wallbox stops. A colleague told me than it's done via degraded mode in the profile but I can't find it and tbh I couldn't translate the whole document in French ;)

For sure it's technicaly faisable but I don't know where it's written in the doc

lbbrhzn, 2022-04-21:
I don't see anywhere in the ocpp specification how to support this. 

You can setup a charging profile with purpose 'ChargePointMaxProfile', this is intended for load balancing among multiple connectors. In addition there are 'TxDefaultProfile' and 'TxProfile' purposes, which are activated when a new transaction is started. There is no mention of a profile that is automatically activated when the connection is lost.

ClemKC, 2022-07-01:
Stale issue message ? Really ?

closed 2022-09-07 by github-actions[bot]