lbbrhzn/ocpp #500: Feature profile detection not working

jonasbkarlsson, 2022-04-22:
Not sure exactly when this started to happen, but rather recently. Could either be when upgrading the integration or when upgrading my charge point firmware (CTEK Chargestorm Connected 2).

```
2022-04-22 14:38:01 INFO (MainThread) [ocpp] XXXXXXX: send [2,"f5d362d5-46a6-4385-aea3-bc1468bacba9","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2022-04-22 14:38:01 INFO (MainThread) [ocpp] XXXXXXX: receive message [3,"f5d362d5-46a6-4385-aea3-bc1468bacba9",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core, FirmwareManagement, LocalAuthListManagement, Reservation, SmartCharging, RemoteTrigger"}]}]
2022-04-22 14:38:01 WARNING (MainThread) [custom_components.ocpp] Unknown feature profile detected ignoring: FirmwareManagement
2022-04-22 14:38:01 WARNING (MainThread) [custom_components.ocpp] Unknown feature profile detected ignoring: LocalAuthListManagement
2022-04-22 14:38:01 WARNING (MainThread) [custom_components.ocpp] Unknown feature profile detected ignoring: Reservation
2022-04-22 14:38:01 WARNING (MainThread) [custom_components.ocpp] Unknown feature profile detected ignoring: SmartCharging
2022-04-22 14:38:01 WARNING (MainThread) [custom_components.ocpp] Unknown feature profile detected ignoring: RemoteTrigger
```

lbbrhzn, 2022-04-22:
Hmm, it maybe due to a code change I introduced a few weeks back. We split the feature list based on ',' but do not remove the space. Should be easy to fix.

lbbrhzn, 2022-04-22:
Please check v0.4.31

closed 2022-04-22 by lbbrhzn

jonasbkarlsson, 2022-04-22:
Great! Works well! Thanks for quick fix.

lbbrhzn, 2022-04-22:
Do you have two connectors / cables on your charger?

jonasbkarlsson, 2022-04-22:
No, one one.