lbbrhzn/ocpp #510: Support for Sync EV Charger

FingalP, 2022-05-10:
**Is your feature request related to a problem? Please describe.**
The ocpp integration doesn't work with my Sync EV charger. It claims to be OCPP 1.6 compliant but it isn't fully. I get a few different errors/issues:
- Sync EV charger doesn't recognise `SupportedFeatureProfiles` - it returns it as an unknown key, rather than `[""]` as the code expects for a charger which only supports Core.
- `ChangeAvailability` requests just time out. By default, this causes an error and the whole thing reconnects, which makes it unusable.
- Some metrics are set with the key as the connector_id, which is an integer. Homeassistant fails to log this because it expects keys to be strings.

**Describe the solution you'd like**
I've made a fork https://github.com/FingalP/ocpp which has workarounds. I'm not sure whether this is something which should be merged into the main repo because it's supporting a charger which isn't fully OCPP compliant, but it would be convenient for me if it was.

**Describe alternatives you've considered**
I have contacted Sync EV to see if they will update their code to make it fully OCPP compliant. No response so far.

**Additional context**
I work for [Carbon Coop](https://carbon.coop/), integrating with EV chargers for the [Powershaper Flex](https://flex.powershaper.io/) project and the [REScoop VPP](https://www.rescoopvpp.eu/) project.

drc38, 2022-05-13:
The SupportedFeatureProfiles and connector_id changes look fine to me. I think there is an easier way to deal with change availability, in `post_connect` move `await self.set_availability()` to the "nice to have" section. It should still timeout but then reconnect successfully. 

Also I'm interested in whether you have an update on the OpenDSR project, it'd be good to do something similar in NZ?

FingalP, 2022-07-19:
Hi @danharvey 

I went to the "Admin Panel" page and enter the PIN (which is just `admin`).

Then you can modify the websocket address (they call it Web Address) in the Backend Details tab. A restart is required after that.

Hope that helps!

closed 2022-09-10 by drc38