lbbrhzn/ocpp #511: Unknown Transaction ID with latest Pulsar Plus firmware 5.7.18

GrantKeymer, 2022-05-11:
**Describe the bug**
Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:1051
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 10:07:22 AM (255 occurrences)
Last logged: 7:00:00 PM

* Unknown transaction detected with id=1936029036
* Unknown transaction detected with id=1652017320

**To Reproduce**
Steps to reproduce the behavior:
1. Install Pulsar Plus firmware 5.7.18
2. Restart HA
3. Restart Pulsar Plus
4. Wait for HA to connect

**Expected behavior**
Charging sessions should start and stop without errors.  Instead, these "Unknown Transaction ID" errors occur every 15 minutes until a charging session starts.  Then the charger behaves normally, but as soon as the charging session stops, the same errors start happening again, every 15 minutes with each set of Meter Values sent by the charger.

**Screenshots**
Here is an excerpt from the Log File showing exactly what happens...

2022-05-11 18:45:14 DEBUG (MainThread) [websockets.server] < TEXT '[2,"532646386","MeterValues",\n{\n"meterValue":...tionId": 1936029036\n}]' [305 bytes]
2022-05-11 18:45:14 INFO (MainThread) [ocpp] : receive message [2,"532646386","MeterValues",
{
"meterValue": [
{
"timestamp": "2022-05-10T09:44:59Z",
"sampledValue": [
{
"format": "Raw",
"location": "Outlet",
"context": "Sample.Clock",
"measurand": "Energy.Active.Import.Register",
"unit": "Wh",
"value": "0"
}
]
}
],
"connectorId": 0,
"transactionId": 1936029036
}]
2022-05-11 18:45:14 WARNING (MainThread) [custom_components.ocpp] Unknown transaction detected with id=1936029036

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser Chrome
 - Version 100

**Additional context**
This problem occurs with both the latest version of OCPP integration v0.4.31 and also v0.4.29

GrantKeymer, 2022-05-16:
Now it's almost a week later and this issue seems to have disappeared for the past several days.  I haven't changed anything so can't explain why.

Best I close this issue, at least for now.

GrantKeymer, 2022-05-19:
Sadly, this issue is back...

Unknown transaction detected with id=1936029036
10:06:27 AM – (WARNING) Open Charge Point Protocol (OCPP) (custom integration) - message first occurred at May 19, 2022, 9:29:20 AM and shows up 82 times

This morning, my Pulsar Plus stopped communicating with OCPP early in the morning.  It needed to be rebooted before OCPP reconnected.  This is the second time it's happened.  Here is a log excerpt showing the sequence of events...

2022-05-20 04:39:02 DEBUG (MainThread) [websockets.server] > PING 35 e9 7d e6 [binary, 4 bytes]
2022-05-20 04:39:02 DEBUG (MainThread) [websockets.server] < PONG 35 e9 7d e6 [binary, 4 bytes]
2022-05-20 04:39:22 DEBUG (MainThread) [websockets.server] > PING f4 8f 9f c0 [binary, 4 bytes]
2022-05-20 04:40:02 DEBUG (MainThread) [websockets.server] > PING d9 93 cb c2 [binary, 4 bytes]
2022-05-20 04:40:42 DEBUG (MainThread) [websockets.server] > PING ac c9 51 bf [binary, 4 bytes]
2022-05-20 04:41:02 DEBUG (MainThread) [websockets.server] = connection is CLOSING
2022-05-20 04:41:02 DEBUG (MainThread) [websockets.server] > CLOSE 1000 (OK) [2 bytes]
2022-05-20 04:41:12 DEBUG (MainThread) [websockets.server] x half-closing TCP connection
2022-05-20 04:41:22 DEBUG (MainThread) [websockets.server] ! timed out waiting for TCP close
2022-05-20 04:41:22 DEBUG (MainThread) [websockets.server] x closing TCP connection
2022-05-20 04:41:22 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2022-05-20 04:41:22 INFO (MainThread) [websockets.server] connection closed
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] < GET / HTTP/1.1
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] < Connection: Upgrade
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] < Host: 192.168.1.6:9000
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Key: 6RMnLxWhj0BL9VO3ZFsAxw==
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Protocol: ocpp1.6
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Version: 13
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] < Upgrade: websocket
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] < User-Agent: WebSocket++/0.8.2
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] > HTTP/1.1 101 Switching Protocols
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] > Upgrade: websocket
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] > Connection: Upgrade
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Accept: EGwY3Go11ZgpPRlag16JikYVWcQ=
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Protocol: ocpp1.6
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] > Date: Thu, 19 May 2022 22:05:39 GMT
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] > Server: Python/3.9 websockets/10.2
2022-05-20 10:05:39 INFO (MainThread) [websockets.server] connection open
2022-05-20 10:05:39 DEBUG (MainThread) [websockets.server] = connection is OPEN

Not sure how long I can continue using this integration if this issue persists.  I really don't want to have to reboot my charger each morning before I can use it.  Obviously the charger still has a cloud connection because restarting via the Wallbox app works.  It's just the OCPP connection which dies.

dansarginson, 2022-05-21:
I also have this issue:

> "measurand": "Energy.Active.Import.Register",
> "unit": "Wh",
> "value": "37645"
> } 
> ]
> } 
> ],
> "connectorId": 1,
> "transactionId": 1936029036
> }]
> 2022-05-21 08:12:28 WARNING (MainThread) [custom_components.ocpp] Unknown transaction detected with id=1936029036
> 2022-05-21 08:12:28 INFO (MainThread) [ocpp] WallboxPulsar: send [3,"37487770",{}]
> 2022-05-21 08:12:28 INFO (MainThread) [ocpp] WallboxPulsar: receive message [2,"1745790417","MeterValues",
> {
> "meterValue": [
> {
> "timestamp": "2022-05-19T03:59:59Z",
> "sampledValue": [
> {
> "format": "Raw",
> "location": "Outlet",
> "context": "Sample.Clock",
> "measurand": "Energy.Active.Import.Register",
> "unit": "Wh",
> "value": "37645"
> } 
> ]
> } 
> ],

GrantKeymer, 2022-05-21:
Interesting that your Transaction ID is exactly the same as mine.  Due to this issue, as well as ongoing problems with frequent reconnects, I'm experimenting with another HA integration for Pulsar Plus which doesn't use OCPP...

https://github.com/hesselonline/wallbox

So far, so good.  Comms to the Wallbox Portal seems to be more reliable than OCPP, and if you install the custom components directly through HACS, you gain access to a Pause/Resume function (available when car is plugged in).  This is exactly what I've wanted from the beginning as it will allow me to suspend charging when insufficient solar power is available.

[image]

dansarginson, 2022-05-21:
Oh thanks for this tip. 

I'd spotted the identical transaction IDs too, but I haven't read the OCPP spec and didn't know whether they supposed to uniquely identify an instance of a transaction or identify a type of transaction.

GrantKeymer, 2022-05-21:
Yeah, I don't know what that particular transaction ID relates to.  What I do know is that this particular error never appeared until I started using version 5.7.18 of the Pulsar Plus firmware.

Searching on Google for problems with this version pointed me to the forum for the Wallbox Portal integration here...

https://community.home-assistant.io/t/wallbox-pulsar-plus-integration/200339/238

Others are using this updated firmware with good results, apart from the indicated Charging Power sometimes dropping to zero, even though it doesn't really.  I don't know if you've noticed it, but the Wallbox App sometimes does the same.  It seems a small price to pay for all the other benefits of that alternative integration.  Time I joined up with their forum and maybe continue this discussion over there.

Many thanks to Lbbrhzn and Drc38 for their help during the time I've spent here.

drc38, 2022-05-22:
Generally a purpose built integration/api for a charger will perform better and have more features than ocpp. Also unfortunately charger manufacturers continue to tweak firmware to change their ocpp messaging, often without a changelog as to what has been changed and why to help with updating the codebase. Thanks for your input @GrantKeymer

tipofthesowrd, 2022-05-24:
I'm having the same issue as described here. Next to that I'm experiencing frequent disconnects of OCPP with my Wallbox Pulsar Plus. However, I can still ping the device.

```
Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:1051
Integration: Open Charge Point Protocol (OCPP) ([documentation](https://github.com/lbbrhzn/ocpp/blob/main/README.md), [issues](https://github.com/lbbrhzn/ocpp/issues))
First occurred: 21 mei 2022 14:35:21 (413 occurrences)
Last logged: 20:12:01

    Unknown transaction detected with id=1653107553
    Unknown transaction detected with id=1936029036
```

I know the other integration mentioned in this thread, however be aware that this is not using local communications but rather relies on the Wallbox Cloud. Where OCPP is all local.

GrantKeymer, 2022-05-24:
> I'm having the same issue as described here. Next to that I'm experiencing frequent disconnects of OCPP with my Wallbox Pulsar Plus. However, I can still ping the device.
> 
And that is exactly what I found.  Here is my history chart showing reconnects...

[image]

The red part at the end where it says '2020' is after I upgraded the Pulsar Plus firmware to v5.7.18.  For many hours, the reconnects sat at 0, but for some unexplained reason they returned for a few hours at a time and eventually were as bad as ever.  I worked out a complicated sequence to restart HA, disable OCPP, restart Pulsar Plus, then re-enable OCPP and was able to restore reconnects to almost zero for a day or so, but then something would change and I'd be back to square one.

It all got too much in the end so I thought I'd give the alternate integration a try.  Here is my charger status now...

[image]

Since I started using this integration on May 21, there has been one short period showing 'Disconnected' for about 23 minutes, but apart from that, it just shows Paused or Charging.  That is what I've been looking for 😄 

> 
> I know the other integration mentioned in this thread, however be aware that this is not using local communications but rather relies on the Wallbox Cloud. Where OCPP is all local.

Yes I know it relies on the Wallbox Cloud.  However, the bizarre thing is that Wallbox Cloud is far more reliable than local OCPP!

If all else fails, I can revert to Bluetooth and the Wallbox App for emergencies.

drc38, 2022-05-27:
@lbbrhzn  any ideas on what's going on with the latest Wallbox firmware?

lbbrhzn, 2022-06-04:
I'm running 5.7.18 and it works fine. At most one reconnect per day.

tipofthesowrd, 2022-06-05:
When you say reconnect do you mean you have to manually restart the charger
via Bluetooth? Or does it reconnect automatically?

I can only reconnect it by removing power or rebooting via the app
connected via Bluetooth

closed 2022-08-12 by github-actions[bot]