lbbrhzn/ocpp #514: make the value of idTag in "RemoteStartTransaction" configurable

marcelvandorp, 2022-05-11:
I have an Alfen ICU Eve Mini with RFID reader as a charging point. I am able to connect this unit to HA. I can use the switch to make the Connector available, but I am unable to start a Charge Transaction.

When I try to start it, I can see in the Log file of the Charging point that a RemoteStartTransaction was rejected, because it contains an invalid IdTag.

The IdTag in the message is the serial number of the charging point, but the charging point expects a valid RFID Id.
An RFID ID contains a Hex number, so only 0-9A-F are accepted. And the Serial of my Charging point is EVC_0123 (fake ID, but syntactically correct). I cannot add that serial number to my Whitelist of accepted RFID IdTags.

Can this iDTag value made configurable?

drc38, 2022-05-13:
Hi, the Ocpp standard allows the idTag to be a 20 character string. Where are you limited to entering  hex characters only?

lukis101, 2022-06-27:
In OCPP 1.6 standard, section 3.9 it says that the tag data can be any format and that 
> ... Charge Points MUST NOT make any presumptions as to the format or content of such data...

**However** ran into different problem: using serial directly is no-go, because serial is defined as `CiString25Type`, while idTag is `CiString20Type` and thus it fails to send RemoteStart altogether whenever serial is longer than 20

drc38, 2022-06-27:
Try #552

lukis101, 2022-06-27:
That does the trick, since any(?) authrize request seems to get accepted.
Long-term still recommend having it configurable so one can use a real RFID tag to manually stop charge session (as it has to match) physically at the charger

drc38, 2022-06-27:
Take a look at #370 on how to configure and use rfid tags at the charger. If you start using remote start from the cms why can you not simply remove the cable from the car to stop the transaction or use remote stop from the cms?

closed 2022-09-03 by github-actions[bot]