lbbrhzn/ocpp #529: Reconfiguration menu

javierin, 2022-05-30:
## Version of the custom_component
main as of May 20, 2022

## Configuration
Not really sure how to get there

## Describe the bug
I have just installed this addon. However, I am being clumsy enough to not be able to find the configuration menu after set up. I know I can remove the integration and add it again to be able to take a screenshot of the settings, but I can't figure out how to get a modal window with the settings, just to see them or if I need to change the port because I need to make port forwarding of a different one.

Most probably the issue is sitting between the keyboard and the chair, but after readingthedocs.io I can't find out how to modify this addon's configuration.

Thanks

dreimer1986, 2022-05-30:
You can use the... manual way as I did sometimes already. If you have access to the /config/.storage folder (for example with the file editor addin)  you could access and edit the file "core.config_entries". Search for ocpp in it and you find the settings you once put in the intial configuration setup. Like host, port etcetc. But be careful to not destoy the yaml syntax in there.

But yes, a configuration menu would be a good idea.

dreimer1986, 2022-07-30:
Still a good idea, so I would say not stale

archerjm, 2022-09-16:
Now that the Force Smart Charging option has been added this has become more pressing.

drc38, 2022-09-16:
Adding an options flow to achieve this will be quite a bit of work. Personally I don't have the need for it, but happy to review a PR if someone wants to take it on.

GuzziDoug, 2022-11-10:
Easy way is to delete the app, then re-enable. This resets it to default. (Newbie way)

closed 2023-01-16 by github-actions[bot]