lbbrhzn/ocpp #550: Charger not available, how to diagnose errors

hackex, 2022-06-19:
Hey,
How can I find out what the problem is with the connection of a EVBox Elvi with this OCPP Backend in Home Assistant?
All the entities of the integration show "not available". Is there a way to see if Wallbox and integration try to communicate?
Thanks

drc38, 2022-06-20:
Check out some of the early issues with @dreimer1986 on how to diagnose using wireshark etc

closed 2022-08-26 by github-actions[bot]

ddlaat, 2025-04-01:
Was there ever a solution for this? I have the same problem, no entities show up.

dreimer1986, 2025-04-01:
Yes, as @drc38 said, check the reports where we fought with my EVBox ELVI and in general the logs of the addon are useful for getting information, too. There is no magic way to find out what's wrong. Get the logs as explained in the report again and read a bit in the old report and then open a report with your logs if you did not manage to find a solution in there. We managed to get my messy Wallbox running rock solid for years now and if you show some effort, maybe yours too. A good start would be not thinking we have a reliable glass bulb here. Put stuff like your current Home Assistant version, current Addon version, what Wallbox on what firmware and network info in your report. There is a template for making a report for a reason.