lbbrhzn/ocpp #596: charger_maximum_current list index out of range

bananaslug123, 2022-08-08:
**Describe the bug**
When attempting to change charger_maximum_current using the slide, HA reports list index out of range.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to HA Settings > Devices & Services > OCPP >charger device
2. Under Controls, move charger.Maximum_Current slider

**Expected behavior**
The new value selected on the slider would be transmitted to the EVSE.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Safari
 - Version: 103.0.1

**Additional context**
Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/ocpp/api.py:782
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 10:28:49 AM (7 occurrences)
Last logged: 10:45:45 AM

[140116270849696] list index out of range
[140114989849600] list index out of range
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1713, in async_call task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1750, in _execute_service await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 676, in entity_service_call future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 930, in async_request_call await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 713, in _handle_entity_call await result
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 111, in async_set_value await entity.async_set_native_value(native_value)
  File "/config/custom_components/ocpp/number.py", line 127, in async_set_native_value resp = await self.central_system.set_max_charge_rate_amps(
  File "/config/custom_components/ocpp/api.py", line 269, in set_max_charge_rate_amps return await self.charge_points[cp_id].set_charge_rate(limit_amps=value)
  File "/config/custom_components/ocpp/api.py", line 557, in set_charge_rate resp = await self.get_configuration(
  File "/config/custom_components/ocpp/api.py", line 782, in get_configuration value = resp.configuration_key[0][om.value.value]
IndexError: list index out of range

ccorderor, 2022-08-12:
I have the same problem (V2C Trydan).
The maximum current slider seems to be broken. I've seen a few changes in the last versions related to the number component, not sure if that might be related.

drc38, 2022-08-12:
Are you on the latest HA version? If not you need to be on at least 2022.8. Otherwise try a reinstall of the integration ensuring you set the max current to match your charger specs.

ccorderor, 2022-08-12:
> Are you on the latest HA version? If not you need to be on at least 2022.8. Otherwise try a reinstall of the integration ensuring you set the max current to match your charger specs.

Yes, i’m using the latest .8 version
It’s a 32A charger. The slider is fixed at 32A, and when i change it, nothing happens, it just goes back to 32A.

drc38, 2022-08-13:
@bananaslug123 it looks like your charger is isn't conforming to the ocpp standard. You'll need to log the actual messages and look at the response before the index error is thrown

drc38, 2022-09-10:
See how you get on with the latest release

closed 2022-11-18 by github-actions[bot]