lbbrhzn/ocpp #609: WARNING (MainThread) [custom_components.ocpp] Unknown transaction detected with id=xxx

Rep-Hueman, 2022-09-04:
Version v0.4.33

## Configuration

```yaml
Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:898
Integration: Open Charge Point Protocol (OCPP) ([documentation](https://github.com/lbbrhzn/ocpp/blob/main/README.md), [issues](https://github.com/lbbrhzn/ocpp/issues))
First occurred: 20:37:49 (173 occurrences)
Last logged: 23:29:35

Unknown transaction detected with id=1662160391
```

## Describe the bug

Unknown transaction detected with id=1662160391 in the log!

Charger is plug and charge. I don't use authorization.
Everything works perfectly but I keep getting this log message!

## Debug log

```text

2022-09-04 20:36:57.741 INFO (MainThread) [ocpp] CHHome-1: send [2,"41aafcac-da84-4a40-8660-a1d54fc1a55b","GetConfiguration",{"key":["SupportedFeatuCHrofiles"]}]
2022-09-04 20:36:57.783 INFO (MainThread) [ocpp] CHHome-1: receive message [3,"41aafcac-da84-4a40-8660-a1d54fc1a55b",{"configurationKey":[{"key":"SupportedFeatuCHrofiles","readonly":true,"value":"Core,FirmwareManagement,Reservation,LocalAuthListManagement,RemoteTrigger,SmartCharging"}]}]
2022-09-04 20:36:57.786 INFO (MainThread) [ocpp] CHHome-1: send [2,"6971c0d1-2c1c-4179-bb92-71b9cb4a9e02","GetConfiguration",{"key":["NumberOfConnectors"]}]
2022-09-04 20:36:57.834 INFO (MainThread) [ocpp] CHHome-1: receive message [3,"6971c0d1-2c1c-4179-bb92-71b9cb4a9e02",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}]}]
2022-09-04 20:36:57.838 INFO (MainThread) [ocpp] CHHome-1: send [2,"833f1b36-2e35-47fc-b390-0b1333dbbc8b","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2022-09-04 20:36:58.076 INFO (MainThread) [ocpp] CHHome-1: receive message [3,"833f1b36-2e35-47fc-b390-0b1333dbbc8b",{"status":"Accepted"}]
2022-09-04 20:36:58.078 INFO (MainThread) [ocpp] CHHome-1: send [2,"2e071d1c-5c67-476c-a047-93fc0c91c94e","GetConfiguration",{"key":["HeartbeatInterval"]}]
2022-09-04 20:36:58.116 INFO (MainThread) [ocpp] CHHome-1: receive message [3,"2e071d1c-5c67-476c-a047-93fc0c91c94e",{"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"3600"}]}]
2022-09-04 20:36:58.117 INFO (MainThread) [ocpp] CHHome-1: send [2,"a2b82540-0fa2-4f87-a650-b280d1ee596b","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2022-09-04 20:36:58.158 INFO (MainThread) [ocpp] CHHome-1: receive message [3,"a2b82540-0fa2-4f87-a650-b280d1ee596b",{"configurationKey":[{"key":"WebSocketPingInterval","readonly":false,"value":"60"}]}]
2022-09-04 20:36:58.159 INFO (MainThread) [ocpp] CHHome-1: send [2,"ef0611b6-6bbc-41d8-a0b2-071d16aa7065","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2022-09-04 20:36:58.198 INFO (MainThread) [ocpp] CHHome-1: receive message [3,"ef0611b6-6bbc-41d8-a0b2-071d16aa7065",{"configurationKey":[{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register,Power.Active.Import,Power.Factor,Current.Import,Current.Offered,Voltage,Frequency,Temperature"}]}]
2022-09-04 20:36:58.199 INFO (MainThread) [ocpp] CHHome-1: send [2,"4f2f130f-742a-448c-a24e-c2589470a02d","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2022-09-04 20:36:58.268 INFO (MainThread) [ocpp] CHHome-1: receive message [3,"4f2f130f-742a-448c-a24e-c2589470a02d",{"configurationKey":[{"key":"MeterValueSampleInterval","readonly":false,"value":"60"}]}]
2022-09-04 20:36:58.269 INFO (MainThread) [ocpp] CHHome-1: send [2,"be19852b-e378-48e2-ac82-a91bb02f643d","GetConfiguration",{"key":["ClockAlignedDataInterval"]}]
2022-09-04 20:36:58.309 INFO (MainThread) [ocpp] CHHome-1: receive message [3,"be19852b-e378-48e2-ac82-a91bb02f643d",{"configurationKey":[{"key":"ClockAlignedDataInterval","readonly":false,"value":"900"}]}]
2022-09-04 20:36:58.312 INFO (MainThread) [ocpp] CHHome-1: send [2,"3c9bb8cc-f8f4-4df9-8e22-5be836bcf90d","TriggerMessage",{"requestedMessage":"BootNotification"}]
2022-09-04 20:36:58.359 INFO (MainThread) [ocpp] CHHome-1: receive message [3,"3c9bb8cc-f8f4-4df9-8e22-5be836bcf90d",{"status":"Accepted"}]
2022-09-04 20:36:58.361 INFO (MainThread) [ocpp] CHHome-1: send [2,"9f2b53b6-b6b8-4ba5-bdc1-51198f6bb64c","TriggerMessage",{"requestedMessage":"StatusNotification","connectorId":0}]
2022-09-04 20:36:58.399 INFO (MainThread) [ocpp] CHHome-1: receive message [3,"9f2b53b6-b6b8-4ba5-bdc1-51198f6bb64c",{"status":"Accepted"}]
2022-09-04 20:36:58.401 INFO (MainThread) [ocpp] CHHome-1: send [2,"7e82cb39-9ec2-41a3-9de6-7afadf374263","TriggerMessage",{"requestedMessage":"StatusNotification","connectorId":1}]
2022-09-04 20:36:58.441 INFO (MainThread) [ocpp] CHHome-1: receive message [3,"7e82cb39-9ec2-41a3-9de6-7afadf374263",{"status":"Accepted"}]
2022-09-04 20:37:00.782 INFO (MainThread) [ocpp] CHHome-1: receive message [2,"16081","BootNotification",{"chargePointVendor":"Alfen BV","chargePointModel":"NG910-60023","chargePointSerialNumber":"ace0025893","chargeBoxSerialNumber":"CHHome-1","firmwareVersion":"5.8.0-4120"}]
2022-09-04 20:37:00.788 INFO (MainThread) [ocpp] CHHome-1: send [3,"16081",{"currentTime":"2022-09-04T18:37:00Z","interval":3600,"status":"Accepted"}]
2022-09-04 20:37:10.850 INFO (MainThread) [ocpp] CHHome-1: receive message [2,"16082","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available"}]
2022-09-04 20:37:10.855 INFO (MainThread) [ocpp] CHHome-1: send [3,"16082",{}]
2022-09-04 20:37:10.880 INFO (MainThread) [ocpp] CHHome-1: receive message [2,"16083","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available"}]
2022-09-04 20:37:10.880 INFO (MainThread) [ocpp] CHHome-1: send [3,"16083",{}]
2022-09-04 20:37:49.665 INFO (MainThread) [ocpp] CHHome-1: receive message [2,"16084","MeterValues",{"connectorId":1,"transactionId":1662160391,"meterValue":[{"timestamp":"2022-09-04T18:37:11Z","sampledValue":[{"value":"3058620.000","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L3"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Factor","location":"Outlet"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L3"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Offered","location":"Outlet","unit":"A"},{"value":"221.400","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L1-N"},{"value":"219.100","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L2-N"},{"value":"221.300","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-N"},{"value":"383.100","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L1-L2"},{"value":"383.000","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L2-L3"},{"value":"385.800","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-L1"},{"value":"50.050","context":"Sample.Periodic","measurand":"Frequency","location":"Outlet"},{"value":"43.000","context":"Sample.Periodic","measurand":"Temperature","location":"Body","unit":"Celsius"}]}]}]
2022-09-04 20:37:49.671 WARNING (MainThread) [custom_components.ocpp] Unknown transaction detected with id=1662160391
2022-09-04 20:37:49.674 INFO (MainThread) [ocpp] CHHome-1: send [3,"16084",{}]
2022-09-04 20:38:46.793 INFO (MainThread) [ocpp] CHHome-1: receive message [2,"16085","MeterValues",{"connectorId":1,"transactionId":1662160391,"meterValue":[{"timestamp":"2022-09-04T18:38:11Z","sampledValue":[{"value":"3058620.000","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"0.000", [... cut ...] "Voltage","location":"Outlet","unit":"V","phase":"L2-L3"},{"value":"386.100","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-L1"},{"value":"50.050","context":"Sample.Periodic","measurand":"Frequency","location":"Outlet"},{"value":"43.125","context":"Sample.Periodic","measurand":"Temperature","location":"Body","unit":"Celsius"}]}]}]
2022-09-04 20:39:28.441 WARNING (MainThread) [custom_components.ocpp] Unknown transaction detected with id=1662160391
```

drc38, 2022-09-05:
Possibly because you have two connectors, does it occur if you connect to connectorId:0 to charge?

Rep-Hueman, 2022-09-05:
I have 1 connector (Alfen Eve Single Pro-line)

drc38, 2022-09-06:
Sorry I was getting confused with the two status notifications in the log. Id 0 represents the charger and not a connector. Your log does not show the startTransaction from the charger however: 
`[2,"422d87b9-2818-151e-232b-d8a079ff5091","StartTransaction",{"idTag":"ENX-AU21060165","timestamp":"2022-09-01T08:48:56Z","meterStart":1613328,"connectorId":1}]`

the integration will then respond with the transactionId:
`[3,"422d87b9-2818-151e-232b-d8a079ff5091",{"transactionId":1662022130,"idTagInfo":{"status":"Accepted"}}]`

this Id is then used to validate the MeterValues messages

closed 2022-11-12 by github-actions[bot]