lbbrhzn/ocpp #649: Wallbox umable to connect

craigdabbs, 2022-10-14:
Wallbox pulsar is not able to conect to HA, i have a copy of the log file below
`This error originated from a custom integration.

Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:813
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 13:23:04 (3 occurrences)
Last logged: 13:24:11

Unexpected exception in connection to 'central': 'TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "'Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energy.Active.Import.Interval,Energy.Reactive.Import.Interval,Power.Active.Import,Power.Reactive.Import,Power.Offered,Power.Factor,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N,Frequency,RPM,SoC,Temperature,Current.Export.L1,Current.Export.L2,Current.Export.L3,Energy.Active.Export.Register,Energy.Reactive.Export.Register,Energy.Active.Export.Interval,Energy.Reactive.Export.Interval,Power.Active.Export,Power.Reactive.Export' is too long"}'
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/ocpp/messages.py", line 200, in validate_payload
    validator.validate(message.payload)
  File "/usr/local/lib/python3.10/site-packages/jsonschema/validators.py", line 304, in validate
    raise error
jsonschema.exceptions.ValidationError: 'Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energy.Active.Import.Interval,Energy.Reactive.Import.Interval,Power.Active.Import,Power.Reactive.Import,Power.Offered,Power.Factor,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N,Frequency,RPM,SoC,Temperature,Current.Export.L1,Current.Export.L2,Current.Export.L3,Energy.Active.Export.Register,Energy.Reactive.Export.Register,Energy.Active.Export.Interval,Energy.Reactive.Export.Interval,Power.Active.Export,Power.Reactive.Export' is too long

Failed validating 'maxLength' in schema['properties']['configurationKey']['items']['properties']['value']:
    {'maxLength': 500, 'type': 'string'}

On instance['configurationKey'][0]['value']:
    'Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energy.Active.Import.Interval,Energy.Reactive.Import.Interval,Power.Active.Import,Power.Reactive.Import,Power.Offered,Power.Factor,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N,Frequency,RPM,SoC,Temperature,Current.Export.L1,Current.Export.L2,Current.Export.L3,Energy.Active.Export.Register,Energy.Reactive.Export.Register,Energy.Active.Export.Interval,Energy.Reactive.Export.Interval,Power.Active.Export,Power.Reactive.Export'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 920, in run
    await asyncio.gather(*self.tasks)
  File "/config/custom_components/ocpp/api.py", line 415, in post_connect
    await self.configure(
  File "/config/custom_components/ocpp/api.py", line 813, in configure
    resp = await self.call(req)
  File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 289, in call
    validate_payload(response, self._ocpp_version)
  File "/usr/local/lib/python3.10/site-packages/ocpp/messages.py", line 209, in validate_payload
    raise TypeConstraintViolationError(
ocpp.exceptions.TypeConstraintViolationError: TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "'Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energy.Active.Import.Interval,Energy.Reactive.Import.Interval,Power.Active.Import,Power.Reactive.Import,Power.Offered,Power.Factor,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N,Frequency,RPM,SoC,Temperature,Current.Export.L1,Current.Export.L2,Current.Export.L3,Energy.Active.Export.Register,Energy.Reactive.Export.Register,Energy.Active.Export.Interval,Energy.Reactive.Export.Interval,Power.Active.Export,Power.Reactive.Export' is too long"}
`

HA Versions :
Home Assistant 2022.10.3
Supervisor 2022.10.0
Operating System 9.2

Wallbox is configured to connect to HA with ip address    192.168.0.25:9000
ID:  central
Pass:  charger

Hope someone can help, it did connect once and was able to control the charger but it lasted a day.

Many thanks

Craig

lbbrhzn, 2022-10-14:
Which sw version are you running on the wallbox? I've been running 5.7.18 for months now without issues.

craigdabbs, 2022-10-14:
hi,  ive taken a screenshot from the wallbox web portal

[image]

microfrost1, 2022-10-14:
Exactly the same issue for me. Happened when upgrading from HA 2022.10.1 to 2022.10.3. Rolling back HA has resolved.

craigdabbs, 2022-10-14:
i had trouble on older version of HA which is why i tried updating to latest with no effect.

drc38, 2022-10-14:
It looks like the issue is with an upstream package that validates the json passed between HA and the charger. Try reinstalling HA and only select a couple of measurands. If that doesn't throw the error then maybe the message will need to be split somehow.

craigdabbs, 2022-10-14:
i have installed HA onto my laptop and redirected the wallbox to connect to the new OCPP integration. Im met with the same error message. 

My laptop runs Arch Linux (up to date).

drc38, 2022-10-14:
Enable debug logging so the full ocpp message sent or received is available

craigdabbs, 2022-10-14:
This is the full log i think, i enable debug logging as shown in the docs.

Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/ocpp/messages.py", line 200, in validate_payload
validator.validate(message.payload)
File "/usr/local/lib/python3.10/site-packages/jsonschema/validators.py", line 304, in validate
raise error
jsonschema.exceptions.ValidationError: 'Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energy.Active.Import.Interval,Energy.Reactive.Import.Interval,Power.Active.Import,Power.Reactive.Import,Power.Offered,Power.Factor,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N,Frequency,RPM,SoC,Temperature,Current.Export.L1,Current.Export.L2,Current.Export.L3,Energy.Active.Export.Register,Energy.Reactive.Export.Register,Energy.Active.Export.Interval,Energy.Reactive.Export.Interval,Power.Active.Export,Power.Reactive.Export' is too long
Failed validating 'maxLength' in schema['properties']['configurationKey']['items']['properties']['value']:
{'maxLength': 500, 'type': 'string'}
On instance['configurationKey'][0]['value']:
'Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energy.Active.Import.Interval,Energy.Reactive.Import.Interval,Power.Active.Import,Power.Reactive.Import,Power.Offered,Power.Factor,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N,Frequency,RPM,SoC,Temperature,Current.Export.L1,Current.Export.L2,Current.Export.L3,Energy.Active.Export.Register,Energy.Reactive.Export.Register,Energy.Active.Export.Interval,Energy.Reactive.Export.Interval,Power.Active.Export,Power.Reactive.Export'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/config/custom_components/ocpp/api.py", line 920, in run
await asyncio.gather(*self.tasks)
File "/config/custom_components/ocpp/api.py", line 415, in post_connect
await self.configure(
File "/config/custom_components/ocpp/api.py", line 813, in configure
resp = await self.call(req)
File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 289, in call
validate_payload(response, self._ocpp_version)
File "/usr/local/lib/python3.10/site-packages/ocpp/messages.py", line 209, in validate_payload
raise TypeConstraintViolationError(
ocpp.exceptions.TypeConstraintViolationError: TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "'Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energy.Active.Import.Interval,Energy.Reactive.Import.Interval,Power.Active.Import,Power.Reactive.Import,Power.Offered,Power.Factor,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N,Frequency,RPM,SoC,Temperature,Current.Export.L1,Current.Export.L2,Current.Export.L3,Energy.Active.Export.Register,Energy.Reactive.Export.Register,Energy.Active.Export.Interval,Energy.Reactive.Export.Interval,Power.Active.Export,Power.Reactive.Export' is too long"}
2022-10-14 22:15:56.757 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to 'central'
2022-10-14 22:15:56.798 INFO (MainThread) [custom_components.ocpp] Charger central disconnected from 0.0.0.0:9000.
2022-10-14 22:15:56.799 INFO (MainThread) [websockets.server] connection closed

drc38, 2022-10-14:
What are the ocpp messages immediately before the error?

microfrost1, 2022-10-15:
Does this help?

```
2022-10-15 06:34:30.079 INFO (MainThread) [websockets.server] connection open
2022-10-15 06:34:30.079 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2022-10-15 06:34:30.079 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/Wallbox
2022-10-15 06:34:30.080 INFO (MainThread) [custom_components.ocpp] Charger Wallbox reconnected to 0.0.0.0:9000.
2022-10-15 06:34:30.080 DEBUG (MainThread) [custom_components.ocpp] Reconnect websocket to Wallbox
2022-10-15 06:34:30.163 INFO (MainThread) [ocpp] Wallbox: receive message [2,"888482339","StatusNotification",
{
"info": "",
"vendorId": "com.wallbox",
"vendorErrorCode": "",
"connectorId": 0,
"errorCode": "NoError",
"status": "Available",
"timestamp": "2022-10-15T05:34:29Z"
}]
2022-10-15 06:34:30.164 INFO (MainThread) [ocpp] Wallbox: send [3,"888482339",{}]
2022-10-15 06:34:30.205 INFO (MainThread) [ocpp] Wallbox: receive message [2,"1874960596","StatusNotification",
{
"info": "",
"vendorId": "com.wallbox",
"vendorErrorCode": "",
"connectorId": 1,
"errorCode": "NoError",
"status": "Available",
"timestamp": "2022-10-15T05:34:29Z"
}]
2022-10-15 06:34:30.205 INFO (MainThread) [ocpp] Wallbox: send [3,"1874960596",{}]
2022-10-15 06:34:32.082 INFO (MainThread) [ocpp] Wallbox: send [2,"c59a0556-dd71-4fc1-a8ea-992290886693","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2022-10-15 06:34:32.111 INFO (MainThread) [ocpp] Wallbox: receive message [3,"c59a0556-dd71-4fc1-a8ea-992290886693",
{
"unknownKey": [],
"configurationKey": [
{
"readonly": true,
"key": "SupportedFeatureProfiles",
"value": "Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger"
} 
]
}]
2022-10-15 06:34:32.112 DEBUG (MainThread) [custom_components.ocpp] Feature profiles returned: Profiles.AUTH|REM|SMART|FW|CORE
2022-10-15 06:34:32.113 INFO (MainThread) [ocpp] Wallbox: send [2,"01198455-ed73-4a5d-85a2-30e193c2c6e1","GetConfiguration",{"key":["NumberOfConnectors"]}]
2022-10-15 06:34:32.130 INFO (MainThread) [ocpp] Wallbox: receive message [3,"01198455-ed73-4a5d-85a2-30e193c2c6e1",
{
"unknownKey": [],
"configurationKey": [
{
"readonly": true,
"key": "NumberOfConnectors",
"value": "1"
} 
]
}]
2022-10-15 06:34:32.131 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for NumberOfConnectors: 1
2022-10-15 06:34:32.131 INFO (MainThread) [ocpp] Wallbox: send [2,"e9156c80-ecce-4459-9dd0-ff19235d221e","GetConfiguration",{"key":["HeartbeatInterval"]}]
2022-10-15 06:34:32.166 INFO (MainThread) [ocpp] Wallbox: receive message [3,"e9156c80-ecce-4459-9dd0-ff19235d221e",
{
"unknownKey": [],
"configurationKey": [
{
"readonly": false,
"key": "HeartbeatInterval",
"value": "3600"
} 
]
}]
2022-10-15 06:34:32.167 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for HeartbeatInterval: 3600
2022-10-15 06:34:32.167 INFO (MainThread) [ocpp] Wallbox: send [2,"d01bed74-d753-452a-9511-951d4470c43d","GetConfiguration",{"key":["MeterValuesSampledData"]}]
[... 14 lines cut ...]
    validator.validate(message.payload)
  File "/usr/local/lib/python3.10/site-packages/jsonschema/validators.py", line 304, in validate
    raise error
jsonschema.exceptions.ValidationError: 'Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energy.Active.Import.Interval,Energy.Reactive.Import.Interval,Power.Active.Import,Power.Reactive.Import,Power.Offered,Power.Factor,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N,Frequency,RPM,SoC,Temperature,Current.Export.L1,Current.Export.L2,Current.Export.L3,Energy.Active.Export.Register,Energy.Reactive.Export.Register,Energy.Active.Export.Interval,Energy.Reactive.Export.Interval,Power.Active.Export,Power.Reactive.Export' is too long
Failed validating 'maxLength' in schema['properties']['configurationKey']['items']['properties']['value']:
    {'maxLength': 500, 'type': 'string'}
On instance['configurationKey'][0]['value']:
    'Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energy.Active.Import.Interval,Energy.Reactive.Import.Interval,Power.Active.Import,Power.Reactive.Import,Power.Offered,Power.Factor,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N,Frequency,RPM,SoC,Temperature,Current.Export.L1,Current.Export.L2,Current.Export.L3,Energy.Active.Export.Register,Energy.Reactive.Export.Register,Energy.Active.Export.Interval,Energy.Reactive.Export.Interval,Power.Active.Export,Power.Reactive.Export'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 920, in run
    await asyncio.gather(*self.tasks)
  File "/config/custom_components/ocpp/api.py", line 415, in post_connect
    await self.configure(
  File "/config/custom_components/ocpp/api.py", line 813, in configure
    resp = await self.call(req)
  File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 289, in call
    validate_payload(response, self._ocpp_version)
  File "/usr/local/lib/python3.10/site-packages/ocpp/messages.py", line 209, in validate_payload
    raise TypeConstraintViolationError(
ocpp.exceptions.TypeConstraintViolationError: TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "'Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energ [... cut ...] .L2,Current.Export.L3,Energy.Active.Export.Register,Energy.Reactive.Export.Register,Energy.Active.Export.Interval,Energy.Reactive.Export.Interval,Power.Active.Export,Power.Reactive.Export' is too long"}
2022-10-15 06:34:32.190 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to 'Wallbox'
2022-10-15 06:34:32.215 INFO (MainThread) [custom_components.ocpp] Charger Wallbox disconnected from 0.0.0.0:9000.
2022-10-15 06:34:32.215 INFO (MainThread) [websockets.server] connection closed
```

drc38, 2022-10-15:
As per above it suggests you've selected too many measurands when first setting up the integration

microfrost1, 2022-10-15:
I removed the integration and readded it with just a couple of measurands via 'Devices and Setvices' before the logs. Do I need to remove and reinstall via HACS? If it does need to be a new HA install, I can try and test later on a new instance.

drc38, 2022-10-15:
Should be fine removing via devices and services. Is the error still occurring?

microfrost1, 2022-10-15:
Yeah same error unfortunately.

drc38, 2022-10-15:
The issue is coming from the charger, can you use the Occp service Set Configuration to change MeterValuesSampledData to Energy.Active.Import.Register for example?

craigdabbs, 2022-10-15:
thanks for the replies drc38,  ive removed the integration numerous times with different settings and measurands.  Ive also done a fresh install of HA on a different platform with the same errors.
Im unsure how i go about your previous suggestion, could you give me some guidance of where to start?

many thanks
craig

drc38, 2022-10-15:
[image]

drc38, 2022-10-15:
I think you'll need to comment out line 813 in api.py before using the service call above:
`resp = await self.call(req)`

drc38, 2022-10-15:
> Do i need to edit the api.py file before i try and call the service. I only ask as that service does not exist in my setup, is this becuase the wallbox has not connected correctly to start with? [image]

Yes edit the api first, you may not need to call the service as it should setup itself with the measurands selected.

craigdabbs, 2022-10-15:
Its staying connected for longer now, it still randomly disconnects.  I was able to turn on a charge session but could not adjust the charge amps.  Then it would become unavailable again.
I have 2 errors now 1 regarding set max charge rate and a task exception.

-------------------------------------------------------------------------------------------------------
This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/ocpp/api.py:299
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 12:16:17 (11 occurrences)
Last logged: 12:23:09

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 299, in update
    for ent in entity_registry.async_entries_for_device(er, dev.id):
AttributeError: 'NoneType' object has no attribute 'id'

----------------------------------------------------------------------------------------------------------

This error originated from a custom integration.

Logger: homeassistant.core
Source: custom_components/ocpp/api.py:787
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 12:22:57 (1 occurrences)
Last logged: 12:22:57

Error executing service: <ServiceCall number.set_value (c:01GFDP2A45D3H16Q279J2STNPF): value=6.0, entity_id=['number.charger_maximum_current']>
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/queues.py", line 159, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 274, in call
    await self._get_specific_response(call.unique_id,
  File "/config/custom_components/ocpp/api.py", line 852, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
  File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 307, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(),
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1756, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 207, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 678, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 109, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/ocpp/number.py", line 127, in async_set_native_value
    resp = await self.central_system.set_max_charge_rate_amps(
  File "/config/custom_components/ocpp/api.py", line 271, in set_max_charge_rate_amps
    return await self.charge_points[cp_id].set_charge_rate(limit_amps=value)
  File "/config/custom_components/ocpp/api.py", line 578, in set_charge_rate
    resp = await self.get_configuration(
  File "/config/custom_components/ocpp/api.py", line 787, in get_configuration
    resp = await self.call(req)
  File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 277, in call
    raise asyncio.TimeoutError(
asyncio.exceptions.TimeoutError: Waited 30s for response on [2,"e41a861e-1dae-46eb-b5e9-4dff0134b7ba","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}].

craigdabbs, 2022-10-15:
I removed the OCPP integration completely to rese everything back to default.  Its connecting fine now, but still no control over the max charge rate.

craigdabbs, 2022-10-15:
Max charge rate was working i do apologise, you need to have a active charge session running for it to take effect.  I dont know what is different this time but it seems pretty stable and all working.  Ill give it a few days.

microfrost1, 2022-10-16:
I can confirm commenting out the lines in api.py got everything back up a running.

@drc38 Is there anything I can do to help to get a a permanent fix merged in?

GrantKeymer, 2022-11-21:
@microfrost1 @craigdabbs if you're still unhappy with the reliability of this OCPP integration with your Wallbox, there is an alternative...

[https://www.home-assistant.io/integrations/wallbox]

I struggled on with OCPP for several months before giving up and switching to the integration mentioned above.  It was an easy change and I haven't looked back.

closed 2023-01-28 by github-actions[bot]