lbbrhzn/ocpp #677: PropertyConstraintViolationError: When chargingProfilePurpose =ChargingStationMaxProfile, chargingProfileKind SHALL NOT be Relative

skygnome, 2022-11-06:
**Describe the bug**
I've managed to get an Autel Maxicharger setup with HA / OCPP. Most of the functions are working and the charger is reporting the correct details back. One of the features is failing to work. I'm unable to change the charging amps. It shows the following error in the logs and on the dashboard when making the change.

 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [2759196832] PropertyConstraintViolationError: When chargingProfilePurpose =ChargingStationMaxProfile, chargingProfileKind SHALL NOT be Relative, {}

**To Reproduce**
Steps to reproduce the behavior:
Open the charger device page from Settings -> integrations
Attempt to change the Maximum_Current

**Expected behavior**
Charge currently to change on the device when changed within HA

**Additional context**
HA Version: 2022.11.1 (Docker)
OCPP Version: 0.4.36

Debug Log:

2022-11-06 11:46:23.582 INFO (MainThread) [ocpp] charger: receive message [2,"2022110611462200000002","StatusNotification",{"connectorId":0,"status":"Available","errorCode":"NoError"}]
2022-11-06 11:46:23.583 INFO (MainThread) [ocpp] charger: send [3,"2022110611462200000002",{}]
2022-11-06 11:46:24.816 INFO (MainThread) [ocpp] charger: receive message [2,"2022110611462200000003","StatusNotification",{"connectorId":1,"status":"Preparing","errorCode":"NoError","info":"eCp_9V"}]
2022-11-06 11:46:24.817 INFO (MainThread) [ocpp] charger: send [3,"2022110611462200000003",{}]
2022-11-06 11:46:37.533 INFO (MainThread) [ocpp] charger: send [2,"3e5a4a74-5f0d-4336-89d1-57f2096dd50c","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2022-11-06 11:46:38.959 INFO (MainThread) [ocpp] charger: receive message [3,"3e5a4a74-5f0d-4336-89d1-57f2096dd50c",{"configurationKey":[{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"}]}]
2022-11-06 11:46:38.961 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargingScheduleAllowedChargingRateUnit: Current,Power
2022-11-06 11:46:38.961 INFO (MainThread) [custom_components.ocpp] Charger supports setting the following units: Current,Power
2022-11-06 11:46:38.961 INFO (MainThread) [custom_components.ocpp] If more than one unit supported default unit is Amps
2022-11-06 11:46:38.962 INFO (MainThread) [ocpp] charger: send [2,"8a86b976-9f4c-4e42-9159-4b2e4f332fac","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2022-11-06 11:46:40.356 INFO (MainThread) [ocpp] charger: receive message [3,"8a86b976-9f4c-4e42-9159-4b2e4f332fac",{"configurationKey":[{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"20"}]}]
2022-11-06 11:46:40.358 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargeProfileMaxStackLevel: 20
2022-11-06 11:46:40.362 INFO (MainThread) [ocpp] charger: send [2,"e55ec995-492c-4e35-89b8-4d4d1e451cb4","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":20,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":11.0}]}}}]
2022-11-06 11:46:40.891 INFO (MainThread) [ocpp] charger: receive message [4,"e55ec995-492c-4e35-89b8-4d4d1e451cb4","PropertyConstraintViolation","When chargingProfilePurpose =ChargingStationMaxProfile, chargingProfileKind SHALL NOT be Relative",{}]
2022-11-06 11:46:40.894 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [2759196832] PropertyConstraintViolationError: When chargingProfilePurpose =ChargingStationMaxProfile, chargingProfileKind SHALL NOT be Relative, {}
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 200, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1744, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1781, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 208, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 678, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 943, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 109, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/ocpp/number.py", line 125, in async_set_native_value
    resp = await self.central_system.set_max_charge_rate_amps(
  File "/config/custom_components/ocpp/api.py", line 271, in set_max_charge_rate_amps
    return await self.charge_points[cp_id].set_charge_rate(limit_amps=value)
  File "/config/custom_components/ocpp/api.py", line 600, in set_charge_rate
    resp = await self.call(req)
  File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 274, in call
    await self._get_specific_response(call.unique_id,
  File "/config/custom_components/ocpp/api.py", line 855, in _get_specific_response
    raise resp.to_exception()
ocpp.exceptions.PropertyConstraintViolationError: PropertyConstraintViolationError: When chargingProfilePurpose =ChargingStationMaxProfile, chargingProfileKind SHALL NOT be Relative, {}

drc38, 2022-11-06:
Looks like a quirk of your charger. You can try changing the line of code to `Absolute` or `Recurring` instead

dmarafino, 2022-11-17:
Can you share what other changes you made to get your configuration working? Mine is working but gives errors occasionally and frequently becomes unavailable. 

https://github.com/lbbrhzn/ocpp/discussions/669

manzanita0412, 2022-12-11:
I am also having this same issue with my new Autel Maxicharger. Managed to get it to connect to my OCPP instance, but see this same error when attempting to change the current limit with this integration via Homeassistant.

closed 2023-02-16 by github-actions[bot]