lbbrhzn/ocpp #715: ocpp with vestle 04

avico78, 2022-12-17:
would be really great if this intergation could explain what functionality it can provide ,
specifically asking ,would it be possible to turn on /off charging ?
setting current amperage ?

vestle 04

N3rdix, 2022-12-20:
All possible, I use a Vestel EVC04 as well.
The device is also listed in the docs, so I guess this one can be closed...

Matthias0705, 2022-12-29:
> All possible, I use a Vestel EVC04 as well. The device is also listed in the docs.

Hi, I'm having the issue that I can't update for settings sice aprox. one week. To you have comparable issue? Brefore everything went fine....e.g I still can start the charging with the change of the wallbox_charge_control parameter but it's always giving off as response.

N3rdix, 2022-12-30:
First difference might be that I am still on version 0.33, but I assume for you it's not an update issue.

Do you see any errors in the HA log?

I am happy to help, so let's collect insights:
- Firmware version of your Vestel
- Anything set in OCPP parameter "MeterValuesSampledData"? If yes, what?
- You can check the logs from the Wallbox to see what happens when you start charging (before sharing the whole zip file make sure to delete sensitive parts, as far as I remember it collects a lot of environment data)
- Additionally set the logger on HA side to DEBUG before you start charging, e.g. until next restart with this service call:
`service: logger.set_level
data:
  ocpp: debug`

Matthias0705, 2023-01-11:
Dear N3rdix, 
thanks for responding and supporting. Before coming to the details. I thought the topic was solved after deactivating and activating the OCPP in the Wallbox but after one week the issue occured again. 

Regarding your questions:
- Firmware version of your Vestel: v3.89.0
- Anything set in OCPP parameter "MeterValuesSampledData"? If yes, what?: Yes, set to Energy.Active.Import.Register
- You can check the logs from the Wallbox to see what happens when you start charging:

11.01.2023-20:10:43 [INFO] CLIENT: Client RX: [2,"xxx","GetConfiguration",{"key":["AuthorizeRemoteTxRequests"]}]11.01.2023-20:10:43 [INFO] CMD: "GetConfiguration"
11.01.2023-20:10:43 [INFO] UID: xxx
11.01.2023-20:10:43 [INFO] JSON STRING: GetConfiguration",{"key":["AuthorizeRemoteTxRequests"]}
11.01.2023-20:10:43 [INFO] Get Configuration Message Received :
11.01.2023-20:10:43 [INFO] Object expected
11.01.2023-20:10:43 [INFO] Unexpected key: {"key":["AuthorizeRemoteTxRequests"]}
11.01.2023-20:10:43 [INFO] - key: ["AuthorizeRemoteTxRequests"]
11.01.2023-20:10:43 [INFO] GetConfig requested_key : ["AuthorizeRemoteTxRequests"]
11.01.2023-20:10:43 [INFO] Unexpected key: AuthorizeRemoteTxRequests
11.01.2023-20:10:43 [INFO] applicationGetBUFFER : ["AuthorizeRemoteTxRequests"]
11.01.2023-20:10:43 [INFO] Response is coming:1 
11.01.2023-20:10:43 [INFO] array length: 1
array index: 0
11.01.2023-20:10:43 [INFO] 
header.message_length: 21011.01.2023-20:10:43 [INFO] getConfigurationsResponseWithMoreKeys
11.01.2023-20:10:43 [INFO] send data2
11.01.2023-20:10:43 [INFO] Write Data: [3,"xxx",{
    "configurationKey": [
        {
            "key": "AuthorizeRemoteTxRequests",
            "readonly": false,
            "value": "FALSE"
        }
    ]
    }]
11.01.2023-20:10:43 [INFO] CLIENT: Client RX: [2,"xxx","RemoteStartTransaction",{"idTag":"1","connectorId":1}]11.01.2023-20:10:43 [INFO] CMD: "RemoteStartTransaction"
11.01.2023-20:10:43 [INFO] UID: xxx
11.01.2023-20:10:43 [INFO] JSON STRING: RemoteStartTransaction",{"idTag":"1","connectorId":1}
11.01.2023-20:10:43 [INFO] Remote Start Transaction Message Received :
11.01.2023-20:10:43 [INFO] Object expected
11.01.2023-20:10:43 [INFO] Unexpected key: {"idTag":"1","connectorId":1}
11.01.2023-20:10:43 [INFO] upper idtag: 1
11.01.2023-20:10:43 [INFO] rsv enable: 0
11.01.2023-20:10:43 [INFO] idtag res: 
11.01.2023-20:10:43 [INFO] upper idtag: 1
11.01.2023-20:10:43 [INFO] 
header.message_length: 6411.01.2023-20:10:43 [INFO] remote_id_send : 1
11.01.2023-20:10:43 [INFO] rfid_tag_send : 0
11.01.2023-20:10:43 [INFO] free_mode_send : 0
11.01.2023-20:10:43 [INFO] BUFFER: 

{
    "type": "AuthorizationResponse",
    "idTagInfo": {
        "status": "Accepted",
        "idTag": "1",
        "remote": "Start"
    }
    }11.01.2023-20:10:43 [INFO] BUFFER: 

{
    "type": "command",
    "data": {
        "commandId": 1,
        "payload": ""
    }
    }11.01.2023-20:10:43 [INFO] send data2
11.01.2023-20:10:43 [INFO] Write Data: [3,"xxx",{"status":"Accepted"}]

I'm pretty new with HA. Where to put the debug comment?

N3rdix, 2023-01-11:
Looks all fine for me, except 2 weird info messages in your logs which I currently can't point:
> 11.01.2023-20:10:43 [INFO] Unexpected key: {"idTag":"1","connectorId":1}
> 11.01.2023-20:10:43 [INFO] Unexpected key: {"key":["AuthorizeRemoteTxRequests"]}

Besides those message I hope the debug log from HA side will make it more obvious.

To enable debug on HA side either go to developer tools -> services -> call the service i mentioned earlier,
or to persist the settings over HA restarts just put this into your configuration.yaml file:
```
logger:
  logs:
    ocpp: debug
```

Matthias0705, 2023-01-12:
Also have those massages (Could this cause any issue?):
[image]

N3rdix, 2023-01-12:
I think I had a similar warning a year ago, what helped was to reduce the amount of measurands (the list in the warning) until I found the one not supported (can't remember which one it was).

I am not sure if the measurands sent to the Wallbox can be adjusted by selecting less options in the setup process (https://github.com/lbbrhzn/ocpp/blob/main/docs/user-guide.md#configuring-the-central-system).
I did it the "dirty" way by adjusting the HA Core config directly. I will send you the list of measurands working for me tomorrow if this helps.

maybe it's worth a try to configure OCPP once again with just 1 or 2 measurands to see if this resolves the issue.

drc38, 2023-01-12:
If the charger documentation isn't explicit on what it supports then best to get them off @n3rdix. To change in HA remove and reinstall integration, selecting only those you need and are supported.

N3rdix, 2023-01-12:
Thanks @drc38, appreciate that good hint.

Unfortunately the documentation from Vestel is unbelievably bad, that's the reason why I did trial and error to find the maximum set of measurands some time ago :-) 

I can add it to the documentation of recommended measurands tomorrow to give a better starting point for new Vestel users

N3rdix, 2023-01-13:
> Also have those massages (Could this cause any issue?): [image]

Hi @Matthias0705, the working set of measurands for my Vestel are the following, maybe you can report back once you re-configued OCPP and tested.

- Active energy imported from the grid during last interval
- Active energy imported from the grid
- Instantaneous current flow to EV
- Instantaneous active power imported by EV
- Maximum power offered to EV

closed 2023-03-22 by github-actions[bot]