lbbrhzn/ocpp #716: Integration crashing while communicating with Weidmüller Business (Wallbe backend)

linuxstony, 2022-12-18:
**Describe the bug**
I am able to connect the Wallbox with your integration just fine - the values are beeing displayed  - then after a few seconds the integration seems to crash and to reboot. 

**To Reproduce**
Hard to tell.
**Expected behavior**
No crashes :)

**Screenshots**
[image]
[image]
[image]

**Additional context**

The Wallbox uses a Wallbe-Backend which claims to be OCPP 1.6 compatible "

A part of the logfile:

2022-12-18 12:58:27.446 ERROR (MainThread) [custom_components.ocpp] Unexpected exception in connection to '': 'argument of type 'int' is not iterable'
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 920, in run
    await asyncio.gather(*self.tasks)
  File "/config/custom_components/ocpp/api.py", line 451, in post_connect
    if prof.SMART in self._attr_supported_features:
TypeError: argument of type 'int' is not iterable
2022-12-18 12:58:27.496 INFO (MainThread) [ocpp] : receive message [3,"e9a5be5f-734e-4b9d-bd68-5c07375afb33",{"status":"Rejected"}]
2022-12-18 12:58:27.499 WARNING (MainThread) [custom_components.ocpp] Rejected while setting MeterValuesSampledData to Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energy.Active.Import.Interval,Energy.Reactive.Import.Interval,Power.Active.Import,Power.Reactive.Import,Power.Offered,Power.Factor,Current.Import,Current.Offered,Voltage,Frequency,SoC,Temperature
2022-12-18 12:58:27.509 INFO (MainThread) [ocpp] : send [2,"d2b65ea1-d04d-47a6-92c6-95ba134c85d5","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2022-12-18 12:58:27.618 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 419, in post_connect
    await self.configure(
  File "/config/custom_components/ocpp/api.py", line 813, in configure
    resp = await self.call(req)
  File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 271, in call
    await self._send(call.to_json())
  File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 325, in _send
    await self._connection.send(message)
  File "/usr/local/lib/python3.10/site-packages/websockets/legacy/protocol.py", line 635, in send
    await self.ensure_open()
  File "/usr/local/lib/python3.10/site-packages/websockets/legacy/protocol.py", line 953, in ensure_open
    raise self.connection_closed_exc()
websockets.exceptions.ConnectionClosedOK: sent 1000 (OK); then received 1000 (OK)
2022-12-18 12:58:29.045 WARNING (MainThread) [homeassistant.components.camera] Updating xiaomi_cloud_map_extractor camera took longer than the scheduled update interval 0:00:05
2022-12-18 12:58:34.047 WARNING (MainThread) [homeassistant.components.camera] Updating xiaomi_cloud_map_extractor camera took longer than the scheduled update interval 0:00:05
2022-12-18 12:58:39.054 WARNING (MainThread) [homeassistant.components.camera] Updating xiaomi_cloud_map_extractor camera took longer than the scheduled update interval 0:00:05
2022-12-18 12:58:49.065 WARNING (MainThread) [homeassistant.components.camera] Updating xiaomi_cloud_map_extractor camera took longer than the scheduled update interval 0:00:05
2022-12-18 12:59:19.080 WARNING (MainThread) [homeassistant.components.camera] Updating xiaomi_cloud_map_extractor camera took longer than the scheduled update interval 0:00:05
2022-12-18 12:59:24.081 WARNING (MainThread) [homeassistant.helpers.entity] Update of camera.xiaomi_cloud_map_extractor is taking over 10 seconds
2022-12-18 12:59:24.084 WARNING (MainThread) [homeassistant.components.camera] Updating xiaomi_cloud_map_extractor camera took longer than the scheduled update interval 0:00:05
2022-12-18 12:59:24.252 WARNING (SyncWorker_0) [custom_components.xiaomi_cloud_map_extractor.camera] Unable to retrieve map data
2022-12-18 12:59:28.238 INFO (MainThread) [ocpp] : receive message [2,"16","BootNotification",{"chargePointVendor":"wallbe","chargePointModel":"Smart Charge Control","chargePointSerialNumber":"00000000596","firmwareVersion":"BF-01.04.13"}]
2022-12-18 12:59:28.242 INFO (MainThread) [ocpp] : send [3,"16",{"currentTime":"2022-12-18T12:59:28Z","interval":3600,"status":"Accepted"}]
2022-12-18 12:59:28.561 INFO (MainThread) [ocpp] : receive message [2,"17","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","vendorErrorCode":"0","timestamp":"2022-12-18T12:59:28.002Z"}]
2022-12-18 12:59:28.566 INFO (MainThread) [ocpp] : send [3,"17",{}]
2022-12-18 12:59:28.688 INFO (MainThread) [ocpp] : receive message [2,"18","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","vendorErrorCode":"0","timestamp":"2022-12-18T12:59:28.003Z"}]
2022-12-18 12:59:28.692 INFO (MainThread) [ocpp] : send [3,"18",{}]
2022-12-18 12:59:29.966 INFO (MainThread) [ocpp] : send [2,"78f2bb8c-c5b8-4f00-bc68-bbcb0c354b3c","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2022-12-18 12:59:30.032 INFO (MainThread) [ocpp] : receive message [3,"78f2bb8c-c5b8-4f00-bc68-bbcb0c354b3c",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"true"}],"unknownKey":[]}]
2022-12-18 12:59:30.034 WARNING (MainThread) [custom_components.ocpp] Unknown feature profile detected ignoring: true
2022-12-18 12:59:30.037 INFO (MainThread) [ocpp] : send [2,"a3fb4c6e-d0ad-41af-82b9-4ea3c7124737","GetConfiguration",{"key":["NumberOfConnectors"]}]
2022-12-18 12:59:30.107 INFO (MainThread) [ocpp] : receive message [3,"a3fb4c6e-d0ad-41af-82b9-4ea3c7124737",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}],"unknownKey":[]}]
2022-12-18 12:59:30.110 INFO (MainThread) [ocpp] : send [2,"51ec4307-76e1-42d5-8296-083740551d1a","GetConfiguration",{"key":["HeartbeatInterval"]}]
2022-12-18 12:59:30.237 INFO (MainThread) [ocpp] : receive message [3,"51ec4307-76e1-42d5-8296-083740551d1a",{"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"3600"}],"unknownKey":[]}]
2022-12-18 12:59:30.245 INFO (MainThread) [ocpp] : send [2,"728dd2d7-8912-4356-ad74-025887f59bf4","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2022-12-18 12:59:30.363 INFO (MainThread) [ocpp] : receive message [3,"728dd2d7-8912-4356-ad74-025887f59bf4",{"configurationKey":[{"key":"MeterValuesSampledData","readonly":true,"value":"true"}],"unknownKey":[]}]
2022-12-18 12:59:30.369 WARNING (MainThread) [custom_components.ocpp] MeterValuesSampledData is a read only setting
2022-12-18 12:59:30.376 INFO (MainThread) [ocpp] : send [2,"e6ab4db5-adb5-40c6-814a-03677678dc27","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2022-12-18 12:59:30.437 INFO (MainThread) [ocpp] : receive message [3,"e6ab4db5-adb5-40c6-814a-03677678dc27",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"true"}],"unknownKey":[]}]
[... 84 lines cut ...]
2022-12-18 13:00:35.115 ERROR (MainThread) [custom_components.ocpp] Unexpected exception in connection to '': 'argument of type 'int' is not iterable'
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 920, in run
    await asyncio.gather(*self.tasks)
  File "/config/custom_components/ocpp/api.py", line 451, in post_connect
    if prof.SMART in self._attr_supported_features:
TypeError: argument of type 'int' is not iterable
2022-12-18 13:00:35.217 INFO (MainThread) [ocpp] : receive message [3,"08e11d44-2b95-48b5-8b25-f2d0288da05e",{"status":"Rejected"}]
2022-12-18 13:00:35.239 WARNING (MainThread) [custom_components.ocpp] Rejected while setting MeterValuesSampledData to Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energy.Active.Import.Interval,Energy.Reactive.Import.Interval,Power.Active.Import,Power.Reactive.Import,Power.Offered,Power.Factor,Current.Import,Current.Offered,Voltage,Frequency,SoC,Temperature
2022-12-18 13:00:35.276 INFO (MainThread) [ocpp] : send [2,"485407b6-566d-4b2e-8ead-bc53b31b07e2","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2022-12-18 13:00:35.358 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 419, in post_connect
    await self.configure(
  File "/config/custom_components/ocpp/api.py", line 813, in configure
    resp = await self.call(req)
  File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 271, in call
    await self._send(call.to_json())
  File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 325, in _send
    await self._connection.send(message)
  File "/usr/local/lib/python3.10/site-packages/websockets/legacy/protocol.py", line 635, in send
    await self.ensure_open()
  File "/usr/local/lib/python3.10/site-packages/websockets/legacy/protocol.py", line 953, in ensure_open
    raise self.connection_closed_exc()
websockets.exceptions.ConnectionClosedOK: sent 1000 (OK); then received 1000 (OK)

drc38, 2022-12-19:
It looks like your charger is not ocpp compliant, read through thread #442 by way of example. You will need to go through and comment out the messages it does not support.

linuxstony, 2022-12-20:
Alright - thank you for pointing me into the right direction. 

I also asked the vendor for a firmware upgrade and reported the problems with the OCPP interface - but i doubt, that they will be of much help because they even declined me any additional information about the backend at all - or even a password to access the web frontend (luckily the password was available somewhere on the internet)

I will debug alongsinde your mentioned thread and when I am finished write a guide what to do exactly for others to reproduce to get this one working. 

I appreciate your all time for looking into my problem.

drc38, 2022-12-21:
You could also try adding `self._attr_supported_features |= prof.CORE` before line 511 in `api.py`

linuxstony, 2022-12-21:
I removed the integration altogether and reinstalled it. I chose another version to install "main" and activated the options on the bottom.  

I activated the option to skip the OCPP-scheme-validation and to force the Smart-Charging-Profile. (Try and error)

Since then I got the Wallbox functioning without any hassle or reboot so far. 

I think with these options this box can be added to the supported devices just fine - when its desired I can summarize which options are available and which are not (e.g. SoC or Frequency are not returned as values so i removed them from the dashboard.

The slider for setting the Maximum Current is available and moveable - I wasn't yet able to confirm, that this is really set in the box though. Controlling the charging and the returning of values works like a charm. 

This issue (which wasn't really one) could therefore be closed. 

If you need me to collect some data for the inclusion into your documentation please tell me - I will do so immediately.

[image]
[image]

closed 2022-12-21 by linuxstony