lbbrhzn/ocpp #738: NewMotion unable to connect

alexoman2030, 2023-01-12:
Hello,

I tried to use the integration to connect on my charger. NewMotion advanced 3.0.
After the setup phase I got the error message “Failed to set up”

In the setup I only set the IP address of the charger and the port.
The following address allows me to connect locally through the internet browser to the charger.

http://192.168.0.2:12800/user/user.html

I’m using the version OCPP v0.4.36
Home assistant 2023.1.2
Supervisor 2022.12.1
Operating system 9.4

Here below the logs
——————————-
This error originated from a custom integration.

Logger: homeassistant.config_entries
Source: custom_components/ocpp/api.py:184 
Integration: Open Charge Point Protocol (OCPP) (documentation, issues) 
First occurred: 8:20:26 PM (3 occurrences) 
Last logged: 8:33:06 PM

Error setting up entry central for ocpp
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 382, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/ocpp/__init__.py", line 71, in async_setup_entry
    central_sys = await CentralSystem.create(hass, entry)
  File "/config/custom_components/ocpp/api.py", line 184, in create
    server = await websockets.server.serve(
  File "/usr/local/lib/python3.10/site-packages/websockets/legacy/server.py", line 1117, in __await_impl__
    server = await self._create_server()
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 1505, in create_server
    raise OSError(err.errno, 'error while attempting '
OSError: [Errno 99] error while attempting to bind on address ('192.168.0.2', 12800): address not available

drc38, 2023-01-12:
Websockets (as opposed to the http port) for ocpp usually is on port 9000, you'll need to refer to your charger documentation and setup if it is different.

lbbrhzn, 2023-01-12:
The charger should contact the server, not the other way around. The IP address of the charger should not be used for configuring the OCPP server. The default and recommended setting is 0.0.0.0, which is a wildcard for 'all local cards'. You should only use something different from 0.0.0.0 when your server has multiple network cards, and you only want to to make the OCPP server available on one of them. The error indicates there is no local card with the specified IP address. Which makes sense, since it is the address used by your charger. 

Please check the documentation of your charger. You should configure your charger to contact the OCPP server.

alexoman2030, 2023-01-12:
Thank you for your quick feedback. 

I tried now with the 0.0.0.0 setting and the 9000 port. 
In this case there is no error anymore. I can go to see the entities in the "Charger" but they are all "Unavailable"

In the standard manual I can't find any reference to a Websocket. Is there another way to find it ? 
https://assets.ctfassets.net/ulfvrpf1itxm/2QE3p5LidW2gC0CeKeGAQy/6f4e2639a701843a18139185c9a7b835/Installation_Instruction_-_Business_Pro___Lite_2.0.pdf

On this page, https://github.com/ShellRechargeSolutionsEU/ocpp   , there is a reference to a port 8017
 ws://localhost:8017/ocppws 1.5,1.6"
I tried it also without result. 

When you say "You should configure your charger to contact the OCPP server." , you mean there should be parameters somewhere ? Admin rights might be required to do so? 

On the Shell site, there are also API explanation but no reference to websockets
https://developer.shell.com/api-catalog

closed 2023-03-22 by github-actions[bot]