lbbrhzn/ocpp #766: Reduce notifications due to car rejecting start of the charging session

raulliive, 2023-02-06:
**Is your feature request related to a problem? Please describe.**
I'm getting a notification message every time when the automation tries to start charging, but car is rejecting due to battery has reached the target charge threshold.

I got over last night 1300 messages about it, while I'm running the automation every 20 seconds.

**Describe the solution you'd like**
I'd like to not receive notifications in such's cases or to have those notifications configurable, so that everyone can decide if they like them or not.

**Describe alternatives you've considered**
Alternative solution would be to poll the car battery charge % often enough to allow the automation to back off. Unfortunately I can't utilize it due to Hyundai Bluelink/Kia UVO API has limitation on daily polling. Plus the cars have issues with 12v battery depletion.

**Additional context**

Adding a screenshot example of the notifications:
[image]

drc38, 2023-02-06:
Why are you needing to poll the car every 20s to start charging? You should be able to use `sensor.charger_status_connector` state to start a charge when the car plugs into the charger. Also to determine whether a charge is stopped by the EV which can be used as a condition to stop further script executions.

raulliive, 2023-02-19:
Car starts to automatically charging after plugging it in. I'm polling every 20s to be able to stop the charging, in case the other requirements are not met.

The other poll is executed to start charging when the electricity price is low.

The events noise comes in the moments when the car battery reaches the charging threshold during charging time. I'm unable to get the battery % from car in real time. I can ping car only ~10 times a day.

We have in here hourly pricing for electricity where the price for every hour can differentiate vastly, in some cases the difference can be more than 10 times. 

This means that you want to have the car to charge on the cheapest time of the day.

drc38, 2023-02-19:
Have you checked whether you can disable notifications from the integration in the `configuration.yaml` like you can with history and logging?

closed 2023-04-28 by github-actions[bot]