lbbrhzn/ocpp #778: Autel maxi charge

morten-w, 2023-02-27:
I would be nice if this integration supported the Autel maxi charge.

Mabnat, 2023-03-05:
I have it mostly working on my Autel.  There are still some issues, like it seems to disconnect/reconnect every couple of minutes and I can’t change the current limit, but I can read the data from it and start and stop the charger.

I have mine turning on when the sun comes up and off when the sun goes down so that it’s always using some of my solar power.  I wanted to try to adjust the current limit so that it uses excess solar power, but I haven’t figured out how to get past the error that I see when I try to adjust the current limit.

I had to comment out some lines, but I wasn’t making notes, so I can’t really say what was the part that fixed it, but I just tried the things that other people tried for other chargers.

Mabnat, 2023-03-21:
I just saw that an update for my Autel charger was available today.

Charge Control Module: v1.18.00
Power Control Module: v1.12.00

After installing the update, I saw that the connection was more stable.  I deleted the OCPP integration and reinstalled, in hope that it would erase all of tweaking that I’d done to try to get it working.  It looks like I’ve got it back to the original code.

Now my Autel charger has been connected solidly for about an hour so far, and my Latency Pong has gone from a solid 20,000ms to actual, realistic numbers.

There are still some issues that show up in the log and I can’t change the charge current, but at least it is working more reliably.

RickDG2, 2023-04-13:
@Mabnat, how did you delete the integration and re-install it?

I had modified mine a bit, while testing some idTag-related stuff with my Autel EVSE.

Then  I updated the firmware and now it won't reconnect to the HA OCPP central system.  I can reset the OCPP configuration of the Autel back to their servers and it connects to them just fine.

I can also run an OCPP charge point emulator and see the log entries in HA, but never from the Autel.

I know I am missing something pretty basic here.  I just want to force HACS to re-download and re-install the OCPP integration.

Mabnat, 2023-04-13:
If you go to HACS -> Integrations -> OCPP, you can click on the three dots in the upper-right corner and select “Remove”.

That will let you re-install fresh.

I still couldn’t get it to do what I wanted, which was to set it up to use my excess solar power to charge.  It’s fine if you want to just monitor it in HA and turn the charging on or off.  If I can’t adjust the current limit, it’s useless for what I want.

I tried to set it up with Charge HQ, and while I could connect to their OCPP server just fine, I couldn’t adjust the power there, either.  The Charge HQ dev even set up a special test server to try a few different methods to make it work, but he couldn’t fix it either.  There’s something that’s not OCPP compliant which prevents changing the current the “right” way.

I also was talking with the Autel reps, because the charger is supposed to be able to charge on solar natively with a power meter, but it won’t work with either of the two meters that I already have.  I also sent them the error logs from Charge HQ, and they said that they’ll look at them. Maybe they will fix it, maybe they won’t.

Charge HQ said that I might try sniffing the traffic, but I don’t feel motivated enough to mess with it anymore.

I ended up finding a really cheap, brand-new Wallbox Pulsar Plus 48A on eBay that someone broke an inductor off of the PCB while they were trying to hook up the power wires.  It cost me $1.30 for a new inductor, so I repaired it and it works great.  It works perfectly with HA OCPP and Charge HQ without needing to change anything.

Ultimately, this is an Autel firmware problem, so it’s up to them to fix it.

zhangela, 2023-05-12:
For the Autel Maxicharge owners in this thread, are you all able to get vehicle state of charge (SoC) information from MetricValues in OCPP? How about some form of vehicle identification (e.g. id tag, vin, make, model, year, etc)?

Mabnat, 2023-05-13:
No, you really can’t get that information on a residential-type Level 2 charger.  There isn’t much information communicated between the EVSE and the car beyond charging states and current limits.

zhangela, 2023-05-15:
That's really good to know! Is that a technical limitation (e.g. related to the spec) or just something that chargers don't implement because it's typically not needed?

lbbrhzn, 2023-07-20:
closing issue, sice it appears to be solved

closed 2023-07-20 by lbbrhzn

MeirSD, 2023-12-12:
I just bought this charger and after speaking with support, they no longer allow OCCP for residential installations. They haven't updated their documentation to reflect that but I thought that might be useful information to have here as I looked here before purchasing.