lbbrhzn/ocpp #786: Connection to Blink HQ 200 closed by server with `No handler for TransactionStarted registered`

njurgens, 2023-03-05:
## Version of the custom_component

`v0.4.37`

## Configuration

```yaml
2023-03-04 18:34:43.408 INFO (MainThread) [custom_components.ocpp] config = {}
```

## Describe the bug

I have a Blink HQ 200 configured in Home Assistant / OCPP.  The server continually closes the websocket connection with the error `No handler for TransactionStarted registered.`.  From what I understand, `TransactionStarted` is a standard message for when the EVSE wants to start a charging session.  

Is `TransactionStarted` not implemented or is there possibly some issue in my setup?

## Debug log

```text
2023-03-04 18:35:46.072 INFO (MainThread) [ocpp] Blink_EVSE_01: receive message [2,"7209cba2-b89a-45ec-bdcf-c89e9df8f416","TransactionStarted",{"connectorId":1,"transactionId":0,"idTag":{"id":"Engineer.Charge.Card"},"timestamp":"2023-02-25T20:46:17.425767854Z","values":[{"timestamp":"2023-02-25T20:46:17.425910435Z","value":"1439008.0","context":"Transaction.Begin","measurand":"Energy.Active.Import.Session","unit":"Wh"},{"timestamp":"2023-02-25T20:46:17.427054246Z","value":"0.0","context":"Transaction.Begin","measurand":"Current.Import","unit":"Amp"},{"timestamp":"2023-02-25T20:46:17.427103995Z","value":"241.9","context":"Transaction.Begin","measurand":"Voltage","unit":"Volt"},{"timestamp":"2023-02-25T20:46:17.427151744Z","value":"19.0","context":"Transaction.Begin","measurand":"Temperature","unit":"Celsius"},{"timestamp":"2023-02-25T20:46:17.427192910Z","value":"0.0","context":"Transaction.Begin","measurand":"Power.Peak","unit":"W"}]}]
2023-03-04 18:35:46.073 ERROR (MainThread) [ocpp] Error while handling request '<Call - unique_id=7209cba2-b89a-45ec-bdcf-c89e9df8f416, action=TransactionStarted, payload={'connectorId': 1, 'transactionId': 0, 'idTag': {'id': 'Engineer.Charge.Card'}, 'timestamp': '2023-02-25T20:46:17.425767854Z', 'values': [{'timestamp': '2023-02-25T20:46:17.425910435Z', 'value': '1439008.0', 'context': 'Transaction.Begin', 'measurand': 'Energy.Active.Import.Session', 'unit': 'Wh'}, {'timestamp': '2023-02-25T20:46:17.427054246Z', 'value': '0.0', 'context': 'Transaction.Begin', 'measurand': 'Current.Import', 'unit': 'Amp'}, {'timestamp': '2023-02-25T20:46:17.427103995Z', 'value': '241.9', 'context': 'Transaction.Begin', 'measurand': 'Voltage', 'unit': 'Volt'}, {'timestamp': '2023-02-25T20:46:17.427151744Z', 'value': '19.0', 'context': 'Transaction.Begin', 'measurand': 'Temperature', 'unit': 'Celsius'}, {'timestamp': '2023-02-25T20:46:17.427192910Z', 'value': '0.0', 'context': 'Transaction.Begin', 'measurand': 'Power.Peak', 'unit': 'W'}]}>'
File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 174, in _handle_call
File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 153, in route_message
File "/config/custom_components/ocpp/api.py", line 905, in _handle_call
File "/usr/local/lib/python3.10/site-packages/ocpp/charge_point.py", line 176, in _handle_call
ocpp.exceptions.NotSupportedError: NotSupportedError: Request Action is recognized but not supported by the receiver, {'cause': 'No handler for TransactionStarted registered.'}
2023-03-04 18:35:46.076 INFO (MainThread) [ocpp] Blink_EVSE_01: send [4,"7209cba2-b89a-45ec-bdcf-c89e9df8f416","NotSupported","Request Action is recognized but not supported by the receiver",{"cause":"No handler for TransactionStarted registered."}]
2023-03-04 18:35:46.533 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to 'Blink_EVSE_01'
2023-03-04 18:35:46.543 INFO (MainThread) [custom_components.ocpp] Charger Blink_EVSE_01 disconnected from 0.0.0.0:9000.

```

drc38, 2023-03-06:
`TransactionStarted` does not appear as part of the ocpp 1.6 specification, so it is not implemented. 
You would need to add your own code to `api.py` to handle these custom messages.

njurgens, 2023-03-06:
My brain misread `StartTransaction` as `TransactionStarted` and when I skimmed the spec :smile:.  I'll try adding the new messages to `api.py`. Thanks for the suggestion.

closed 2023-03-06 by njurgens