lbbrhzn/ocpp #827: OCPP management with 2 connectors on same wallbox

ClemKC, 2023-04-21:
Hello,
I bought a Schneider Electric charging station. The model is called EVLINK Parking (EVP2PE44R). This charging station is compatible with OCPP JSON 1.6.
I can't separate the 2 left-right connectors. The charging point allows it via OCPP to send a maximum current per side, start charging on one side but not the other, reserve on one side and make unavailable on one side but not the other.
However with this extension that you offer on Home Assistant I can't do it, is this normal?
Thank you for your feedback.
Regards

neilvangeffen, 2023-05-23:
I have a dual port one also, and can't make it work... Would love an option for this

lbbrhzn, 2023-07-20:
There is a pull request now #803 but we need someone to add a testcase for it, and to try it out in practice.
@ClemKC , @neilvangeffen, would you be able to assist here?

neilvangeffen, 2023-07-20:
I can try, yeah, I have a BCPC-D2N-P which is a dual socketed unit.

I might need someone to hold my hand on how to install the correct code though, as im pretty new to github and pull requests etc, but happy to try for sure!

neilvangeffen, 2023-07-20:
I had previously tried this version, but it had a lot of missing information for me
https://github.com/lbbrhzn/ocpp/discussions/594 

So i just download the the custom_component ocpp folder from here right? https://github.com/ernst77/ocpp/tree/main/custom_components

Anything in particular you need me to check or test?

And how do i do the following, all i want is to be able to graph the charger ports as individual items in my home assistant power graphs, and be able to do load balancing and slow down charge, but in general i want the charger to be ready to charge when someone plugs in. I wasnt able to get that working before, but not sure if thats just because the code i was using wasnt quite right, or if its not possible, or if im an idiot and its really obvious

neilvangeffen, 2023-07-21:
I installed it at home, runs good, but goes unavailable regularly... all hard wired with ethernet, so doubt its an internet thing. Also i added every sensor to see what would happen, and many are not available. all the charger ones are unknown.

Here are some screenshots:
https://photos.app.goo.gl/mi9Er3eCi4uLbebK9

When i move the slider to set max current, it doesnt do anything on the EVSE end... But DLB isnt working either, so might be an issue with the unit

closed 2023-09-27 by github-actions[bot]