lbbrhzn/ocpp #844: index out of range error

darkmordu, 2023-05-08:
My setup is latest version of ocpp plugin.
Using a openevse charger

When trying to change value of set_max_charge_rate_amps I get an error from home assitant:
ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139720090513664] list index out of range

debug traces show:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service
await hass.services.async_call(
File "/usr/src/homeassistant/homeassistant/core.py", line 1849, in async_call
task.result()
File "/usr/src/homeassistant/homeassistant/core.py", line 1889, in _execute_service
await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 226, in handle_service
await service.entity_service_call(
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 798, in entity_service_call
future.result() # pop exception if have
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 981, in async_request_call
await coro
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 838, in _handle_entity_call
await result
File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 106, in async_set_value
await entity.async_set_native_value(native_value)
File "/config/custom_components/ocpp/number.py", line 127, in async_set_native_value
resp = await self.central_system.set_max_charge_rate_amps(
File "/config/custom_components/ocpp/api.py", line 271, in set_max_charge_rate_amps
return await self.charge_points[cp_id].set_charge_rate(limit_amps=value)
File "/config/custom_components/ocpp/api.py", line 564, in set_charge_rate
resp = await self.get_configuration(
File "/config/custom_components/ocpp/api.py", line 789, in get_configuration
value = resp.configuration_key[0][om.value.value]
IndexError: list index out of range

When I look in file api.py at line 789 I notice that it checks for "configuration_key" before checking for "unknown_key"
Inverting order and managing case "None" at line 572 seems to fix the problem

imle, 2023-06-13:
Do you have this fix somewhere others can use?

darkmordu, 2023-06-16:
Here is my fix I did in my install of ha
I changed 2 functions :
- get_configuration: to manage case unknown_key before and avoid calling resp.configuration_key[0][om.value.value] which generates the error. I also removed the `await self.notify_ha` to avoid zounds of notifications in ha
- set_charge_rate to manage the case "None"
```
    async def get_configuration(self, key: str = ""):
        """Get Configuration of charger for supported keys else return None."""
        if key == "":
            req = call.GetConfigurationPayload()
        else:
            req = call.GetConfigurationPayload(key=[key])
        resp = await self.call(req)
        if resp.unknown_key is not None:
            _LOGGER.warning("Get Configuration returned unknown key for: %s", key)
            return None
        if resp.configuration_key is not None:
            value = resp.configuration_key[0][om.value.value]
            _LOGGER.debug("Get Configuration for %s: %s", key, value)
            self._metrics[cdet.config_response.value].value = datetime.now(
                tz=timezone.utc
            )
            self._metrics[cdet.config_response.value].extra_attr = {key: value}
            return value
```

```
    async def set_charge_rate(self, limit_amps: int = 32, limit_watts: int = 22000):
        """Set a charging profile with defined limit."""
        if prof.SMART in self._attr_supported_features:
            resp = await self.get_configuration(
                ckey.charging_schedule_allowed_charging_rate_unit.value
            )
            _LOGGER.info(
                "Charger supports setting the following units: %s",
                resp,
            )
            _LOGGER.info("If more than one unit supported default unit is Amps")
            if resp is None:
                lim = limit_amps
                units = ChargingRateUnitType.amps.value
            elif om.current.value in resp:
                lim = limit_amps
                units = ChargingRateUnitType.amps.value
            else:
                lim = limit_watts
                units = ChargingRateUnitType.watts.value
            resp = await self.get_configuration(
                ckey.charge_profile_max_stack_level.value
            )
           ...
```

I hope these changes will make their way to the repository so that I don't need to change it every time I update.

regards

lbbrhzn, 2023-07-20:
Please create a piull request with the changes you propose, so we can review and test. Thanks!

closed 2023-10-13 by github-actions[bot]