lbbrhzn/ocpp #857: StarCharge Aurora 7 mostly working just not SoC sensor

andlo, 2023-06-23:
## Version of the custom_component

V 0.4.39

## Describe the bug

It seems that this charger works with the OCPP which I expected it to as the product description claims it uses OCPP 1.6j 

It connects fine and I can start and stop charging and get some data from it.

But it dosnt report SoC and i dont know if it is because the car isnt reporting it or because the implementation of OCPP in the product dosnt use that data. 

How can I debug that if it is posible ?

lbbrhzn, 2023-06-24:
Although the SoC measurand is indeed part of the OCPP 1.6j standard, and supported by this plugin, there are currently not many chargers that report it. In order for it to work, both the Charger and the EV need to support a a Vehicle to Grid protocol, such as [ISO15118](https://en.wikipedia.org/wiki/ISO_15118).

andlo, 2023-06-26:
Thanks. Great reply.

closed 2023-06-26 by andlo