lbbrhzn/ocpp #859: ZJBeny BCP-A2N OCPP EVSE: Update Firmware

GuzziDoug, 2023-06-28:
### Discussed in https://github.com/lbbrhzn/ocpp/discussions/858

<div type='discussions-op-text'>

<sup>Originally posted by **GuzziDoug** June 28, 2023</sup>
Hi,
 searched but no answers... How do I update the firmware for my Beny? (Must be done from the OCPP Server, HA in this case)

 fyi, for Beny users, Firmware BCP-A2N-P_SW1_0_26_HW1.bin is recently available.
 The readme attached says: 
Please send the URL ( via "update Firmware.req")  to OCPP EV Charger for updating the firmware.

(Do not use other HTTP server to update, maybe incompatible.):

Household EV charger, Signel phase with cable
http://106.15.78.131/BCP-A2N-P_SW1_0_25_HW1.bin
etc... (end)

so I need to point the server at that address, but unsure how to go about it.

I feel this will be an issue for other EVSEs too, because all the Firmwares will require updating at some point. 
Perhaps a howto may be required? (I last performed an update with Steve, but my Steve server is no longer...

regards, Doug
</div>

GuzziDoug, 2023-07-09:
The issue raised is SOLVED. However, I think there is a need for more documentation. I am happy to assist with this, but as a proof reader or minor editor. The areas I had difficulty with were because the charger occasionally lost contact with HA OCPP (May have been early firmware issue on charger) which meant the links were not there when I looked for them: resolved by Charger re-boot by Power cycle.

lbbrhzn, 2023-07-20:
Great. I'll close the issue. Feel free to add a pull request for updating the documentation.

closed 2023-07-20 by lbbrhzn