lbbrhzn/ocpp #871: HA 2023.8.0b0 causes OCPP addin to fail

dreimer1986, 2023-07-26:
**Describe the bug**
Addin fails to load at all

**To Reproduce**
Steps to reproduce the behavior:
- Just update to the new Beta version of HA and try to use the addin

**Expected behavior**
- It should work I guess ^^ 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Windows 11
 - Browser [e.g. chrome, safari] Chrome
 - Version [e.g. 22] 117

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6] Pixel 7
 - OS: [e.g. iOS8.1] Android 14 Beta
 - Browser [e.g. stock browser, safari] Chrome Beta
 - Version [e.g. 22] ?

**Additional context**
Add any other context about the problem here.
```
Logger: homeassistant.config_entries
Source: helpers/device_registry.py:162
First occurred: 26. Juli 2023 um 22:04:57 (1 occurrences)
Last logged: 26. Juli 2023 um 22:04:57

Error setting up entry central for ocpp
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 388, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/__init__.py", line 84, in async_setup_entry
    dr.async_get_or_create(
  File "/usr/src/homeassistant/homeassistant/helpers/device_registry.py", line 500, in async_get_or_create
    device_info_type = _validate_device_info(config_entry, device_info)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/device_registry.py", line 162, in _validate_device_info
    raise DeviceInfoError(
homeassistant.helpers.device_registry.DeviceInfoError: Invalid device info {'default_model': 'OCPP Charge Point', 'identifiers': {('ocpp', 'evbox')}, 'name': 'evbox', 'via_device': ('ocpp', 'e3e34b48b1f328e6666c85df3085125d')} for 'ocpp' config entry: device info needs to either describe a device, link to existing device or provide extra information.
```

drc38, 2023-07-27:
Try changing `model=boot_info.get(om.charge_point_model.name, None),` to `model=boot_info.get(om.charge_point_model.name, "Unknown"),`

drc38, 2023-07-27:
Looks like issue is Hass adding `default_model` to device registry entry which causes the error. Hopefully they'll fix in their beta testing.

dreimer1986, 2023-07-28:
That now looks good indeed. Everything back online.
[image]
[image]
So this fixes it! Thanks for the quick help.