lbbrhzn/ocpp #877: Problem with setting charge limit

mrVoltrix, 2023-08-06:
Hi all, 

I'm using this integration already over a year, everything was working fine but now suddenly I've got a issue with setting te charge limit.
from the integration I got all the time this info back: "Ocpp integration
Warning: Set charging profile failed with response Rejected"

From my charger (Schneider smart wallbox) I got this logs back:

Sun Aug 6 08:32:00 2023 | Sent GetConfigurationConf configurationKeys: 1
Sun Aug 6 08:32:00 2023 | Received SetChargingProfile(connectorId:0 profileId:8 stackLevel:999 ChargePointMaxProfile Relative)
Sun Aug 6 08:32:00 2023 | Sent SetChargingProfileConf: Rejected (Load balancer is disabled)
Sun Aug 6 08:32:00 2023 | Received SetChargingProfile(connectorId:0 profileId:8 stackLevel:998 TxDefaultProfile Relative)
Sun Aug 6 08:32:00 2023 | Sent SetChargingProfileConf: Rejected (EM controlled by external system is disabled)

It's strange because I didn't change anything in my setup

lbbrhzn, 2023-08-06:
Which plug-in release are you using?

Note that this functionality has not been changed in the last few releases. It might be related to a firmware update on your charger.

mrVoltrix, 2023-08-06:
I'm using the V0.4.40 and my chargers firmware was also not updated, it's really strange. Before I had never problems with that.

mrVoltrix, 2023-08-07:
I complety resetted my charger and put the same configuration back as before and everything is working again. Still don't know what the problem was but its solved so this ticket can be closed.

lbbrhzn, 2023-08-07:
Glad to hear you were able to solve the issue.