lbbrhzn/ocpp #878: Error message "NotSupported while setting MeterValuesSampledData to ..." for phase-specific charger MeterValues

robinsmidsrod, 2023-08-12:
**Describe the bug**

Every time I start opp Home Assistant I'm getting this message:

```
INFO (MainThread) [ocpp] 2210070698M: send [2,"af96390d-fbfa-4792-8505-4e7934ad3f64","GetConfiguration",{"key":["MeterValuesSampledData"]}]
INFO (MainThread) [ocpp] 2210070698M: receive message [3, "af96390d-fbfa-4792-8505-4e7934ad3f64", {"configurationKey":[{"key":"MeterValuesSampledData","readonly":false,"value":"Current.Import.L1,Current.Import.L2,Current.Import.L3,Voltage.L1,Voltage.L2,Voltage.L3,Energy.Active.Import.Register"}]}]
INFO (MainThread) [ocpp] 2210070698M: send [2,"26ccee4c-6aac-4ce7-a808-108553c41ea8","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register,Energy.Active.Import.Interval,Power.Active.Import,Power.Offered,Power.Factor,Current.Import,Current.Offered,Voltage,Frequency,RPM,SoC,Temperature,Current.Export,Energy.Active.Export.Register,Energy.Active.Export.Interval,Power.Active.Export"}]
INFO (MainThread) [ocpp] 2210070698M: receive message [3, "26ccee4c-6aac-4ce7-a808-108553c41ea8", {"status":"NotSupported"}]
WARNING (MainThread) [custom_components.ocpp] NotSupported while setting MeterValuesSampledData to Energy.Active.Import.Register,Energy.Active.Import.Interval,Power.Active.Import,Power.Offered,Power.Factor,Current.Import,Current.Offered,Voltage,Frequency,RPM,SoC,Temperature,Current.Export,Energy.Active.Export.Register,Energy.Active.Export.Interval,Power.Active.Export
```

To me it seems like the `Current.Import.L1` and similar MeterValues (per phase values) sent by the charger is not properly detected by this integration.

**To Reproduce**

1. Configure a ChargeAmps Halo charger (OCPP version with a single charging port, 1 phase 32A 230V) and set its OCPP WebSocket endpoint to this integration
2. Configure integration with all checkmarks (to figure out what it supports)
3. Get this error message

**Expected behavior**

Would've expected all unsupported features of the charger to not be enabled and work with what was supported.

Alternatively, inform me which features I asked for that are not supported, so I can manually disable them.

This could probably be done by trying to enable one feature at a time and see which ones fail.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Firefox
 - Version 116.0.2

**Smartphone (please complete the following information):**
 - Device: Xiaomi Mi 12 Pro
 - OS: Android 13
 - Browser Firefox
 - Version 116.2.0

**Additional context**

In addition (which will probably go away if first problem is fixed) I'm getting a persistent notification added several times during the day: `Warning: Start transaction failed with response Rejected`

[image]

This is really annoying and should probably be added as an event entity (now that we have those). If I don't open my Home Assistant web interface for some days I can easily have 30-40 persistent notifications of that kind. You can see that issue in the attached log file as well. If you'd like me to report this as a separate issue I can do that, just let me know.

Complete log file for several days of running the integration: [ocpp-home-assistant_2023-08-12T11-05-45.362Z.log](https://github.com/lbbrhzn/ocpp/files/12327226/ocpp-home-assistant_2023-08-12T11-05-45.362Z.log)

dreimer1986, 2023-08-19:
Well the Rejected shows up because you ask for MeterValuesSampledData that you box does not support. You have to do it vice versa. Remove all checkboxes and then one by one add values that you think it should be able to provide. OR maybe you have a way to get these information from a OCPP Debug log file. That's what I did for my EVBox Elvi.

This is part of my Log and should be values your's should be able to provide, too.
```
MeterValuesSampledData false Current.Import,Current.Offered,Energy.Active.Import.Register,Power.Active.Import,Temperature,Voltage,Frequency
```

robinsmidsrod, 2023-08-20:
@dreimer1986 How can I generate this `OCPP Debug log file` you're talking about? Is it in the information I attached above, or do I need to use some other tool? It is just so incredibly cumbersome, because this integration is missing the *configure* button next to the config entry, so you have to remove the entire integration, type in all the information again to connect to the device, and remember where you previously where in the list of checkboxes and then click the next one. There must be a better/faster way to do this?

dreimer1986, 2023-08-20:
You look for service: ocpp.get_diagnostics
https://github.com/lbbrhzn/ocpp/issues/221#issuecomment-988765666 here and https://github.com/lbbrhzn/ocpp/issues/221#issuecomment-1035989737 here you can see what I did to get my information.

As almost NO manufacturer cares about OCPP compliance athough they claim to be, you have two ways. Trial and error or maybe get the needed information from the logs. I did https://github.com/lbbrhzn/ocpp/issues/221#issuecomment-1044339108

robinsmidsrod, 2023-08-20:
@dreimer1986 After much trial and error I've figured out which meterValues are supported by the ChargeAmps Halo EV charger (firmware version 169). 

Here's the list:

MeterValues accepted:

Energy.Active.Import.Register
Power.Active.Import
Current.Import
Current.Offered
Voltage
Frequency
Temperature

MeterValues not supported:

Energy.Reactive.Import.Register
Energy.Active.Import.Interval
Energy.Reactive.Import.Interval
Power.Reactive.Import
Power.Offered
Power.Factor
RPM
SoC
Current.Export
Energy.Active.Export.Register
Energy.Reactive.Export.Register
Energy.Active.Export.Interval
Energy.Reactive.Export.Interval
Power.Active.Export
Power.Reactive.Export

After some testing I've also noticed that some other things are also wonkey.

- `UnlockConnector` is not supported
- `ChargePointMaxProfile` is not supported by this charger
- `TxDefaultProfile` is rejected (wrong parameters?)
- `"ChangeAvailability",{"connectorId":0,"type":"Operative"}` is accepted, but doesn't change switch in UI (is always shown as off)

This is a charger with a single connector. Are the wrong parameters sent with these commands, or what's the problem? I'd really love to be able to change the charging current between 6A and 32A, which is possible in the web-interface of the charger. How would I go about troubleshooting these problems?

dreimer1986, 2023-08-21:
Let me add something to your list. Status "Unavailable" or "Unknown" of an entity does NOT mean it's not supported. You need to check for the NotSupported reply in your logs. If you did so, wonderful... Looks like your charger is a even bigger mess than mine. If you wanna see how we debugged mine, check the issue report I linked to already. It's from unuseable up to all systems go, so there you can see how to enable debugging properly and then what to send in your report to make things fixable.

Regarding your "Are the wrong parameters sent with these commands, or what's the problem?": In 99% of the cases the problem is the manufacturr being too ... ya know ... to support a standard that is clearly set in stone and not that difficult to understand either. My charger for example is OCPP certified and still did things that clearly are NON standard conformant. That needed some special fixing in the end.

Ah and before I forget. What did you do to "Skip OCPP schema validation" and "Force Smart Charging Profile" settings?

robinsmidsrod, 2023-08-21:
@dreimer1986 Thanks a bunch for helping me out trying to understand all of this. Really appreciated!

I'm fully aware that *NotSupported* and *Unknown* are completely different things, and I pay attention to report the right thing whenever I see it.

How do you go about fixing things when the charger doesn't adhere to spec? Do you just have to go straight into the source code, or are there any settings (file) you can tweak here and there?

I didn't check the options for either `Skip OCPP schema validation` or `Force Smart Charging Profile`. I tried to enable them during my trial and error phase, and it seemed like it didn't have any impact on the behavior I reported earlier.

My charger has a single connector, but I'm a bit puzzled by some things I see in the log, for some commands I see `connectorId:0` and some I see `connectorId:1` (always the same number for the same command). Is it possible there are some off-by-one errors either in this OCPP server, or in the charger firmware? What is the correct behavior spec-wise when it comes to connectorIds? Does all commands start at 0 or 1, or does it vary with command?

Finally, I noticed earlier you mentioned the service calls (ocpp.get_diagnostics, etc.), but when I go into Home Assistant Developer Tools to try them out, the *call service* button is greyed out. I don't see a variable to specify the target/entity_id. Where are these service calls documented? I couldn't find them over at the documentation site.

dreimer1986, 2023-08-21:
The close specifics of the Standard can all be found here: https://www.oasis-open.org/committees/download.php/58944/ocpp-1.6.pdf It's quite a read, but just look for the parameters you wanna know more of and the PDF tells you what you wanna know. SO for example it tells you that ID:0 means the setting is for the whole thing globally and the real connectors must start with 1. (Page 27)

In my case I had not much clue about all the stuff and especially was not the Python coder at all. I just threw logs at the devs and the found out what to do in the end.

Regarding the disgnostics service. Switch to YAML-Mode down left and then you see what you can set up. I set a anonymous FTP for it as upload location. Can be found in my Odyssey, too ^^

robinsmidsrod, 2023-08-21:
@dreimer1986 I finally managed to get diagnostics from the charger!

It seems like the charger states that it has two connectors, one 32A and the other 10A. This is because the charger is also sold with a version that has a Schuko connector (in addition to the Type 2 connector). In my specific case the Schuko connector is not installed, but the firmware still shows it as a connector, even though the bootup flags says that there is no Schuko on this particular model. This is probably why I get some confusion.

Is it possible to specify which connector should be used, and just ignore the second non-functional one?

[halo_2210070698M_20230821_113352.log](https://github.com/lbbrhzn/ocpp/files/12395583/halo_2210070698M_20230821_113352.log)

dreimer1986, 2023-08-21:
Tbh... no clue if multiple connectors are supported right now... @lbbrhzn @drc38 any clue?

drc38, 2023-08-21:
> Tbh... no clue if multiple connectors are supported right now... @lbbrhzn @drc38 any clue?

You can try #803

robinsmidsrod, 2023-10-04:
Since this topic ended up without any solution I decided to try my luck with the Node-RED OCPP custom module, which took some time, but it's now working as I need it to. If you feel this issue is no longer relevant for anyone else, you may close it. It's not vital for me anymore.

closed 2023-10-06 by drc38