lbbrhzn/ocpp #879: Option to clear charging profiles

ghulleman, 2023-08-14:
**Is your feature request related to a problem? Please describe.**
I was unable to charge my EV as the charger stated that the external station only allowed charging at 0A. I was unable to set the maximum current via the OCPP HA addon. this was stuck at 0A. When I set this to 8A or 16A, after a refresh the max was 0A again. I did look at the log (enabled debugging log), but could not find anything related to this. Since the priority was to charge my car, I resolved this using the alternative as described below. 

**Describe the solution you'd like**
I would like the option to send the OCPP 1.6 'Clear charging profiles' to clear the profiles in the charger. 

**Describe alternatives you've considered**
To solve the issue and allow the charge to charge, I used SteVe (see https://github.com/steve-community/steve) to link the charger to this backend, send the clear command and now the carger works again. 

**Additional context**

drc38, 2023-08-16:
There is a service for this already, goto developer tools | services to use it.

closed 2023-08-16 by drc38

ghulleman, 2023-08-16:
Hi drc38. You are correct. I did not find this option as this was not documented (at least not that I could find) and not available to me. 

For anyone finding this topic; there is a service "ocpp.clear_profile" but this service is only available if you chose  “Force Smart Charging feature profile" during the setup.