lbbrhzn/ocpp #880: Boot Loop ABB Terra AC

thomas6779, 2023-08-14:
After adding the ocpp to my home assistant the connection started und I see the entitys for several seconds. Some seconds later the charger reboots again ad again.
-->

## Version of the cust

0.4.40
-->

## Log
[home-assistant_ocpp_2023-08-14T14-30-42.294Z.log](https://github.com/lbbrhzn/ocpp/files/12335941/home-assistant_ocpp_2023-08-14T14-30-42.294Z.log)

Any ideas?

drc38, 2023-08-31:
See the potential fix posted in #684

MinecraftiUndCo, 2023-09-15:
Also having this issue. Could this be resolved in an official release?

jonaswikstrom, 2023-09-26:
Is still this an issue? Planning to integrate my Terra AC

villekui, 2023-09-27:
I got it fixed by commenting a few lines in **/config/custom_components/ocpp/api.py** as described [here](https://github.com/lbbrhzn/ocpp/issues/442#issuecomment-1237651231)

tessierdsl, 2023-11-09:
I got out of the boot loop with this fix. But I can't get any measurements from it. It would be nice if someone posted their working settings.

villekui, 2023-11-10:
> I got it fixed by commenting a few lines in **/config/custom_components/ocpp/api.py** as described [here](https://github.com/lbbrhzn/ocpp/issues/442#issuecomment-1237651231)

It looks like Terra AC firmware 1.6.9 has been fixed lots of compatibility issues with the charger and e.g. this fix is not needed anymore.

drc38, 2023-11-10:
Also try the latest release to auto-detect the supported measurands. You'll need to uninstall   then reinstall the integration once downloaded

woomiez, 2023-11-20:
My TerraAC never seemed to have that bug, but a few others.. like only accepting current values between 6-10. Anything below 6 will lead to charging being paused, and anything above 10 just gets ignored. So, as long as I use OCPP, I can only charge at 7kW max, while with manual control via Bluetooth app I have the full 16A configuration range available. But that looks like a bug in the Terra, not the integration.

lipigab, 2024-01-24:
Hi, I'm running the latest Terra AC firmware and OCPP but my charger keeps rebooting randomly when I change the charging current for example.
So it looks like FW 1.6.9 has not fixed the issue.
Does anyone have a solid solution for this?

closed 2024-04-02 by github-actions[bot]