lbbrhzn/ocpp #882: Log level being ignored

ExcelsiorEyer, 2023-08-17:
Log level configuration is being ignored

Configuration.yaml
```
logger:
  logs:
    custom_components.ocpp: warn

```

**Expected behavior**
Log level setting to be respected.

Sample output from logs
```
2023-08-17 07:09:08.038 INFO (MainThread) [ocpp] Wallbox: send [3,"1467963981",{}]
2023-08-17 07:09:08.304 INFO (MainThread) [ocpp] Wallbox: receive message [2,"1814887560","MeterValues",{"meterValue": [{"timestamp": "2023-08-17T12:09:07Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "2702492.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "27712"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "40"} ]} ],"connectorId": 1}]
2023-08-17 07:09:08.306 INFO (MainThread) [ocpp] Wallbox: send [3,"1814887560",{}]
2023-08-17 07:10:08.346 INFO (MainThread) [ocpp] Wallbox: receive message [2,"1290127955","MeterValues",{"meterValue": [{"timestamp": "2023-08-17T12:10:07Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "2702492.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "27712"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "40"} ]} ],"connectorId": 1}]
2023-08-17 07:10:08.347 INFO (MainThread) [ocpp] Wallbox: send [3,"1290127955",{}]
2023-08-17 07:11:08.419 INFO (MainThread) [ocpp] Wallbox: receive message [2,"1434433518","MeterValues",{"meterValue": [{"timestamp": "2023-08-17T12:11:07Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "2702492.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "27712"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "40"} ]} ],"connectorId": 1}]
2023-08-17 07:11:08.421 INFO (MainThread) [ocpp] Wallbox: send [3,"1434433518",{}]
2023-08-17 07:12:08.468 INFO (MainThread) [ocpp] Wallbox: receive message [2,"1821066342","MeterValues",{"meterValue": [{"timestamp": "2023-08-17T12:12:07Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "2702492.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "27712"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "40"} ]} ],"connectorId": 1}]
2023-08-17 07:12:08.469 INFO (MainThread) [ocpp] Wallbox: send [3,"1821066342",{}]
2023-08-17 07:13:08.501 INFO (MainThread) [ocpp] Wallbox: receive message [2,"550245196","MeterValues",{"meterValue": [{"timestamp": "2023-08-17T12:13:07Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "2702492.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "27712"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "40"} ]} ],"connectorId": 1}]
2023-08-17 07:13:08.502 INFO (MainThread) [ocpp] Wallbox: send [3,"550245196",{}]
2023-08-17 07:14:08.562 INFO (MainThread) [ocpp] Wallbox: receive message [2,"1104627321","MeterValues",{"meterValue": [{"timestamp": "2023-08-17T12:14:07Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "2702492.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "27712"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "40"} ]} ],"connectorId": 1}]
2023-08-17 07:14:08.564 INFO (MainThread) [ocpp] Wallbox: send [3,"1104627321",{}]
2023-08-17 07:15:00.844 INFO (MainThread) [ocpp] Wallbox: receive message [2,"1312994984","MeterValues",{"meterValue": [{"timestamp": "2023-08-17T12:00:00Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Clock","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "2702492.0"} ]} ],"connectorId": 0}]
2023-08-17 07:15:00.844 INFO (MainThread) [ocpp] Wallbox: send [3,"1312994984",{}]
2023-08-17 07:15:00.853 INFO (MainThread) [ocpp] Wallbox: receive message [2,"1763794427","MeterValues",{"meterValue": [{"timestamp": "2023-08-17T12:00:00Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Clock","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "2702492.0"} ]} ],"connectorId": 1}]
2023-08-17 07:15:00.853 INFO (MainThread) [ocpp] Wallbox: send [3,"1763794427",{}]
[... cut ...] ": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "40"} ]} ],"connectorId": 1}]
2023-08-17 07:15:08.641 INFO (MainThread) [ocpp] Wallbox: send [3,"1582152040",{}]

```

Version: v0.4.40

Home Assistant 2023.8.2
Supervisor 2023.08.1
Operating System 10.4
Frontend 20230802.0 - latest

drc38, 2023-08-17:
The messages received are coming from the `ocpp` package dependency. So if you add another line in your yaml they will respect the log level specified.

ExcelsiorEyer, 2023-08-18:
Thanks. The default log level was not respected by the dependency. Had to explicitly set it to WARN and it is no longer showing the INFO level logs.

closed 2023-08-18 by ExcelsiorEyer