lbbrhzn/ocpp #890: Error setting up entry central for ocpp

stigkildegaard, 2023-08-23:
After installing the latest updates for HA, the OCPP integration does not work for me anymore. At every reboot of home assistant, the first entry in the log file is:
Error setting up entry central for ocpp
19:15:19 – (ERROR) helpers/device_registry.py

The contents of the log entry are:
Log Details (ERROR)
Logger: homeassistant.config_entries
Source: helpers/device_registry.py:165
First occurred: 19:15:19 (1 occurrences)
Last logged: 19:15:19

Error setting up entry central for ocpp
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 388, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/__init__.py", line 84, in async_setup_entry
    dr.async_get_or_create(
  File "/usr/src/homeassistant/homeassistant/helpers/device_registry.py", line 511, in async_get_or_create
    device_info_type = _validate_device_info(config_entry, device_info)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/device_registry.py", line 165, in _validate_device_info
    raise DeviceInfoError(
homeassistant.helpers.device_registry.DeviceInfoError: Invalid device info {'default_model': 'OCPP Charge Point', 'identifiers': {('ocpp', 'charger')}, 'name': 'charger', 'via_device': ('ocpp', '9fe9db36dcb0a70c1d954336ca24e288')} for 'ocpp' config entry: device info needs to either describe a device, link to existing device or provide extra information.

I have used the integration happily for months and have made no changes to my setup at all in more than a month.

The charger, an ABB Terra, indicates that it connects to the LAN correctly when rebooted. The rest of my HA setup also works without any issues.

I am running the latest version HA OS on proxmox.

Please let me know if there is any additional information I can provide, which may be helpful for solving the problem.

Thanks in advance!

lbbrhzn, 2023-08-24:
Which home assistant / ocpp version are you using? 
I'm at 2023.8.3 and occp v0.4.40, and I do not have this issue.

janrothkegel, 2023-08-27:
I am running HA 2023.8.4 and faced the same issue. After updating to the latest version of OCPP the issue seems to be gone (but I was not yet able to verify whether everything else in this version works well).

stigkildegaard, 2023-08-27:
Hi and thank you for your responses.

After updating OCPP v0.4.40 the issue is solved. I had missed that there was a new version of OCPP and had only updated home assistant. Sorry.

Thanks again for your assistance.

stigkildegaard, 2023-08-27:
Issue solved in OCPP v0.4.40.

closed 2023-08-27 by stigkildegaard

lbbrhzn, 2023-08-27:
no problem, glad to see your issue is solved..