lbbrhzn/ocpp #899: Grizzl-e Smart v2 support

mkbrown, 2023-09-08:
**Describe the bug**

Grizzl-e Smart v2 does not work with the OCPP integration version 0.4.40

Firmware WiFi Module: GWM-07.013-03

**To Reproduce**

Upon configuring the Grizzl-e, it connects and has the following errors in the HA logs:

```

This error originated from a custom integration.

Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:212 
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: September 7, 2023 at 5:42:35 PM (9 occurrences) 
Last logged: 9:29:52 AM

Skipping websocket subprotocol validation

This error originated from a custom integration.

Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:865 
Integration: Open Charge Point Protocol (OCPP) (documentation, issues) 
First occurred: September 7, 2023 at 5:42:46 PM (8 occurrences) 
Last logged: 9:29:54 AM

Unexpected exception in connection to ‘GRS-170000xxxx’: ‘(“Error code ‘%s’ is not defined by the OCPP specification”, ‘FormationViolation’)’
Traceback (most recent call last):
  File “/config/custom_components/ocpp/api.py”, line 930, in run
    await asyncio.gather(*self.tasks)
  File “/config/custom_components/ocpp/api.py”, line 421, in post_connect
    await self.get_supported_features()
  File “/config/custom_components/ocpp/api.py”, line 495, in get_supported_features
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File “/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py”, line 281, in call
    response = await self._get_specific_response(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File “/config/custom_components/ocpp/api.py”, line 865, in _get_specific_response
    raise resp.to_exception()
          ^^^^^^^^^^^^^^^^^^^
  File “/usr/local/lib/python3.11/site-packages/ocpp/messages.py”, line 439, in to_exception
    raise UnknownCallErrorCodeError(
ocpp.exceptions.UnknownCallErrorCodeError: (“Error code ‘%s’ is not defined by the OCPP specification”, ‘FormationViolation’)

This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/ocpp/api.py:495 
Integration: Open Charge Point Protocol (OCPP) (documentation, issues) 
First occurred: September 7, 2023 at 5:43:16 PM (7 occurrences) 
Last logged: 9:30:24 AM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File “/usr/local/lib/python3.11/asyncio/tasks.py”, line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File “/usr/local/lib/python3.11/asyncio/queues.py”, line 158, in get
    await getter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File “/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py”, line 281, in call
[... 83 lines cut ...]
2023-09-08 15:34:07.813 DEBUG (MainThread) [websockets.server] < TEXT '[4, "aedabe09-d6ed-4f95-9dc2-e9186be0a005", "FormationViolation", ""]' [69 bytes]
2023-09-08 15:34:07.813 DEBUG (MainThread) [websockets.server] < CLOSE 1005 (no status code [internal]) [0 bytes]
2023-09-08 15:34:07.813 DEBUG (MainThread) [websockets.server] x half-closing TCP connection
2023-09-08 15:34:07.813 INFO (MainThread) [ocpp] GRS-170000xxxxx: receive message [4, "aedabe09-d6ed-4f95-9dc2-e9186be0a005", "FormationViolation", ""]
2023-09-08 15:34:07.814 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/config/custom_components/ocpp/api.py", line 421, in post_connect
await self.get_supported_features()
File "/config/custom_components/ocpp/api.py", line 495, in get_supported_features
resp = await self.call(req)
^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 281, in call
response = await self._get_specific_response(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/ocpp/api.py", line 865, in _get_specific_response
raise resp.to_exception()
^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/ocpp/messages.py", line 439, in to_exception
raise UnknownCallErrorCodeError(
ocpp.exceptions.UnknownCallErrorCodeError: ("Error code '%s' is not defined by the OCPP specification", 'FormationViolation')
2023-09-08 15:34:07.832 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2023-09-08 15:34:07.833 INFO (MainThread) [custom_components.ocpp] Charger GRS-170000xxxxx disconnected from 172.20.21.150:9000.
2023-09-08 15:34:07.834 INFO (MainThread) [websockets.server] connection closed

```

mkbrown, 2023-09-08:
[image]
[image]

mkbrown, 2023-09-08:
[image]

8jmm3, 2023-09-09:
I'm not an expert in deciphering these logs but I'm curious to better understand the issue.  I don't have a v2 unit (yet).

The log does indicate "Skipping websocket subprotocol validation" which seems to be required for all versions of the firmware.  It appears that the OCPP integration is attempting to query the feature profile.  For some reason the EVSE is responding that is doesn't like the format of the request (FormationViolation).  I recall reading that in at least some version of the firmware they only support the Core profile.  I don't know about the force profile setting but could that be causing a problem?

The description for FormationViolation violation: Payload for Action is syntactically incorrect or not conform the PDU structure for Action. This is only valid for OCPP 1.6.

Is there a second request made that returns garbage (EVSE in a bad state?), or is the OCPP integration just not handling the FormationViolation error gracefully?

Are you able to capture the contents of the 98 byte websocket request (or even the entire transaction) to compare against the OCPP specification or some other working example?

You could try hacking the integration, perhaps by modifying the get_supported_features() function in api.py.

I could try to capture the exchange on my v1 unit to compare.

8jmm3, 2023-09-09:
Excuse the crude formatting but to follow up my v1 unit performs the following initial exchange (captured with tcpdump -v -X dst grizzle or src grizzle):

```
16:12:59.761679 IP (tos 0x0, ttl 64, id 25330, offset 0, flags [DF], proto TCP (6), length 140)
    hass.9000 > grizzle.60988: Flags [P.], cksum 0x85e0 (incorrect -> 0x2412), seq 206:306, ack 239, win 64002, length 100
	0x0000:  4500 008c 62f2 4000 4006 5218 c0a8 0006  E...b.@.@.R.....
	0x0010:  c0a8 040b 2328 ee3c 7c2d 13ca a6a7 9192  ....#(.<|-......
	0x0020:  5018 fa02 85e0 0000 8162 5b32 2c22 3730  P........b[2,"70
	0x0030:  6633 6632 3437 2d63 3333 362d 3439 6563  f3f247-c336-49ec
	0x0040:  2d38 6139 632d 6262 3834 6632 3662 3839  -8a9c-bb84f26b89
	0x0050:  3466 222c 2247 6574 436f 6e66 6967 7572  4f","GetConfigur
	0x0060:  6174 696f 6e22 2c7b 226b 6579 223a 5b22  ation",{"key":["
	0x0070:  5375 7070 6f72 7465 6446 6561 7475 7265  SupportedFeature
	0x0080:  5072 6f66 696c 6573 225d 7d5d            Profiles"]}]
```

```
16:12:59.831183 IP (tos 0x0, ttl 254, id 507, offset 0, flags [none], proto TCP (6), length 321)
    grizzle.60988 > hass.9000: Flags [P.], cksum 0x2d43 (correct), seq 239:520, ack 306, win 5439, length 281
	0x0000:  4500 0141 01fb 0000 fe06 345a c0a8 040b  E..A......4Z....
	0x0010:  c0a8 0006 ee3c 2328 a6a7 9192 7c2d 142e  .....<#(....|-..
	0x0020:  5018 153f 2d43 0000 81fe 0111 0000 0000  P..?-C..........
	0x0030:  5b33 2c22 3730 6633 6632 3437 2d63 3333  [3,"70f3f247-c33
	0x0040:  362d 3439 6563 2d38 6139 632d 6262 3834  6-49ec-8a9c-bb84
	0x0050:  6632 3662 3839 3466 222c 7b22 636f 6e66  f26b894f",{"conf
	0x0060:  6967 7572 6174 696f 6e4b 6579 223a 5b7b  igurationKey":[{
	0x0070:  226b 6579 223a 2253 7570 706f 7274 6564  "key":"Supported
	0x0080:  4665 6174 7572 6550 726f 6669 6c65 7322  FeatureProfiles"
	0x0090:  2c22 7661 6c75 6522 3a22 436f 7265 2c46  ,"value":"Core,F
	0x00a0:  6972 6d77 6172 654d 616e 6167 656d 656e  irmwareManagemen
	0x00b0:  742c 4c6f 6361 6c41 7574 684c 6973 744d  t,LocalAuthListM
	0x00c0:  616e 6167 656d 656e 742c 5265 7365 7276  anagement,Reserv
	0x00d0:  6174 696f 6e2c 536d 6172 7443 6861 7267  ation,SmartCharg
	0x00e0:  696e 672c 5265 6d6f 7465 5472 6967 6765  ing,RemoteTrigge
	0x00f0:  7222 2c22 7265 6164 6f6e 6c79 223a 7472  r","readonly":tr
	0x0100:  7565 7d2c 7b22 6b65 7922 3a22 5769 4669  ue},{"key":"WiFi
	0x0110:  5369 676e 616c 5374 7265 6e67 7468 222c  SignalStrength",
	0x0120:  2276 616c 7565 223a 222d 3631 222c 2272  "value":"-61","r
	0x0130:  6561 646f 6e6c 7922 3a74 7275 657d 5d7d  eadonly":true}]}
	0x0140:  5d                                       ]
```

djiesr, 2023-09-10:
I have the same error:

I make my WS server to try the command I have the same error. 

This what I do (I'm not a programmer)

`import { WebSocketServer } from 'ws';

const wss = new WebSocketServer({ port: 9000 });

wss.on('connection', function connection(ws) {
  ws.on('message', function message(data) {
    console.log('received: %s', data);
	if (data.includes('BootNotification')) {
	  ws.send('[3,"1481766031",{"currentTime":"2023-09-10T09:46:52Z","interval":3600,"status":"Accepted"}]');
	}
	if (data.includes('1481766032')) {
	  ws.send('[3,"1481766032",{}]');
	}
	if (data.includes('1481766033')) {
	  ws.send('[3,"1481766033",{}]');
	  ws.send('[2,"########-####-####-####-############","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]');
	//         [2,"########-####-####-####-############","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
	}
  });

});`

And I got: received: [4, "49eab272-b834-45e1-8674-90dee33e5fbd", "FormationViolation", ""]

In de documentation, the server can ask: 
[2,"0116359a-f5b2-4f6b-af28-b845dd10c138","GetConfiguration",{}]

When I try, the charger return:
[3,"########-####-####-####-############",{"configurationKey":[{"key":"MaxCurrent","readonly":true,"value":"32000"},{"key":"MinCurrent","readonly":true,"value":"8000"},{"key":"SetCurrent","readonly":false,"value":"32000"},{"key":"StartChargingAfterConnect","readonly":false,"value":"true"},{"key":"AuthorizeRemoteTxRequests","readonly":true,"value":"true"},{"key":"ConnectorPhaseRotation","readonly":true,"value":"NotApplicable"},{"key":"HeartbeatInterval","readonly":false,"value":"3600"},{"key":"LocalAuthorizeOffline","readonly":true,"value":"true"},{"key":"LocalPreAuthorize","readonly":true,"value":"true"},{"key":"MeterValueSampleInterval","readonly":false,"value":"20"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ResetRetries","readonly":true,"value":"1"},{"key":"StopTransactionOnInvalid","readonly":true,"value":"false"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement"},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":true,"value":"false"}]}]

I try with MeterValueSampleInterval, I have the same error. If I try {"key":"SupportedFeatureProfiles"} i got the same thing of [2,"0116359a-f5b2-4f6b-af28-b845dd10c138","GetConfiguration",{}]

drc38, 2023-09-10:
The issue appears to be buggy firmware, you can try getting further diagnostics from the unit. See thread #221 for details.

mkbrown, 2023-09-10:
> The issue appears to be buggy firmware, you can try getting further diagnostics from the unit. See thread #221 for details on using get_diagnostics.

The integration doesn’t get to a state where any ocpp service is available in the developer tools services, so get_diagnostics is not present. I’ve enabled asyncio debug as well, and will try to capture a trace with that and websockets, as well as tcpdump.

drc38, 2023-09-10:
In post_connect go through and comment out any lines the unit is not accepting and see if you can get to the point the integration loads.

djiesr, 2023-09-10:
It's possible to patch if I get [2,"0116359a-f5b2-4f6b-af28-b845dd10c138","GetConfiguration",{}] and select SupportedFeatureProfiles from it.

mkbrown, 2023-09-11:
Ok, so I commented out some stuff from get_supported features and forced the core profile.  

[image]

Looks like it’s having issues with NumberOfConnectors

```
2023-09-10 20:41:08.211 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2023-09-10 20:41:08.215 DEBUG (MainThread) [websockets.server] < GET /GRS-170000xxxxx HTTP/1.1
2023-09-10 20:41:08.216 DEBUG (MainThread) [websockets.server] < Connection: Upgrade
2023-09-10 20:41:08.216 DEBUG (MainThread) [websockets.server] < Host: 172.20.21.150:9000
2023-09-10 20:41:08.216 DEBUG (MainThread) [websockets.server] < User-Agent: ESP32 Websocket Client
2023-09-10 20:41:08.216 DEBUG (MainThread) [websockets.server] < Upgrade: websocket
2023-09-10 20:41:08.216 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Version: 13
2023-09-10 20:41:08.216 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Key: qZP1iJqdMi3Tt+UcdeiyDg==
2023-09-10 20:41:08.216 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Protocol: ocpp1.6
2023-09-10 20:41:08.216 DEBUG (MainThread) [websockets.server] < Authorization: Basic R1JTLTE3MDAwMDNjNjgzOg==
2023-09-10 20:41:08.216 DEBUG (MainThread) [websockets.server] > HTTP/1.1 101 Switching Protocols
2023-09-10 20:41:08.216 DEBUG (MainThread) [websockets.server] > Upgrade: websocket
2023-09-10 20:41:08.217 DEBUG (MainThread) [websockets.server] > Connection: Upgrade
2023-09-10 20:41:08.217 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Accept: Qhqjr2rHLkeVqzJpI/gfKL+CaFc=
2023-09-10 20:41:08.217 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Protocol: ocpp1.6
2023-09-10 20:41:08.217 DEBUG (MainThread) [websockets.server] > Date: Mon, 11 Sep 2023 00:41:08 GMT
2023-09-10 20:41:08.217 DEBUG (MainThread) [websockets.server] > Server: Python/3.11 websockets/11.0.3
2023-09-10 20:41:08.217 INFO (MainThread) [websockets.server] connection open
2023-09-10 20:41:08.217 DEBUG (MainThread) [websockets.server] = connection is OPEN
2023-09-10 20:41:08.217 WARNING (MainThread) [custom_components.ocpp] Skipping websocket subprotocol validation
2023-09-10 20:41:08.220 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/GRS-170000xxxxx
2023-09-10 20:41:08.220 INFO (MainThread) [custom_components.ocpp] Charger GRS-170000xxxxx connected to 172.20.21.150:9000.
2023-09-10 20:41:08.297 DEBUG (MainThread) [websockets.server] < TEXT ‘[2,”1481766072”,”BootNotification”,{“chargePoin...7”,”meterType”:”wifi”}]’ [222 bytes]
2023-09-10 20:41:08.298 INFO (MainThread) [ocpp] GRS-170000xxxxx: receive message [2,”1481766072”,”BootNotification”,{“chargePointModel”:”Grizzl-E Smart”,”chargePointVendor”:”UnitedChargers”,”chargePointSerialNumber”:”GRS-170000xxxxx”,”firmwareVersion”:”GWM-07.013-03_GCW-10.18-05.7”,”meterType”:”wifi”}]
2023-09-10 20:41:08.300 DEBUG (MainThread) [custom_components.ocpp] Received boot notification for GRS-170000xxxxx: {‘charge_point_model’: ‘Grizzl-E Smart’, ‘charge_point_vendor’: ‘UnitedChargers’, ‘charge_point_serial_number’: ‘GRS-170000xxxxx’, ‘firmware_version’: ‘GWM-07.013-03_GCW-10.18-05.7’, ‘meter_type’: ‘wifi’}
2023-09-10 20:41:08.302 INFO (MainThread) [ocpp] GRS-170000xxxxx: send [3,”1481766072”,{“currentTime”:”2023-09-11T00:41:08Z”,”interval”:3600,”status”:”Accepted”}]
2023-09-10 20:41:08.303 DEBUG (MainThread) [websockets.server] > TEXT ‘[3,”1481766072”,{“currentTime”:”2023-09-11T00:4...0,”status”:”Accepted”}]’ [91 bytes]
2023-09-10 20:41:08.303 DEBUG (MainThread) [custom_components.ocpp] Updating device info charger: {‘charge_point_model’: ‘Grizzl-E Smart’, ‘charge_point_vendor’: ‘UnitedChargers’, ‘charge_point_serial_number’: ‘GRS-170000xxxxx’, ‘firmware_version’: ‘GWM-07.013-03_GCW-10.18-05.7’, ‘meter_type’: ‘wifi’}
2023-09-10 20:41:08.398 DEBUG (MainThread) [websockets.server] < TEXT ‘[2,”1481766073”,”StartTransaction”,{“connectorI...0000-00-00T00:00:00Z”}]’ [118 bytes]
2023-09-10 20:41:08.398 INFO (MainThread) [ocpp] GRS-170000xxxxx: receive message [2,”1481766073”,”StartTransaction”,{“connectorId”:0, “idTag”:””, “meterStart”:0, “timestamp”: “0000-00-00T00:00:00Z”}]
2023-09-10 20:41:08.400 DEBUG (MainThread) [custom_components.ocpp] id_tag=‘’ not found in auth_list, default authorization_status=‘Accepted’
2023-09-10 20:41:08.401 INFO (MainThread) [ocpp] GRS-170000xxxxx: send [3,”1481766073”,{“transactionId”:1694392868,”idTagInfo”:{“status”:”Accepted”}}]
2023-09-10 20:41:08.401 DEBUG (MainThread) [websockets.server] > TEXT ‘[3,”1481766073”,{“transactionId”:1694392868,”id...{“status”:”Accepted”}}]’ [79 bytes]
2023-09-10 20:41:09.530 DEBUG (MainThread) [websockets.server] < TEXT ‘[2,”1481766074”,”StatusNotification”,{“connecto...,”status”:”Available”}]’ [98 bytes]
2023-09-10 20:41:09.531 DEBUG (MainThread) [websockets.server] < TEXT ‘[2,”1481766075”,”StatusNotification”,{“connecto...,”status”:”Available”}]’ [98 bytes]
2023-09-10 20:41:09.531 INFO (MainThread) [ocpp] GRS-170000xxxxx: receive message [2,”1481766074”,”StatusNotification”,{“connectorId”:0,”errorCode”:”NoError”,”status”:”Available”}]
2023-09-10 20:41:09.539 INFO (MainThread) [ocpp] GRS-170000xxxxx: send [3,”1481766074”,{}]
2023-09-10 20:41:09.539 DEBUG (MainThread) [websockets.server] > TEXT ‘[3,”1481766074”,{}]’ [19 bytes]
2023-09-10 20:41:09.540 INFO (MainThread) [ocpp] GRS-170000xxxxx: receive message [2,”1481766075”,”StatusNotification”,{“connectorId”:1,”errorCode”:”NoError”,”status”:”Available”}]
2023-09-10 20:41:09.540 INFO (MainThread) [ocpp] GRS-170000xxxxx: send [3,”1481766075”,{}]
[... cut ...]
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/config/custom_components/ocpp/api.py”, line 797, in get_configuration
resp = await self.call(req)
^^^^^^^^^^^^^^^^^^^^
File “/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py”, line 285, in call
raise asyncio.TimeoutError(
TimeoutError: Waited 30s for response on [2,”dd71025b-4c5b-4d93-a772-a5b857a72af7”,”GetConfiguration”,{“key”:[“NumberOfConnectors”]}].
2023-09-10 20:50:31.618 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
```

drc38, 2023-09-11:
Looks like the issue is any get_configuration call that passes a parameter. So they'll all need to be commented out and replaced with a default value.

mkbrown, 2023-09-11:
Ok, I know enough about python to be dangerous (to myself)… the integration still won’t load enough to have get_diagnostics.  Is there a way I can force it to run at the start of get_supported_features so that we can have the diag output before it tries to do anything that will likely fail?

djiesr, 2023-09-11:
like a say before, it's possible to use
[2,"0116359a-f5b2-4f6b-af28-b845dd10c138","GetConfiguration",{}] 
to get all informatgion (like SupportedFeatureProfiles) and select what you want in the list.

But it's not possible to use
[2,"0116359a-f5b2-4f6b-af28-b845dd10c138","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]'
You will receive "FormationViolation"

mkbrown, 2023-09-11:
Ok, so I tried the SteVe add-on, and was able to successfully connect to the Grizzl-e.  Here’s it’s get configuration function

[image]

mkbrown, 2023-09-11:
Looks like Get diagnostics isn’t implemented in the Grizzl-e v2 firmware. 

[image]

8jmm3, 2023-09-11:
They've obviously implemented (prioritized) the Grizzl-E v2 firmware to work with the Grizzl-E Connect app and it is still a work in progress.  Is it possible to capture a trace of the traffic in a Grizzl-E Connect session to see what they are exchanging?  Would this be of any use?

Do you have any idea what charging statistics the app is able to obtain from the EVSE?  I notice that several of the sensors (e.g. Current Export, Current Offered, Power Offered, etc.) don't seem to populate with the v1 hardware.  I don't know if this is a limitation of the firmware and whether it's any better with v2.

djiesr, 2023-09-21:
It's work after I modified 2 sections in api.py

First:

`    async def get_supported_features(self):
        """Get supported features."""
###        req = call.GetConfigurationPayload(key=[ckey.supported_feature_profiles.value])
###        resp = await self.call(req)
###        feature_list = (resp.configuration_key[0][om.value.value]).split(",")
        feature_string = "Core,FirmwareManagement"
        feature_list = feature_string.split(',')
        if feature_list[0] == "":
            _LOGGER.warning("No feature profiles detected, defaulting to Core")
            await self.notify_ha("No feature profiles detected, defaulting to Core")
            feature_list = [om.feature_profile_core.value]
        if self.central.config.get(
            CONF_FORCE_SMART_CHARGING, DEFAULT_FORCE_SMART_CHARGING
        ):
            _LOGGER.warning("Force Smart Charging feature profile")
            self._attr_supported_features |= prof.SMART
        for item in feature_list:
            item = item.strip().replace(" ", "")
            if item == om.feature_profile_core.value:
                self._attr_supported_features |= prof.CORE
            elif item == om.feature_profile_firmware.value:
                self._attr_supported_features |= prof.FW
            elif item == om.feature_profile_smart.value:
                self._attr_supported_features |= prof.SMART
            elif item == om.feature_profile_reservation.value:
                self._attr_supported_features |= prof.RES
            elif item == om.feature_profile_remote.value:
                self._attr_supported_features |= prof.REM
            elif item == om.feature_profile_auth.value:
                self._attr_supported_features |= prof.AUTH
            else:
                _LOGGER.warning("Unknown feature profile detected ignoring: %s", item)
                await self.notify_ha(
                    f"Warning: Unknown feature profile detected ignoring {item}"
                )
        self._metrics[cdet.features.value].value = self._attr_supported_features
        _LOGGER.debug("Feature profiles returned: %s", self._attr_supported_features)`

And this one:

`    async def get_configuration(self, key: str = ""):
        """Get Configuration of charger for supported keys else return None."""
        req = call.GetConfigurationPayload()
        resp = await self.call(req)
        _LOGGER.debug("Complet get config value is %s", resp)

        if key:
            # Recherche de la clé spécifiée
            for config_item in resp.configuration_key:
                if config_item.get("key") == key:
                    value = config_item.get("value")
                    _LOGGER.debug("Get Configuration for %s: %s", key, value)
                    self._metrics[cdet.config_response.value].value = datetime.now(tz=timezone.utc)
                    self._metrics[cdet.config_response.value].extra_attr = {key: value}
                    return value

        if resp.configuration_key is not None:
            value = resp.configuration_key[0][om.value.value]
            _LOGGER.debug("Get Configuration for %s: %s", key, value)
            self._metrics[cdet.config_response.value].value = datetime.now(tz=timezone.utc)
            self._metrics[cdet.config_response.value].extra_attr = {key: value}
            return value

        if resp.unknown_key is not None:
            _LOGGER.warning("Get Configuration returned unknown key for: %s", key)
            await self.notify_ha(f"Warning: charger reports {key} is unknown")
[... 44 lines cut ...]

        req = call.ChangeConfigurationPayload(key=key, value=value)

        resp = await self.call(req)

        if resp.status in [
            ConfigurationStatus.rejected,
            ConfigurationStatus.not_supported,
        ]:
            _LOGGER.warning("%s while setting %s to %s", resp.status, key, value)
            await self.notify_ha(
                f"Warning: charger reported {resp.status} while setting {key}={value}"
            )

        if resp.status == ConfigurationStatus.reboot_required:
            self._requires_reboot = True
            await self.notify_ha(f"A reboot is required to apply {key}={value}")`

And, prevent disconnecting, I modified const.py with this:

DEFAULT_WEBSOCKET_PING_TRIES = 5 #2
DEFAULT_WEBSOCKET_PING_INTERVAL = 60 #20
DEFAULT_WEBSOCKET_PING_TIMEOUT = 30 #20

[image]

windseye, 2023-10-06:
I am very interested in this discussion. I just received a replacement Grizzl-E Smart EVSE for the one that I bricked by loading old firmware. UC Connect says I now have Firmware Version:GWM-07.020-03_GCW-10.18-05.7. The Advanced OCPP Configuration option has been removed from this firmware.

kenttse, 2023-10-06:
> I am very interested in this discussion. I just received a replacement Grizzl-E Smart EVSE for the one that I bricked by loading old firmware. UC Connect says I now have Firmware Version:GWM-07.020-03_GCW-10.18-05.7. The Advanced OCPP Configuration option has been removed from this firmware. I have to decide whether to keep this unit or sell it on the used market because Homeassistant connectivity is a must-have for me. It seems that there is optimism in this group that an OCPP compliant firmware will be developed although not by United Chargers. Am I being overly optimistic? Should I keep the unit and hope for a solution?

The Advanced OCPP Configuration is now hidden in this firmware. On the WiFI configuration page you need to click the "ONLINE"  at the top left 10 times. The Advanced OCPP Configuration button will then appear as before. I haven't had time to test out the above patching to see if it gives me the right functionality yet.

[image]

djiesr, 2023-10-06:
Let me know if this firmware work well with the OCPP intergration. My version didn't work properly, so I make this fork with some modification: https://github.com/djiesr/ocpp

windseye, 2023-10-07:
Thanks djiesr. I am making progress with your fork but get this error:

2023-10-07 09:27:38.423 ERROR (MainThread) [ocpp] Error while handling request '<Call - unique_id=1481765945, action=BootNotification, payload={'chargePointModel': 'Grizzl-E Smart Connect', 'chargePointVendor': 'UnitedChargers', 'chargePointSerialNumber': 'GRS-1700003fb5d', 'firmwareVersion': 'GWM-07.020-03_GCW-10.18-05.7', 'meterType': ''}>'
Traceback (most recent call last):
File "/usr/local/lib/python3.11/site-packages/ocpp/messages.py", line 227, in validate_payload
validator.validate(message.payload)
File "/usr/local/lib/python3.11/site-packages/jsonschema/validators.py", line 435, in validate
raise error
jsonschema.exceptions.ValidationError: 'Grizzl-E Smart Connect' is too long
Failed validating 'maxLength' in schema['properties']['chargePointModel']:
{'maxLength': 20, 'type': 'string'}
On instance['chargePointModel']:
'Grizzl-E Smart Connect'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 153, in route_message
await self._handle_call(msg)
File "/config/custom_components/ocpp/api.py", line 945, in _handle_call
await super()._handle_call(msg)
File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 181, in _handle_call
validate_payload(msg, self._ocpp_version)
File "/usr/local/lib/python3.11/site-packages/ocpp/messages.py", line 240, in validate_payload
raise TypeConstraintViolationError(
ocpp.exceptions.TypeConstraintViolationError: TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "'Grizzl-E Smart Connect' is too long", 'ocpp_message': <Call - unique_id=1481765945, action=BootNotification, payload={'chargePointModel': 'Grizzl-E Smart Connect', 'chargePointVendor': 'UnitedChargers', 'chargePointSerialNumber': 'GRS-1700003fb5d', 'firmwareVersion': 'GWM-07.020-03_GCW-10.18-05.7', 'meterType': ''}>}
2023-10-07 09:27:38.431 INFO (MainThread) [ocpp] GRS-1700003fb5d: send [4,"1481765945","TypeConstraintViolation","Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. \u201csomestring\u201d: 12)",{"cause":"'Grizzl-E Smart Connect' is too long","ocpp_message":"[2,\"1481765945\",\"BootNotification\",{\"chargePointModel\":\"Grizzl-E Smart Connect\",\"chargePointVendor\":\"UnitedChargers\",\"chargePointSerialNumber\":\"GRS-1700003fb5d\",\"firmwareVersion\":\"GWM-07.020-03_GCW-10.18-05.7\",\"meterType\":\"\"}]"}]
2023-10-07 09:27:38.432 DEBUG (MainThread) [websockets.server] > TEXT '[4,"1481765945","TypeConstraintViolation","Payl...eterType\\":\\"\\"}]"}]' [501 bytes]
2023-10-07 09:27:48.458 DEBUG (MainThread) [websockets.server] < PING '' [0 bytes]
2023-10-07 09:27:48.458 DEBUG (MainThread) [websockets.server] > PONG '' [0 bytes]
2023-10-07 09:27:48.634 DEBUG (MainThread) [websockets.server] > PING 8c 55 b3 59 [binary, 4 bytes]
2023-10-07 09:27:48.902 DEBUG (MainThread) [websockets.server] < PONG 8c 55 b3 59 [binary, 4 bytes]
2023-10-07 09:27:48.903 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'GRS-1700003fb5d': ping=1.0 ms, pong=268.0 ms, retry=0
2023-10-07 09:27:58.873 DEBUG (MainThread) [websockets.server] < PING '' [0 bytes]
2023-10-07 09:27:58.873 DEBUG (MainThread) [websockets.server] > PONG '' [0 bytes]
2023-10-07 09:28:08.904 DEBUG (MainThread) [websockets.server] < PING '' [0 bytes]
2023-10-07 09:28:08.904 DEBUG (MainThread) [websockets.server] > PONG '' [0 bytes]
2023-10-07 09:28:08.906 DEBUG (MainThread) [websockets.server] > PING e8 c0 b7 6e [binary, 4 bytes]
2023-10-07 09:28:08.967 DEBUG (MainThread) [websockets.server] < PONG e8 c0 b7 6e [binary, 4 bytes]
2023-10-07 09:28:08.967 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'GRS-1700003fb5d': ping=2.0 ms, pong=60.0 ms, retry=0

djiesr, 2023-10-07:
Ok, in my version is Grizzl-E Smart and in your firmware is "Grizzl-E Smart
Connect". I didn't modify this part, so you would have the same error in
OCPP. Let me look at this.

djiesr, 2023-10-07:
I make a v0.1.5. I just limit the model at 20 character, so you will have a "Grizzl-E Smart Conne". Sorry about that, I just patch the it, thje problem is in "/usr/local/lib/python3.11/site-packages/" and I don't know how resolve that.

windseye, 2023-10-07:
[home-assistant_2023-10-07T15-30-02.492Z.log](https://github.com/lbbrhzn/ocpp/files/12838143/home-assistant_2023-10-07T15-30-02.492Z.log)
I am still having the same problem.

windseye, 2023-10-07:
It appears that the problem is with the United Chargers firmware: OCPP 1.6 defines chargePointModel as CiString20Type
The firmware outputs a string of 22 characters and fails the validation test

drc38, 2023-10-07:
In the `on` function you can add `skip_schema_validation=True`

windseye, 2023-10-07:
That does not fix the problem. I believe the problem is packet validation not schema validationSent from my T-Mobile 5G Device

djiesr, 2023-10-07:
Sorry about that,
In api.py, I already add the limit of 20. Exemple:

`    @on(Action.BootNotification)
    def on_boot_notification(self, **kwargs):
        """Handle a boot notification."""
        resp = call_result.BootNotificationPayload(
            current_time=datetime.now(tz=timezone.utc).strftime("%Y-%m-%dT%H:%M:%SZ"),
            interval=3600,
            status=RegistrationStatus.accepted.value,
        )
        self.received_boot_notification = True
        _LOGGER.debug("Received boot notification for %s: %s", self.id, kwargs)
        # update metrics
        self._metrics[cdet.model.value].value = kwargs.get(
            om.charge_point_model.name[:20], None`
....

But, that did't change for you.

The probleme is in the OCPP 1.6

[image]

You can ask to change 
        "chargePointModel": {
            "type": "string",
            "maxLength": 20
        },

20 for 25

drc38, 2023-10-07:
Another option would be to copy and modify the ocpp package. It needs to go into `lib/python3.11/site-packages/` under your config folder

drc38, 2023-10-07:
In the log above 
>File /usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 181, in _handle_call
validate_payload(msg, self._ocpp_version)

This should not be called if skip_schema_validation is set to True

windseye, 2023-10-07:
I will look at this more carefully MondaySent from my T-Mobile 5G Device

djiesr, 2023-10-08:
When you install the integration, you have a check box "Skip OCPP schema validation".

windseye, 2023-10-09:
I am still getting this error even at  2023-10-09 10:48:37.188 ERROR below even though it is skipping websocket protocol validation. See 2023-10-09 10:48:36.097 WARNING

2023-10-09 10:48:36.088 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2023-10-09 10:48:36.093 DEBUG (MainThread) [websockets.server] < GET /GRS-1700003fb5d HTTP/1.1
2023-10-09 10:48:36.093 DEBUG (MainThread) [websockets.server] < Connection: Upgrade
2023-10-09 10:48:36.093 DEBUG (MainThread) [websockets.server] < Host: 192.168.0.33:9002
2023-10-09 10:48:36.093 DEBUG (MainThread) [websockets.server] < User-Agent: ESP32 Websocket Client
2023-10-09 10:48:36.093 DEBUG (MainThread) [websockets.server] < Upgrade: websocket
2023-10-09 10:48:36.093 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Version: 13
2023-10-09 10:48:36.094 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Key: wP6PRZLaqnSK7nDSnINDiQ==
2023-10-09 10:48:36.094 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Protocol: ocpp1.6
2023-10-09 10:48:36.094 DEBUG (MainThread) [websockets.server] < Authorization: Basic R1JTLTE3MDAwMDNmYjVkOg==
2023-10-09 10:48:36.095 DEBUG (MainThread) [websockets.server] > HTTP/1.1 101 Switching Protocols
2023-10-09 10:48:36.095 DEBUG (MainThread) [websockets.server] > Upgrade: websocket
2023-10-09 10:48:36.095 DEBUG (MainThread) [websockets.server] > Connection: Upgrade
2023-10-09 10:48:36.095 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Accept: NWIgVfERmV48NqZRnGrxTARCm40=
2023-10-09 10:48:36.096 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Protocol: ocpp1.6
2023-10-09 10:48:36.096 DEBUG (MainThread) [websockets.server] > Date: Mon, 09 Oct 2023 14:48:36 GMT
2023-10-09 10:48:36.096 DEBUG (MainThread) [websockets.server] > Server: Python/3.11 websockets/11.0.3
2023-10-09 10:48:36.096 INFO (MainThread) [websockets.server] connection open
2023-10-09 10:48:36.096 DEBUG (MainThread) [websockets.server] = connection is OPEN
2023-10-09 10:48:36.097 WARNING (MainThread) [custom_components.ocpp] Skipping websocket subprotocol validation
2023-10-09 10:48:36.107 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/GRS-1700003fb5d
2023-10-09 10:48:36.108 INFO (MainThread) [custom_components.ocpp] Charger GRS-1700003fb5d connected to 0.0.0.0:9002.
2023-10-09 10:48:36.742 INFO (MainThread) [homeassistant.setup] Setup of domain flux_led took 9.7 seconds
2023-10-09 10:48:37.173 DEBUG (MainThread) [websockets.server] < TEXT '[2,"1481765940","BootNotification",{"chargePoin...,"meterType":"server"}]' [232 bytes]
2023-10-09 10:48:37.181 DEBUG (MainThread) [websockets.server] < PING '' [0 bytes]
2023-10-09 10:48:37.182 DEBUG (MainThread) [websockets.server] > PONG '' [0 bytes]
2023-10-09 10:48:37.182 INFO (MainThread) [ocpp] GRS-1700003fb5d: receive message [2,"1481765940","BootNotification",{"chargePointModel":"Grizzl-E Smart Connect","chargePointVendor":"UnitedChargers","chargePointSerialNumber":"GRS-1700003fb5d","firmwareVersion":"GWM-07.020-03_GCW-10.18-05.7","meterType":"server"}]
2023-10-09 10:48:37.188 ERROR (MainThread) [ocpp] Error while handling request '<Call - unique_id=1481765940, action=BootNotification, payload={'chargePointModel': 'Grizzl-E Smart Connect', 'chargePointVendor': 'UnitedChargers', 'chargePointSerialNumber': 'GRS-1700003fb5d', 'firmwareVersion': 'GWM-07.020-03_GCW-10.18-05.7', 'meterType': 'server'}>'
Traceback (most recent call last):
File "/usr/local/lib/python3.11/site-packages/ocpp/messages.py", line 227, in validate_payload
validator.validate(message.payload)
File "/usr/local/lib/python3.11/site-packages/jsonschema/validators.py", line 435, in validate
raise error
jsonschema.exceptions.ValidationError: 'Grizzl-E Smart Connect' is too long
Failed validating 'maxLength' in schema['properties']['chargePointModel']:
{'maxLength': 20, 'type': 'string'}
On instance['chargePointModel']:
'Grizzl-E Smart Connect'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 153, in route_message
await self._handle_call(msg)
File "/config/custom_components/ocpp/api.py", line 945, in _handle_call
await super()._handle_call(msg)
File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 181, in _handle_call
validate_payload(msg, self._ocpp_version)
File "/usr/local/lib/python3.11/site-packages/ocpp/messages.py", line 240, in validate_payload
raise TypeConstraintViolationError(
ocpp.exceptions.TypeConstraintViolationError: TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "'Grizzl-E Smart Connect' is too long", 'ocpp_message': <Call - unique_id=1481765940, action=BootNotification, payload={'chargePointModel': 'Grizzl-E Smart Connect', 'chargePointVendor': 'UnitedChargers', 'chargePointSerialNumber': 'GRS-1700003fb5d', 'firmwareVersion': 'GWM-07.020-03_GCW-10.18-05.7', 'meterType': 'server'}>}
2023-10-09 10:48:37.204 INFO (MainThread) [ocpp] GRS-1700003fb5d: send [4,"1481765940","TypeConstraintViolation","Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. \u201csomestring\u201d: 12)",{"cause":"'Grizzl-E Smart Connect' is too long","ocpp_message":"[2,\"1481765940\",\"BootNotification\",{\"chargePointModel\":\"Grizzl-E Smart Connect\",\"chargePointVendor\":\"United [... cut ...]
2023-10-09 10:48:56.112 DEBUG (MainThread) [websockets.server] > PING 'P5ٲ' [text, 4 bytes]
2023-10-09 10:48:56.262 DEBUG (MainThread) [websockets.server] < PONG 'P5ٲ' [text, 4 bytes]
2023-10-09 10:48:56.262 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'GRS-1700003fb5d': ping=1.0 ms, pong=150.0 ms, retry=0
2023-10-09 10:49:06.229 DEBUG (MainThread) [websockets.server] < PING '' [0 bytes]
2023-10-09 10:49:06.230 DEBUG (MainThread) [websockets.server] > PONG '' [0 bytes]

drc38, 2023-10-09:
Suggest adding extra logging to `api.py` under this line:
`self.route_map[action]["_skip_schema_validation"] = skip_schema_validation`
to check the skip schema validation is being correctly set for the action

djiesr, 2023-10-09:
When you add the intégration, you select ignore shema validation? like this? (Sorry, is in french)
[image]

windseye, 2023-10-10:
Yes, this box is checked. I reinstalled the integration once more and verified the checkbox. The log shows:

2023-10-10 07:30:26.403 WARNING (MainThread) [custom_components.ocpp] Skipping websocket subprotocol validation

which seems to confirm this.

windseye, 2023-10-10:
I believe that the error is in the packet validation and not the websocket protocol validation. I am operating above my knowledge level here but the problem appears to originate from this section of File "/usr/local/lib/python3.11/site-packages/ocpp/messages.py", line 227, in validate_payload
validator.validate(message.payload):

 226      try:
   227          validator.validate(message.payload)
   228      except SchemaValidationError as e:
   229          if e.validator == SchemaValidators.type.__name__:
   230              raise TypeConstraintViolationError(
   231                  details={"cause": e.message, "ocpp_message": message}
   232              )
   233          elif e.validator == SchemaValidators.additionalProperties.__name__:
   234              raise FormatViolationError(
   235                  details={"cause": e.message, "ocpp_message": message}
   236              )
   237          elif e.validator == SchemaValidators.required.__name__:
   238              raise ProtocolError(details={"cause": e.message})
   239          elif e.validator == "maxLength":
   240              raise TypeConstraintViolationError(
   241                  details={"cause": e.message, "ocpp_message": message}
   242              ) from e
   
Where is the maxLength variable set?

windseye, 2023-10-10:
I found the maxLength variable is set in /usr/local/lib/python3.11/site-packages/ocpp/v16/schemas/BootNotification.json

I will try to change this on my device although it will violate the OCPP 1.6 specification.

I will report the result

windseye, 2023-10-10:
Apparent success! I changed line 12 in /usr/local/lib/python3.11/site-packages/ocpp/v16/schemas/BootNotification.json to 

  10          "chargePointModel": {
    11              "type": "string",
    12              "maxLength": **25**
    13          },

windseye, 2023-10-10:
There is progress but there is now a new recurring error that causes reboots:

2023-10-10 09:05:05.791 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/local/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
return fut.result()
^^^^^^^^^^^^
File "/usr/local/lib/python3.11/asyncio/queues.py", line 158, in get
await getter
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 281, in call
response = await self._get_specific_response(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/ocpp/api.py", line 891, in _get_specific_response
resp = await super()._get_specific_response(unique_id, timeout)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 315, in _get_specific_response
response = await asyncio.wait_for(self._response_queue.get(), timeout)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
raise exceptions.TimeoutError() from exc
TimeoutError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/config/custom_components/ocpp/api.py", line 421, in post_connect
await self.get_supported_features()
File "/config/custom_components/ocpp/api.py", line 498, in get_supported_features
resp = await self.get_configuration(ckey.supported_feature_profiles.value)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/ocpp/api.py", line 801, in get_configuration
resp = await self.call(req)
^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 285, in call
raise asyncio.TimeoutError(
TimeoutError: Waited 30s for response on [2,"8bf1afdb-ac11-4e20-bdb8-08312f885879","GetConfiguration",{}].

windseye, 2023-10-10:
The above error went away when I checked Force Smart Charging feature profile on re-installation. So far . . . stable.

windseye, 2023-10-10:
I am now getting the following:

2023-10-10 11:03:17.521 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/local/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
return fut.result()
^^^^^^^^^^^^
File "/usr/local/lib/python3.11/asyncio/queues.py", line 158, in get
await getter
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 281, in call
response = await self._get_specific_response(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/ocpp/api.py", line 891, in _get_specific_response
resp = await super()._get_specific_response(unique_id, timeout)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 315, in _get_specific_response
response = await asyncio.wait_for(self._response_queue.get(), timeout)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
raise exceptions.TimeoutError() from exc
TimeoutError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/config/custom_components/ocpp/api.py", line 421, in post_connect
await self.get_supported_features()
File "/config/custom_components/ocpp/api.py", line 498, in get_supported_features
resp = await self.get_configuration(ckey.supported_feature_profiles.value)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/ocpp/api.py", line 801, in get_configuration
resp = await self.call(req)
^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 285, in call
raise asyncio.TimeoutError(
TimeoutError: Waited 30s for response on [2,"1c2e8f0d-98cb-4d5b-9b8c-a9cfe486c310","GetConfiguration",{}].

It appears that you have dealt with this before. Please advise me how I should proceed. I see numerous comments in this part of the api.py file in the current version.

windseye, 2023-10-10:
I am now at a stalemate. I cannot seem to fix the error immediately above.

djiesr, 2023-10-11:
Arround the line 492, you can try tu put this par like the original api.py in OCPP.

replace: `
    async def get_supported_features(self):
        """Get supported features."""
###        req = call.GetConfigurationPayload(key=[ckey.supported_feature_profiles.value])
###        resp = await self.call(req)
###        feature_list = (resp.configuration_key[0][om.value.value]).split(",")
###
        resp = await self.get_configuration(ckey.supported_feature_profiles.value)
#        _LOGGER.warning("Feature list is: %s", resp)
        feature_list = resp.split(',')
###       
###        feature_string = "Core,FirmwareManagement"
###        feature_list = feature_string.split(',')
        if feature_list[0] == "":
`

By:
`    async def get_supported_features(self):
        """Get supported features."""
        req = call.GetConfigurationPayload(key=[ckey.supported_feature_profiles.value])
        resp = await self.call(req)
        feature_list = (resp.configuration_key[0][om.value.value]).split(",")
        if feature_list[0] == "":
`

windseye, 2023-10-15:
I will try to get back to this in a few days

windseye, 2023-10-24:
GOOD NEWS! I went back to the original /config/custom_components/ocpp/api.py with the modification to line 12 in /usr/local/lib/python3.11/site-packages/ocpp/v16/schemas/BootNotification.json and the EVSE is connected to the server and providing useful data. I am going to wait a few days to see if the connection remains stable before declaring the problem solved. One note: it is essential to specify the server in the following form (with the trailing backslash) ws://<ip>:<port>/ orit will not connect.

windseye, 2023-10-25:
It has been over 24 hours. The EVSE has been reporting useful data. There have been no spontaneous reboots. I consider this problem SOLVED. Thanks so much to djiesr and others for help. I will leave it to others more knowledgeable than I to update the repository for the problem is with the current United Chargers firmware:

OCPP 1.6 defines chargePointModel as CiString20Type
The current firmware outputs a string of 22 characters and fails the validation test. 

Thanks to all

drc38, 2023-10-25:
Good to hear you got it sorted. Note my comment on where you need to store your changes https://github.com/lbbrhzn/ocpp/issues/899#issuecomment-1751807194 otherwise when the underlying OCPP package is refresehd/updated (eg HA version upgrade) your changes will be overwritten.

BjarneBitscrambler, 2023-10-25:
@windseye - Great that you got it figured out!  Have you passed your observation back to 
United Chargers? It will be easier in the long run if they fix their non-conformances.

8jmm3, 2023-10-28:
After seeing the progress here I decided to go ahead and try one of the v2 units.  I can get it to connect to the United Chargers OCPP server and I can get it to attempt a connection to my local OCPP server (which currently fails because I'm still using the stock integration).

My current issue is that my v2 unit broadcasts an essentially open wifi network (password = "password") with the SSID being the serial number of the device.  Upon connecting to that network I can go to 192.168.4.1 to configure the device.  With the v1 units this network would disappear after configuring the wifi settings but now it appears to be persistent.  The problem is that ANYBODY can connect to this persistent wifi network allowing them to view your home wifi password and/or reconfigure the device.  Am I the only one experiencing this?

Firmware WiFi Module: GWM-07.020-03
Firmware Power Board: GCW-10.18-05.7

8jmm3, 2023-10-29:
As a proof of concept I modified the firmware (version GWM-07.020-03) and shortened the reported chargePointModel to be within the proper character limit.  Upon flashing the modified firmware it appears to be happy.

Flash using: `esptool.py --chip esp32 --baud 921600 write_flash 0x10000 GWM-07.020-03-mod.bin
`

Note that this is for a Grizzl-E Smart Connect (v2) and the charger must be power cycled after flashing the update.  Use at your own risk.  I've also included the unmodified firmware in case you need to revert to the original.

Firmware WiFi Module: GWM-07.020-03
Firmware Power Board: GCW-10.18-05.7

[GWM-07.020-03-mod.zip](https://github.com/lbbrhzn/ocpp/files/13197345/GWM-07.020-03-mod.zip)
[GWM-07.020-03.zip](https://github.com/lbbrhzn/ocpp/files/13197370/GWM-07.020-03.zip)

djiesr, 2023-10-29:
8jmm

> GWM-07.020-03-mod.zip

You have a uncompiled version of the bin?

8jmm3, 2023-10-29:
> 8jmm
> 
> > GWM-07.020-03-mod.zip
> 
> You have a uncompiled version of the bin?

If only I did.  No, I just patched the binary that came on the device.

djiesr, 2023-10-29:
Ok, I never try it to open a bin before. You don't have a error with "chargePointVendor"?

windseye, 2023-10-30:
Sorry folks, a new problem to add. The unit was working fine and giving usable information. I am in the process of trying to set up a wireguard tunnel which resulted in the EVSE being out of communication with the central server for a few minutes. When i got things reconnected I am getting reconnects every 2 minutes or so. Apparently an unsupported feature request. Here is the log:

2023-10-30 13:11:25.206 DEBUG (MainThread) [websockets.server] ! failing connection with code 1006
2023-10-30 13:11:25.207 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2023-10-30 13:11:25.209 DEBUG (MainThread) [websockets.server] x half-closing TCP connection
2023-10-30 13:11:25.209 INFO (MainThread) [custom_components.ocpp] Charger GRS-1700003fb5d disconnected from 0.0.0.0:9002.
2023-10-30 13:11:25.210 INFO (MainThread) [websockets.server] connection closed
2023-10-30 13:11:25.233 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2023-10-30 13:11:25.240 DEBUG (MainThread) [websockets.server] < GET /GRS-1700003fb5d HTTP/1.1
2023-10-30 13:11:25.240 DEBUG (MainThread) [websockets.server] < Connection: Upgrade
2023-10-30 13:11:25.241 DEBUG (MainThread) [websockets.server] < Host: 192.168.0.33:9002
2023-10-30 13:11:25.241 DEBUG (MainThread) [websockets.server] < User-Agent: ESP32 Websocket Client
2023-10-30 13:11:25.241 DEBUG (MainThread) [websockets.server] < Upgrade: websocket
2023-10-30 13:11:25.241 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Version: 13
2023-10-30 13:11:25.241 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Key: bLRMJrsukWeCPK05taX1cQ==
2023-10-30 13:11:25.241 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Protocol: ocpp1.6
2023-10-30 13:11:25.242 DEBUG (MainThread) [websockets.server] < Authorization: Basic R1JTLTE3MDAwMDNmYjVkOg==
2023-10-30 13:11:25.242 DEBUG (MainThread) [websockets.server] > HTTP/1.1 101 Switching Protocols
2023-10-30 13:11:25.242 DEBUG (MainThread) [websockets.server] > Upgrade: websocket
2023-10-30 13:11:25.243 DEBUG (MainThread) [websockets.server] > Connection: Upgrade
2023-10-30 13:11:25.243 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Accept: +ARFEFB/3aylKuGLRLQzsbjuIvU=
2023-10-30 13:11:25.243 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Protocol: ocpp1.6
2023-10-30 13:11:25.243 DEBUG (MainThread) [websockets.server] > Date: Mon, 30 Oct 2023 17:11:25 GMT
2023-10-30 13:11:25.243 DEBUG (MainThread) [websockets.server] > Server: Python/3.11 websockets/11.0.3
2023-10-30 13:11:25.244 INFO (MainThread) [websockets.server] connection open
2023-10-30 13:11:25.244 DEBUG (MainThread) [websockets.server] = connection is OPEN
2023-10-30 13:11:25.244 WARNING (MainThread) [custom_components.ocpp] Skipping websocket subprotocol validation
2023-10-30 13:11:25.244 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/GRS-1700003fb5d
2023-10-30 13:11:25.244 INFO (MainThread) [custom_components.ocpp] Charger GRS-1700003fb5d reconnected to 0.0.0.0:9002.
2023-10-30 13:11:26.269 DEBUG (MainThread) [websockets.server] < TEXT '[2,"1481767829","BootNotification",{"chargePoin..."meterType":"timeout"}]' [233 bytes]
2023-10-30 13:11:26.270 INFO (MainThread) [ocpp] GRS-1700003fb5d: receive message [2,"1481767829","BootNotification",{"chargePointModel":"Grizzl-E Smart Connect","chargePointVendor":"UnitedChargers","chargePointSerialNumber":"GRS-1700003fb5d","firmwareVersion":"GWM-07.020-03_GCW-10.18-05.7","meterType":"timeout"}]
2023-10-30 13:11:26.272 INFO (MainThread) [ocpp] GRS-1700003fb5d: send [4,"1481767829","NotImplemented","Requested Action is not known by receiver",{"cause":"Failed to validate action: BootNotification"}]
2023-10-30 13:11:26.273 DEBUG (MainThread) [websockets.server] > TEXT '[4,"1481767829","NotImplemented","Requested Act...on: BootNotification"}]' [133 bytes]
2023-10-30 13:11:27.283 DEBUG (MainThread) [websockets.server] < TEXT '[2,"1481767830","BootNotification",{"chargePoin...-05.7","meterType":""}]' [226 bytes]
2023-10-30 13:11:27.283 INFO (MainThread) [ocpp] GRS-1700003fb5d: receive message [2,"1481767830","BootNotification",{"chargePointModel":"Grizzl-E Smart Connect","chargePointVendor":"UnitedChargers","chargePointSerialNumber":"GRS-1700003fb5d","firmwareVersion":"GWM-07.020-03_GCW-10.18-05.7","meterType":""}]
2023-10-30 13:11:27.286 INFO (MainThread) [ocpp] GRS-1700003fb5d: send [4,"1481767830","NotImplemented","Requested Action is not known by receiver",{"cause":"Failed to validate action: BootNotification"}]
2023-10-30 13:11:27.286 DEBUG (MainThread) [websockets.server] > TEXT '[4,"1481767830","NotImplemented","Requested Act...on: BootNotification"}]' [133 bytes]
2023-10-30 13:11:37.298 DEBUG (MainThread) [websockets.server] < PING '' [0 bytes]
2023-10-30 13:11:37.298 DEBUG (MainThread) [websockets.server] > PONG '' [0 bytes]
2023-10-30 13:11:45.247 DEBUG (MainThread) [websockets.server] > PING '\x1e\x138\x06' [text, 4 bytes]
2023-10-30 13:11:45.353 DEBUG (MainThread) [websockets.server] < PONG '\x1e\x138\x06' [text, 4 bytes]
2023-10-30 13:11:55.322 DEBUG (MainThread) [websockets.server] < PING '' [0 bytes]
2023-10-30 13:11:55.322 DEBUG (MainThread) [websockets.server] > PONG '' [0 bytes]
2023-10-30 13:11:55.606 DEBUG (MainThread) [websockets.server] < TEXT '[2,"1481767831","BootNotification",{"chargePoin...-05.7","meterType":""}]' [226 bytes]
[... cut ...]
2023-10-30 13:12:15.393 DEBUG (MainThread) [websockets.server] > PONG '' [0 bytes]
2023-10-30 13:12:25.422 DEBUG (MainThread) [websockets.server] ! failing connection with code 1006
2023-10-30 13:12:25.423 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2023-10-30 13:12:25.424 DEBUG (MainThread) [websockets.server] x half-closing TCP connection
2023-10-30 13:12:25.425 INFO (MainThread) [custom_components.ocpp] Charger GRS-1700003fb5d disconnected from 0.0.0.0:9002.
2023-10-30 13:12:25.426 INFO (MainThread) [websockets.server] connection closed

8jmm3, 2023-10-30:
> Ok, I never try it to open a bin before. You don't have a error with "chargePointVendor"?

No error with "chargePointVendor" as it appears to fall without the allowable length stated in the OCPP specification, if I'm not mistaken.  "chargePointModel" was the only problem I noticed in this regard.  I implemented my firmware patch to avoid having to modify anything below the OCPP integration level so I can move my focus to the problems abated via api.py and such. This was easy because it just involved modifying a text string, not the binary code.  Of note, I changed chargePointModel to "Grizzl-E Smart" rather than "Grizzl-E Smart Connect", which would be evident in the logs.  Not sure whether that has any effect if trying to connect to the United Chargers OCPP server as I haven't tried yet.  I'm still in the process of further analyzing the firmware binary.  At least it doesn't appear to contain unreleased GPL code like the last iteration.

I'm currently back to playing with api.py since the above patch takes care of the BootNotification error.  I'm still getting the FormatValidation error and a reconnect every minute or two, but it otherwise seems to be initializing the sensors.  Perhaps I missed something in the prior discussion on this.  Will keep investigating.

djiesr, 2023-10-30:
I ask for the chargePointVendor beacause you had some space before like that  ...,    "chargePointVendor"... I just ask if you see a difference, but its good if not.
It's strange you get FormatValidation after you change it. 
If you check "ignore the validation of the OCPP scheme" when you add it, this is ok?

devzwf, 2023-10-31:
> My current issue is that my v2 unit broadcasts an essentially open wifi network (password = "password") with the SSID being the serial number of the device. Upon connecting to that network I can go [...]

Hello,
i am a bit concerned by that as well, even if you need to be close to the unit , still a red flag for me 

BTW thanks djiesr for all the effort :)

kenttse, 2023-10-31:
> > My current issue is that my v2 unit broadcasts an essentially open wifi network (password = "password") with the SSID being the serial number of the device. Upon connecting to that network I can go to 192.168.4.1 to configure the device. With the v1 units this network would disappear after configuring the wifi settings but now it appears to be persistent. The problem is that ANYBODY can connect to this persistent wifi network allowing them to view your home wifi password and/or reconfigure the device. Am I the only one experiencing this?
> 
> Hello, i am a bit concerned by that as well, even if you need to be close to the unit , still a red flag for me
> 
> BTW thanks @djiesr for all the efforts :)

I reported this as a security issue to United Chargers. Unfortunately, they did not give me an ETA on a fix or a patch to either allow for changing the password or turning the hotspot off when it is configured. I suggest everyone send in a bug report.

devzwf, 2023-10-31:
do you have a link handy for the bug report ?

[36 more replies]

closed 2024-05-16 by github-actions[bot]