lbbrhzn/ocpp #906: Can't connect two identical Terra AC chargers

hansenelectric, 2023-09-17:
It seems like I'm facing issues where the OCPP server mixes up the serial numbers for the two Terra AC chargers that I'm trying to connect to the same OCPP server @port 9000. Can you please advice how the device & entities is completely deleted or how it can be rectified? When I connect both the OCPP mixes them up and none of the chargers are working. Only one at the time where the only way I can identify the right charger is to disconnect the other one from the OCPP.
[home-assistant_ocpp_2023-09-17T16-45-26.794Z.log](https://github.com/lbbrhzn/ocpp/files/12643282/home-assistant_ocpp_2023-09-17T16-45-26.794Z.log)

lbbrhzn, 2023-09-17:
The home assistant OCPP integration currently only supports one charger per server. If you want to use multiple chargers then you should create multiple servers, with different ports (e.g. 9000 and 9001)

hansenelectric, 2023-09-18:
Okey. I created two servers and changed the names of the servers and chargers to new ones that haven't been used before and used the Alias from the Terra chargers. Then it did work. Thanks for such quick reply.

closed 2023-09-18 by hansenelectric