lbbrhzn/ocpp #912: Failed to set Up Integration on HA2023.9.3

TNTLarsn, 2023-09-25:
```

Logger: homeassistant.setup
Source: custom_components/ocpp/__init__.py:12
Integration: Open Charge Point Protocol (OCPP) ([documentation](https://github.com/lbbrhzn/ocpp/blob/main/README.md), [issues](https://github.com/lbbrhzn/ocpp/issues))
First occurred: 11:25:13 (1 occurrences)
Last logged: 11:25:13

Setup failed for custom integration ocpp: Unable to import component: cannot import name '_validators' from 'jsonschema' (/usr/local/lib/python3.11/site-packages/jsonschema/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 818, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/ocpp/__init__.py", line 12, in <module>
    from ocpp.v16.enums import AuthorizationStatus
  File "/usr/local/lib/python3.11/site-packages/ocpp/v16/__init__.py", line 1, in <module>
    from ocpp.charge_point import ChargePoint as cp
  File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 11, in <module>
    from ocpp.messages import Call, MessageType, unpack, validate_payload
  File "/usr/local/lib/python3.11/site-packages/ocpp/messages.py", line 12, in <module>
    from jsonschema import _validators as SchemaValidators
ImportError: cannot import name '_validators' from 'jsonschema' (/usr/local/lib/python3.11/site-packages/jsonschema/__init__.py)

```

dragorex71, 2023-09-25:
Same!
Here my setup

`<html>
<body>
<h2 id="title" class="mdc-dialog__title" style="-webkit-font-smoothing: antialiased; font-family: var(--mdc-typography-headline6-font-family,var(--mdc-typography-font-family,Roboto,sans-serif)); font-size: var(--mdc-typography-headline6-font-size,1.25rem); line-height: var(--mdc-typography-headline6-line-height,2rem); font-weight: var(--mdc-typography-headline6-font-weight,500); letter-spacing: var(--mdc-typography-headline6-letter-spacing,.0125em); text-decoration: var(--mdc-typography-headline6-text-decoration,inherit); text-transform: var(--mdc-typography-headline6-text-transform,inherit); display: block; position: relative; flex-shrink: 0; box-sizing: border-box; margin: 0px 0px 1px; padding: 24px 24px 0px; color: var(--mdc-dialog-heading-ink-color,rgba(0,0,0,.87));"><div class="header_title" style="margin-right: 32px; margin-inline-end: 32px; margin-inline-start: initial; direction: var(--direction);">Informazioni di sistema</div><ha-icon-button dialogaction="close" class="header_button" style="display: inline-block; outline: 0px; position: absolute; right: 16px; top: 14px; text-decoration: none; color: inherit; inset-inline-start: initial; inset-inline-end: 16px; direction: var(--direction);"><mwc-icon-button title="Chiudi" style="display: inline-block; outline: 0px; --mdc-ripple-color: currentcolor; -webkit-tap-highlight-color: transparent; vertical-align: top; --mdc-theme-on-primary: currentColor; --mdc-theme-text-disabled-on-light: var(--disabled-text-color);"><button class="mdc-icon-button mdc-icon-button--display-flex" aria-label="Chiudi" style="font-size: 24px; width: var(--mdc-icon-button-size,48px); height: var(--mdc-icon-button-size,48px); padding: calc((var(--mdc-icon-button-size,48px) - var(--mdc-icon-size,24px))/ 2); display: inline-flex; position: relative; box-sizing: border-box; border: none; outline: 0px; background-color: transparent; fill: currentcolor; color: inherit; text-decoration: none; cursor: pointer; user-select: none; z-index: 0; overflow: visible; align-items: center; justify-content: center; vertical-align: top;"><span><slot><ha-svg-icon style="display: block; align-items: center; justify-content: center; position: relative; vertical-align: middle; fill: currentcolor; width: var(--mdc-icon-size,24px); height: var(--mdc-icon-size,24px);"><svg preserveAspectRatio="xMidYMid meet" focusable="false" role="img" aria-hidden="true" viewBox="0 0 24 24"><g><path d="M19,6.41L17.59,5L12,10.59L6.41,5L5,6.41L10.59,12L5,17.59L6.41,19L12,13.41L17.59,19L19,17.59L13.41,12L19,6.41Z"></path></g></svg></ha-svg-icon></slot></span></button></mwc-icon-button></ha-icon-button></h2><div id="content" class="mdc-dialog__content" style="-webkit-font-smoothing: antialiased; font-family: var(--mdc-typography-body1-font-family,var(--mdc-typography-font-family,Roboto,sans-serif)); font-size: 14px; line-height: var(--mdc-typography-body1-line-height,1.5rem); font-weight: 400; letter-spacing: var(--mdc-typography-body1-letter-spacing,.03125em); text-decoration-line: ; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; text-transform: none; flex-grow: 1; box-sizing: border-box; margin: 0px; overflow: auto; color: rgb(255, 255, 255); padding: var(--dialog-content-padding,24px); position: var(--dialog-content-position,relative); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; orphans: 2; text-align: start; text-indent: 0px; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal;"><slot id="contentSlot" style="margin-top: 0px; margin-bottom: 0px;"><div><div>

Versione | core-2023.9.3
-- | --
Tipo di installazione | Home Assistant OS
Sviluppo | false
Supervisor | true
Docker | true
Utente | root
Ambiente virtuale | false
Versione Python | 3.11.5
Famiglia del Sistema Operativo | Linux
Versione del Sistema Operativo | 6.1.21-v8
Architettura della CPU | aarch64
Fuso orario | Europe/Rome
Cartella di configurazione | /config

</div></div></slot></div>
</body>
</html>`

`<html>
<body>
<div class="card-header" style="justify-content: space-between; display: flex; align-items: center; color: rgb(255, 255, 255); font-family: Roboto, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: 0.4375px; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgba(25, 25, 25, 0.7); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><h3>Home Assistant Community Store</h3><a class="manage" href="http://192.168.107.100:8123/hacs" style="color: var(--primary-color); text-decoration: none;"><mwc-button style="display: inline-flex; outline: 0px; -webkit-tap-highlight-color: transparent; vertical-align: top;"><button id="button" class="mdc-button   " aria-label="" style="-webkit-font-smoothing: antialiased; font-family: var(--mdc-typography-button-font-family,var(--mdc-typography-font-family,Roboto,sans-serif)); font-size: var(--mdc-typography-button-font-size,.875rem); line-height: inherit; font-weight: var(--mdc-typography-button-font-weight,500); letter-spacing: var(--mdc-typography-button-letter-spacing,.0892857143em); text-decoration: var(--mdc-typography-button-text-decoration,none); text-transform: var(--mdc-typography-button-text [... cut ...] utton__label" style="position: relative;"></span><span class="slot-container   " style="display: inline-flex; align-items: center; justify-content: center;"><slot>GESTISCI</slot></span><span class="trailing-icon"><slot name="trailingIcon"></slot></span></button></mwc-button></a></div>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4625
Installed Version | 1.33.0
Stage | running
Available Repositories | 1301
Downloaded Repositories | 28

</body>
</html>`

koersie, 2023-09-25:
Had the same issue. Already reverted back to last nights backup so unable to pull up the logs.

andlo, 2023-09-25:
Same issue for me - havnt reverted back to backup yet  hoping for a quickie solution :)
Full log is atatched 
[home-assistant_2023-09-25T12-15-59.667Z.log](https://github.com/lbbrhzn/ocpp/files/12714893/home-assistant_2023-09-25T12-15-59.667Z.log)

javiermitchell, 2023-09-25:
Same issue with my System

oekinger, 2023-09-25:
Same for me

philgra, 2023-09-25:
Same for me

dreimer1986, 2023-09-25:
Wanna bet that @lbbrhzn and @drc38 already understood that this is a problem for everyone who updated HA?!

gburtscher, 2023-09-25:
same at 9.3

drc38, 2023-09-25:
It's an issue with an upstream library https://github.com/mobilityhouse/ocpp/issues/471

drc38, 2023-09-25:
If anyone's able to try adding `jsonschema==4.19.0` to the manifest file and see if that works as a temporary fix

robinsmidsrod, 2023-09-25:
@drc38 I tried adding `jsonschema==4.19.0` to manifest.json, and then I didn't get any errors, but I only got this in the log:

```
2023-09-25 21:03:06.725 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration ocpp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-09-25 21:03:31.101 INFO (MainThread) [custom_components.ocpp] config = {}
```

drc38, 2023-09-25:
@robinsmidsrod try unloading and reloading ocpp integration

victors21, 2023-09-25:
@drc38 
redownloaded ocpp extension then add jsonschema==4.19.0 to manifest and then add integration. now all works for me as before, thanks.

vetranolorenzo, 2023-09-25:
same problem for me too, just updated it no longer loads the ocpp integration

victors21, 2023-09-25:
> same problem for me too, just updated it no longer loads the ocpp integration

remove it the make same as me here 

> @drc38 redownloaded ocpp extension then add jsonschema==4.19.0 to manifest and then add integration. now all works for me as before, thanks.

mpch13, 2023-09-25:
you guys mean set the jsonschema version in the manifest like this?

```
"requirements": [ 
     "occp>=0.14.1",
      "websockets>=10.2",
      "jsonschema==4.19.0"
],
"version": "v0.4.40"
```

victors21, 2023-09-25:
> you guys mean set the jsonschema version in the manifest like this?
> 
> ```
> "requirements": [ 
>      "occp>=0.14.1",
>       "websockets>=10.2",
>       "jsonschema==4.19.0"
> ],
> "version": "v0.4.40"
> ```

thats it

mpch13, 2023-09-25:
> > you guys mean set the jsonschema version in the manifest like this?
> > ```
> > "requirements": [ 
> >      "occp>=0.14.1",
> >       "websockets>=10.2",
> >       "jsonschema==4.19.0"
> > ],
> > "version": "v0.4.40"
> > ```
> 
> thats it

I confirm it works, remenber de **colon** at the end of websockets if you add the new jsonschema version after it.

vetranolorenzo, 2023-09-25:
> > > you guys mean set the jsonschema version in the manifest like this?
> > > ```
> > > "requirements": [ 
> > >      "occp>=0.14.1",
> > >       "websockets>=10.2",
> > >       "jsonschema==4.19.0"
> > > ],
> > > "version": "v0.4.40"
> > > ```
> > 
> > 
> > thats it
> 
> I confirm it works, remember de **colon** at the end of websockets if you add the new jsonschema version after it.

It doesn't work for me, it doesn't even allow me to delete the integration....

victors21, 2023-09-25:
> It doesn't work for me, it doesn't even allow me to delete the integration....

just remove addon in hacs presing ignore integration installed

vetranolorenzo, 2023-09-25:
> just remove addon in hacs presing ignore integration installed

forgive me, but I can't understand how to uninstall the integration, should I delete the folder in custom component?

robinsmidsrod, 2023-09-25:
> @robinsmidsrod try unloading and reloading ocpp integration

Got it working with the schema change. Turns out it just took a while to refresh data after restart.

vetranolorenzo, 2023-09-25:
> > @robinsmidsrodprova a scaricare e ricaricare l'integrazione ocpp
> 
> Funziona con la modifica dello schema. Risulta che ci è voluto solo un po' di tempo per aggiornare i dati dopo il riavvio.

How long did you wait? I've been rebooting and updating for a while now and it can't load the integration

victors21, 2023-09-25:
> > > @robinsmidsrodprova a scaricare e ricaricare l'integrazione ocpp
> > 
> > 
> > Funziona con la modifica dello schema. Risulta che ci è voluto solo un po' di tempo per aggiornare i dati dopo il riavvio.
> 
> How long did you wait? I've been rebooting and updating for a while now and it can't load the integration

Just go to HACS and remove OCPP it's custom integration

lbbrhzn, 2023-09-25:
Created a new release  v0.4.41 with the workaround, pinning jsonschema to 4.19.0.

lbbrhzn, 2023-09-25:
Can someone please confirm that the release v0.4.41 solves the issue?

jrlegg, 2023-09-25:
> Can someone please confirm that the release v0.4.41 solves the issue?

I've just updated and can confirm its now working... Thankyou.

oekinger, 2023-09-26:
Works like a charm again :-D. Thank you.

dragorex71, 2023-09-26:
> Can someone please confirm that the release v0.4.41 solves the issue?

Confirm!

vetranolorenzo, 2023-09-26:
> Creata una nuova versione v0.4.41 con la soluzione alternativa, bloccando jsonschema su 4.19.0.

so do I have to delete the jsonshema 4.19 line?I updated to version 4.41 but it gives me an error in the configuration even though at least hacs gives me the possibility to configure it

koersie, 2023-09-26:
I did the HA upgrade again today. Followed by the occp upgrade through hacs.
All working fine now.

vetranolorenzo, 2023-09-26:
@koersie in various attempts yesterday I deleted the integration and reinstalled it. and now I can't configure it anymore

closed 2023-12-04 by github-actions[bot]