lbbrhzn/ocpp #918: Charge Control not working with EVLink Wallbox

stibbzy, 2023-09-25:
I have an issue whereby my EVLink Wallbox is rejecting Charge Control turn on requests after a turn off request has been issued. I can successfully issue a turn off request using the UI toggle switch, and the charger will stop charging. However, if I then toggle the switch on, it switches off again and I get this error in the log:

2023-09-26 09:18:58.585 INFO (MainThread) [ocpp] b41725: send [2,"0cf4db24-978b-4d1b-9e01-82988679b1ab","GetConfiguration",{"key":["AuthorizeRemoteTxRequests"]}]
2023-09-26 09:18:58.794 INFO (MainThread) [ocpp] b41725: receive message [3,"0cf4db24-978b-4d1b-9e01-82988679b1ab",{"configurationKey":[{"key":"AuthorizeRemoteTxRequests","readonly":true,"value":"false"}]}]
2023-09-26 09:18:58.796 INFO (MainThread) [ocpp] b41725: send [2,"5e2f5bbb-c7ef-4327-80d3-15b126847da9","RemoteStartTransaction",{"idTag":"b41725","connectorId":1}]
2023-09-26 09:18:58.821 INFO (MainThread) [ocpp] b41725: receive message [3,"5e2f5bbb-c7ef-4327-80d3-15b126847da9",{"status":"Rejected"}]
2023-09-26 09:18:58.821 WARNING (MainThread) [custom_components.ocpp] Failed with response: Rejected

The light on the charging station will also turn red. However, if I unplug the cable, the light turns green. I can then toggle the charge control on, and it works as expected, until another stop request is issued followed by a start request.

I've configured the charger to allow remote start etc, but given I can start it remotely after unplugging it, authentication doesn't seem to be the issue. It's almost like the resistor network in the charger needs to detect an open connection before allowing another start request.

This is the charger log:

Mon Sep 25 23:18:57 2023 Received RemoteStartTransaction _connectorId=1 _idTag=b41725
Mon Sep 25 23:18:57 2023 Sent RemoteStartTransactionConf: Rejected
Mon Sep 25 23:19:21 2023 Received ChangeAvailability connectorId=0 type=Operative
Mon Sep 25 23:19:21 2023 Sent ChangeAvailabilityConf: Accepted
Mon Sep 25 23:19:25 2023 Sent GetConfigurationConf configurationKeys: 1
Mon Sep 25 23:19:25 2023 Received RemoteStartTransaction _connectorId=1 _idTag=b41725
Mon Sep 25 23:19:25 2023 Sent RemoteStartTransactionConf: Rejected
Mon Sep 25 23:20:21 2023 Received ChangeAvailability connectorId=0 type=Operative
Mon Sep 25 23:20:21 2023 Sent ChangeAvailabilityConf: Rejected
Mon Sep 25 23:21:01 2023 Sending StatusNotificationReq: connectorld=1 Available
Mon Sep 25 23:21:01 2023 Received StatusNotificationConf
Mon Sep 25 23:21:20 2023 Sending StatusNotificationReq: connectorld=1 Preparing
Mon Sep 25 23:21:20 2023 Received StatusNotificationConf
Mon Sep 25 23:21:28 2023 Sent GetConfigurationConf configurationKeys: 1
Mon Sep 25 23:21:28 2023 Received RemoteStartTransaction _connectorId=1 _idTag=b41725
Mon Sep 25 23:21:28 2023 Sent RemoteStartTransactionConf: Accepted Mon Sep 25 23:21:31 2023 Sending StartTransactionReq tag:b41725
Mon Sep 25 23:21:31 2023 Received StartTransactionConf transactionId: 1695684092 status: Accepted
Mon Sep 25 23:21:33 2023 Sending StatusNotificationReq: connectorld=1 Charging
Mon Sep 25 23:21:33 2023 Received StatusNotificationConf

Version of the custom_component: v0.4.41

## Configuration

# Evse Conf

[FUNCTIONAL]
operatingMode=           1
clusterManagement=       0
simplifiedMode3=         true
ventilation=             false
localisation=            false
stateFDelayPwmOff=       0
stateFDelayPwmOn=        3
evChargeTimeout=         30
cpwRateStep=             0
isolatedInput1=          0
isolatedInput2=          0
diStatusExpected=        0x0002
diStatusEnable=          0x2602
autoTestEnable=          0x7FFFFFFF
SM3CurrentLowThreshold=  1
SM3MaxCurrent=           10
SM3StopDelay=            3600
postponeCharge=          false
allowPluggedCable=       true
enableEvDetection=       false
schukoWithNoDetectionEndOfCharge=       60
updateSetPointPeriodInSec=       15
remoteSetPointPeriodInSec=       5
stationNetworkType=       3
dhcp=                     1
strictDhcpMode=           0
slaveHasPublicIp=         1
stationName=              N 20265 20 007 003
ssdp=                     1
[... 194 lines cut ...]
port=8000

# Update Conf

[UPDATE_GLOBAL]
GlobalFwUpdateProtocol=0
DefaultPackageName=evse

[UPDATE_HTTP]
HttpLogin=
HttpPassword=2F94F0C34B68214E48D66C08D6893B2B
HttpServerAddr=192.168.0.101
HttpPort=80
HttpPath=

[UPDATE_FTP]
FtpLogin=
FtpPassword=2F94F0C34B68214E48D66C08D6893B2B
FtpServerAddr=
FtpPort=21
FtpPath=

[UPDATE_TFTP]
TftpServerAddr=
TftpPath=

drc38, 2023-09-26:
I have a different charger but it works in the same way as yours, only option seems to be to manually unplug or reset. Instead I simply set max charge rate to 0 rather than use switch.

stibbzy, 2023-10-05:
> I have a different charger but it works in the same way as yours, only option seems to be to manually unplug or reset. Instead I simply set max charge rate to 0 rather than use switch.

Yeah I figured out the same after lots of trial and error. The issue with that is that it makes it hard to integrate as a custom charger into evcc.io. Evcc expects to be able to issue a charger enable/disable, and while I can automate a disable command to '0' amps, I can't automate an enable command to a variable charge amps.

drc38, 2023-10-05:
evcc.io has its own ocpp interface, what happens when you use it directly rather than through HA?

stibbzy, 2023-10-05:
Unfortunately, it doesn’t appear to conform to OCPP 1.6 perfectly. It has the same issue as home assistant when evcc tries to set the charger to off. This is when it’s set up under an OCPP template in evcc. 
Evcc does have a dedicated EVLink integration, but it requires a sponsor token.

closed 2024-01-05 by github-actions[bot]