lbbrhzn/ocpp #928: Unable to set maximum current

markusgillmeister, 2023-10-03:
**Describe the bug**
At the current version of Homeassistant (2023.9.3) I'm unable to change the "charger_maximum_current" via input control / slider in the ui - nothing happens. The slider does not set a new value or trigger a message to the wallbox (debug logging does not show anything / any error message).
It worked in earlier versions of HA but unfortunately I cannot say at which version it stopped working. (I do not change the maximum ampere of the wallbox very often). 

Even with an automation script I am not able to set a new value anymore.  If I'm using the entity id "number.charger_maximum_current" from the ocpp integration I'll get an error message in the log "Referenced entities number.charger_maximum_current are missing or not currently available".  (and yes its available, I can see the value and slider in the ui).   
Below here is the sample automation test which does not work:
```
  - service: input_number.set_value
    data:
      value: "{{ states('input_number.ladestrom') | int }}"
    target:
      entity_id: number.charger_maximum_current
```

If I use the automation UI to create some action on the "maximum current" I'll get something like
```
...
  - device_id: 74091774572966d60c609f191ba3bb1e
    domain: number
    entity_id: 2f434d8011c3f567e7b51c891b5d5158
...
``` 
which results also in an error message about an unknown entity. 

The only way (and thats curious) to set the "number.charger_maximum_current" is via development tools menu and setting a status manually. If I change this value, the value is set and the sliders also reflect that change in the ui. 

Any ideas what could possibly the problem or changed in the last versions of HA so that the number-field from the ocpp integration does not work anymore?

**To Reproduce**
Don't know if its possible to reproduce it for you guys. 

I use the latest HA and OCPP integration together with an Alfen Eve Pro wallbox which is configured for OCPP 1.6.  Wallbox has firmware 6.4.0-4210 (latest). As mentioned above it worked some days before perfectly. 

Best Regards
Markus

drc38, 2023-10-06:
Sounds like an issue in HA itself if it is reporting the entity is missing. Not ideal, but suggest try removing the ocpp integration, restarting HA and reinstalling again.

markusgillmeister, 2023-10-21:
Sorry for the delay.  I did remove the integration, restarted HA , reinstalled/reconfigured the charger.  But that does not help - setting is impossible.  
Setting "number.charger_maximum_current" via development tools does also not work (for some seconds the new value is visible but then resets to the old value". 
Any ideas how to debug the ocpp connection / checking the request/responses ?

lbbrhzn, 2023-10-21:
For debugging, see https://home-assistant-ocpp.readthedocs.io/en/latest/debugging.html.

closed 2023-12-29 by github-actions[bot]