lbbrhzn/ocpp #952: Automtically detect supported measurands

bardahlm, 2023-10-26:
**Background**
I have set up OCPP with my Schneider Electric EVlink Smart Wallbox for years. When installing the latest firmware, 3.4.0.7, I noticed that the charge box now support **Power.Active.Import**.  I need that measurand for my automation so I wanted to enable it.

**Problem**
It is tricky to enable measurands, because if you select one that is unsupported you get none. You then have to delete the integration and try again untill you get what you want. You can't change measurands on an already configured integration.

**Possible solution**
One solution could be to have the integration loop over all possible measurements, one by one, and present the working ones to the user. This could be run now and then and present new found measurands to the user. It would also be great if measurands could be changed without deleting the integration.

lbbrhzn, 2023-11-04:
Some integrations create components (sensors, switches, etc) that are disabled by default. 
The user can then enable the components they want to see by going to the device panel. 
I think this could be a nice way to allow measurands to be changed.

lbbrhzn, 2023-11-04:
Autodetection of the supported measurands would be nice feature, but is a bit tricky to implement. 

There is no message to request a list of supported measurands supported by the charger. The central management system just sends a list of measurands that must be reported periodically by the charger in a ChangeConfiguration request.  If the list contains measurands that are not supported, then the charger will simply reject the request (without indicating which measurand is not supported). Hence, auto detecting the list of supported measurands requires trial and error. Checking each measurand separately will take some time. To reduce the number of messages required, we could start with a full list and split it in two each time we get a rejection.

closed 2023-11-10 by lbbrhzn