lbbrhzn/ocpp #987: TypeConstraintViolationError - the message is too long

mazzy89, 2023-11-26:
## Version of the custom_component

v0.5.0

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

I get this error message complaining that the message is too long 

```
User didn\'t disconnect vehicle within 00:10:00 minutes after charging has ended." is too long'
```

## Debug log

```text

  File "/usr/local/lib/python3.11/site-packages/ocpp/messages.py", line 226, in validate_payload
  File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 186, in route_message
  File "/config/custom_components/ocpp/api.py", line 944, in _handle_call
  File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 214, in _handle_call
    validate_payload(msg, self._ocpp_version)
  File "/usr/local/lib/python3.11/site-packages/ocpp/messages.py", line 240, in validate_payload
ocpp.exceptions.TypeConstraintViolationError: TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': '"User didn\'t disconnect vehicle within 00:10:00 minutes after charging has ended." is too long', 'ocpp_message': <Call - unique_id=24, action=StatusNotification, payload={'connectorId': 1, 'status': 'Finishing', 'errorCode': 'NoError', 'info': "User didn't disconnect vehicle within 00:10:00 minutes after charging has ended.", 'timestamp': '2023-11-26T13:30:31.573Z', 'vendorId': 'Etrel', 'vendorErrorCode': 'UserDidntUnplugVehicleAfterCharging'}>}
```

What do you recommend here? To wire a PR and have accepted this length (not sure whether this would be possible OR disable validation?

mazzy89, 2023-11-26:
Actually I've found in my CS an option to shrink length of the messages

lbbrhzn, 2023-11-26:
The error is due to strict OCPP validation. You can select to disable it the configuration.

mazzy89, 2023-11-26:
@lbbrhzn an error like that would prevent to register session time? I'm wondering why I cannot get session time registered. In my case I've put a debug clause after and it did not get executed.