lbbrhzn/ocpp #994: OCPP network connection to my ABB Terra box not connecting

maxcim32, 2023-12-02:
Trying to establish OCPP connection (using OCPP Integration) to my ABB Terra Charger (ABB Terra AC W7-G5-R-O) but each time I enter the IP of my OCPP server (HA) in the charger config the charger drops the network connection. If I leave the default OCPP connection to the ABB server the network connection is stable (tried both wired and wireless connections). Using the ABB TerraConfig app (version 2.0.8.150) to enter the IP address of the OCPP Server (my HA server is ASUSTOR NAS box with Linux VM for HA).  All firmware is latest on all devices and talking to ABB technical support they say there is nothing wrong with the Terra box or the OCPP firmware.  

I know this should work but just can establish a connection. Suspicious that it could be something to do with HA running on the VM on the NAS box.

A beginner with HA but I am slowly learning.

Log entry created when I try to connect to the charger using the OCPP Integration

Logger: homeassistant.config_entries
Source: config_entries.py:402
First occurred: 11:26:18 (1 occurrences)
Last logged: 11:26:18

Error setting up entry central for ocpp
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 402, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/__init__.py", line 70, in async_setup_entry
    central_sys = await CentralSystem.create(hass, entry)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 194, in create
    server = await websockets.server.serve(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/websockets/legacy/server.py", line 1127, in __await_impl__
    server = await self._create_server()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1525, in create_server
    raise OSError(err.errno, 'error while attempting '
OSError: [Errno 98] error while attempting to bind on address ('0.0.0.0', 9000): address in use

Any suggestions.

kovlo, 2023-12-16:
For my Terra I set the OCPP in TerraConfig: ws://myhomeassistantIP:9000
As far as I remember I set up the host address as the manual said: 0.0.0.0, without https
The charger and HA are bot in the same VLAN and both have a static IP. The HA runs in a VM as well.

What services do you use in your HA installation? Check the port allocation, the last line of your log might show something related to this.

maxcim32, 2023-12-20:
Thanks for your input, sorry for the delay in responding, have been out for work.  By services do you mean what integrations I am running ? Still looking into a potential port allocation problem, calling on a friend to help me look further into that.

Danton73, 2024-02-03:
> For my Terra I set the OCPP in TerraConfig: ws://myhomeassistantIP:9000 As far as I remember I set up the host address as the manual said: 0.0.0.0, without https The charger and HA are bot in the [...]

Thank you.. It's 3 days that i try to communicate with my Terra AC in HA and by follow your instruction now work.
Just one think: the voltage shown is multiplicate to 10 (I watch 2199 V).. do you know why?

kovlo, 2024-02-04:
> the voltage shown is multiplicate to 10 (I watch 2199 V).. do you know why?

No, in my setup I always had correct values.

If If you connect to the charger with the mobile app (Charger Sync) via Bluetooth (wifi does not work for me since I enabled my own ocpp setup), do you see proper values?

Danton73, 2024-02-05:
> > the voltage shown is multiplicate to 10 (I watch 2199 V).. do you know why?
> 
> No, in my setup I always had correct values.
> 
> If If you connect to the charger with the mobile app (Charger Sync) via Bluetooth (wifi does not work for me since I enabled my own ocpp setup), do you see proper values?

Yes, I see the proper values via Bluetooth..
Is someting strange, because when the charger didn't work (like now) show the wrong voltage but when I charge the car the voltage is correct.
The picture below is when the charger is off

[image]

janrothkegel, 2024-02-13:
@maxcim32 Have you tried setting up your config in the TerraConfig portal instead of the app? Or does it work now?

Danton73, 2024-02-13:
> @maxcim32 Have you tried setting up your config in the TerraConfig portal instead of the app? Or does it work now?

is there any special setting in terraconfig app?
I noticed that when the device is not connected or charging, it reports an incorrect voltage (multiplied x10) but reports it correct when charging

janrothkegel, 2024-02-13:
@Danton73 At this point in time my ABB Terra is not connected to my car and it reports 229.7 V. Have you checked the raw communication data by [enabling both loggings](https://github.com/lbbrhzn/ocpp/wiki/debugging) (ocpp and websocket)?

Danton73, 2024-02-15:
> @Danton73 At this point in time my ABB Terra is not connected to my car and it reports 229.7 V. Have you checked the raw communication data by [enabling both loggings](https://github.com/lbbrhzn/ocpp/wiki/debugging) (ocpp and websocket)?

I activated both loggings, but honestly I don't understand much about them.
I attach the registry log, I just deleted the number of my device by putting "xxx"

[OCPP_LOG_Report.txt](https://github.com/lbbrhzn/ocpp/files/14292810/OCPP_LOG_Report.txt)

janrothkegel, 2024-02-16:
@Danton73 I am also not an expert regarding OCPP. But let me share my findings.

In my log the voltage is missing at all when not charging:
2024-02-16 15:15:00.353 INFO (MainThread) [ocpp] TAxxxxxxxxxx: receive message [2, "2409998", "MeterValues", {"connectorId": 1, "meterValue": [{"timestamp": "2024-02-16T14:15:00.000Z", "sampledValue": [{"value": "2008322", "context": "Sample.Clock", "format": "Raw", "measurand": "Energy.Active.Import.Register", "location": "Inlet"}]}]}]

From your log I can see that you are using firmware version 1.6.9. Maybe it makes more sense to compare our ABB Terras with equal firmware. So could you maybe upgrade to 1.6.91 via the TerraConfig app?

Danton73, 2024-02-20:
@janrothkegel Thanks for the info.
In these days I updated the firmware two time: first time at 1.7.0, second time at 1.8.0 but noting is change. 
Same problem. Do you need an upgrade of the register?

closed 2024-04-28 by github-actions[bot]

Penbanc, 2024-08-29:
Any advice about how the 'port issue' was resolved ? Think I have the same issue with a different charger (Simpson Partners)