lbbrhzn/ocpp #1015: Meter values are not selected to send after HA rebooting

mazzy89, 2023-12-14:
## Version of the custom_component

0.5

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

In my charge point, I have the following menu

[image]

where I can select the meter values that can be sent to the backend system to collect data

I select them and click save. However, when home assistant is rebooted (and OCPP integration is rebooted) the checkboxes are no longer ticked.

## Debug log

```text

2023-12-14 23:49:03.610 INFO (MainThread) [homeassistant.setup] Setup of domain ocpp took 0.0 seconds
2023-12-14 23:49:03.810 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.ocpp
2023-12-14 23:49:06.225 INFO (MainThread) [homeassistant.components.switch] Setting up switch.ocpp
2023-12-14 23:49:06.235 INFO (MainThread) [homeassistant.components.number] Setting up number.ocpp
2023-12-14 23:49:12.532 INFO (MainThread) [homeassistant.components.button] Setting up button.ocpp
2023-12-14 23:49:16.608 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2023-12-14 23:49:16.608 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/charger
2023-12-14 23:49:16.608 INFO (MainThread) [custom_components.ocpp] Charger charger connected to 0.0.0.0:9000.
2023-12-14 23:49:18.614 INFO (MainThread) [ocpp] charger: send [2,"edb781c6-ccb0-4b89-8712-4e510c4d5708","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2023-12-14 23:49:18.823 INFO (MainThread) [ocpp] charger: receive message [3,"edb781c6-ccb0-4b89-8712-4e510c4d5708",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger"}],"unknownKey":[]}]
2023-12-14 23:49:18.832 DEBUG (MainThread) [custom_components.ocpp] Feature profiles returned: Profiles.CORE|FW|SMART|RES|REM|AUTH
2023-12-14 23:49:18.835 INFO (MainThread) [ocpp] charger: send [2,"5520aa13-805d-4198-8e82-d2b7ecf2f409","GetConfiguration",{"key":["NumberOfConnectors"]}]
2023-12-14 23:49:18.843 INFO (MainThread) [ocpp] charger: receive message [3,"5520aa13-805d-4198-8e82-d2b7ecf2f409",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}],"unknownKey":[]}]
2023-12-14 23:49:18.845 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for NumberOfConnectors: 1
2023-12-14 23:49:18.847 INFO (MainThread) [ocpp] charger: send [2,"282f3c92-4759-4cbe-8f52-3dad8bc8335e","GetConfiguration",{"key":["HeartbeatInterval"]}]
2023-12-14 23:49:18.854 INFO (MainThread) [ocpp] charger: receive message [3,"282f3c92-4759-4cbe-8f52-3dad8bc8335e",{"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"1800"}],"unknownKey":[]}]
2023-12-14 23:49:18.856 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for HeartbeatInterval: 1800
2023-12-14 23:49:18.857 DEBUG (MainThread) [custom_components.ocpp] 'charger' trying measurand 'Current.Export'
2023-12-14 23:49:18.858 INFO (MainThread) [ocpp] charger: send [2,"96640980-cfca-4c2d-8f68-201442b285d9","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Export"}]
2023-12-14 23:49:20.331 INFO (MainThread) [ocpp] charger: receive message [3,"96640980-cfca-4c2d-8f68-201442b285d9",{"status":"Rejected"}]
2023-12-14 23:49:20.345 DEBUG (MainThread) [custom_components.ocpp] 'charger' trying measurand 'Current.Import'
2023-12-14 23:49:20.345 INFO (MainThread) [ocpp] charger: send [2,"d1f13bfd-3543-4c0d-ad1b-291640f1cd7b","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
2023-12-14 23:49:20.505 INFO (MainThread) [ocpp] charger: receive message [3,"d1f13bfd-3543-4c0d-ad1b-291640f1cd7b",{"status":"Rejected"}]
2023-12-14 23:49:20.511 DEBUG (MainThread) [custom_components.ocpp] 'charger' trying measurand 'Current.Offered'
2023-12-14 23:49:20.511 INFO (MainThread) [ocpp] charger: send [2,"82358933-1d09-4483-859c-e8e2c7765ad4","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
2023-12-14 23:49:20.685 INFO (MainThread) [ocpp] charger: receive message [3,"82358933-1d09-4483-859c-e8e2c7765ad4",{"status":"RebootRequired"}]
2023-12-14 23:49:20.689 DEBUG (MainThread) [custom_components.ocpp] 'charger' trying measurand 'Energy.Active.Export.Interval'
2023-12-14 23:49:20.690 INFO (MainThread) [ocpp] charger: send [2,"9df2f0da-50ac-4c32-a777-5db607a47fb4","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Export.Interval"}]
2023-12-14 23:49:20.763 INFO (MainThread) [ocpp] charger: receive message [3,"9df2f0da-50ac-4c32-a777-5db607a47fb4",{"status":"Rejected"}]
2023-12-14 23:49:20.765 DEBUG (MainThread) [custom_components.ocpp] 'charger' trying measurand 'Energy.Active.Export.Register'
2023-12-14 23:49:20.766 INFO (MainThread) [ocpp] charger: send [2,"f8b07251-89d2-41ce-9863-0416722d3cce","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Export.Register"}]
2023-12-14 23:49:20.832 INFO (MainThread) [ocpp] charger: receive message [3,"f8b07251-89d2-41ce-9863-0416722d3cce",{"status":"Rejected"}]
2023-12-14 23:49:20.833 DEBUG (MainThread) [custom_components.ocpp] 'charger' trying measurand 'Energy.Active.Import.Interval'
2023-12-14 23:49:20.834 INFO (MainThread) [ocpp] charger: send [2,"93b6aae7-a150-4677-a02d-dd186cd9e476","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Interval"}]
2023-12-14 23:49:20.894 INFO (MainThread) [ocpp] charger: receive message [3,"93b6aae7-a150-4677-a02d-dd186cd9e476",{"status":"Rejected"}]
2023-12-14 23:49:20.896 DEBUG (MainThread) [custom_components.ocpp] 'charger' trying measurand 'Energy.Active.Import.Register'
2023-12-14 23:49:20.897 INFO (MainThread) [ocpp] charger: send [2,"e2b0cc30-2e67-4511-8f24-b5c0d799d2ff","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
2023-12-14 23:49:20.995 INFO (MainThread) [ocpp] charger: receive message [3,"e2b0cc30-2e67-4511-8f24-b5c0d799d2ff",{"status":"RebootRequired"}]
2023-12-14 23:49:20.996 DEBUG (MainThread) [custom_components.ocpp] 'charger' trying measurand 'Energy.Reactive.Export.Interval'
2023-12-14 23:49:20.997 INFO (MainThread) [ocpp] charger: send [2,"22e36f24-708a-4303-b58f-411684d7cb51","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Export.Interval"}]
2023-12-14 23:49:21.084 INFO (MainThread) [ocpp] charger: receive message [3,"22e36f24-708a-4303-b58f-411684d7cb51",{"status":"Rejected"}]
2023-12-14 23:49:21.085 DEBUG (MainThread) [custom_components.ocpp] 'charger' trying measurand 'Energy.Reactive.Export.Register'
2023-12-14 23:49:21.086 INFO (MainThread) [ocpp] charger: send [2,"40f1a282-bc32-47f5-b9f7-a4e6487bb34f","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Export.Register"}]
2023-12-14 23:49:21.142 INFO (MainThread) [ocpp] charger: receive message [3,"40f1a282-bc32-47f5-b9f7-a4e6487bb34f",{"status":"Rejected"}]
2023-12-14 23:49:21.144 DEBUG (MainThread) [custom_components.ocpp] 'charger' trying measurand 'Energy.Reactive.Import.Interval'
2023-12-14 23:49:21.145 INFO (MainThread) [ocpp] charger: send [2,"a8d27cbb-272b-4838-b1b6-550d9b1ddadd","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Import.Interval"}]
[... 41 lines cut ...]
2023-12-14 23:49:22.412 INFO (MainThread) [ocpp] charger: receive message [3,"73684b93-5c22-4575-883d-c9fa57877f2c",{"status":"RebootRequired"}]
2023-12-14 23:49:22.414 INFO (MainThread) [ocpp] charger: send [2,"2fbf815e-836e-4c51-a0cf-83615dfb2295","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2023-12-14 23:49:22.420 INFO (MainThread) [ocpp] charger: receive message [3,"2fbf815e-836e-4c51-a0cf-83615dfb2295",{"configurationKey":[{"key":"MeterValueSampleInterval","readonly":false,"value":"10"}],"unknownKey":[]}]
2023-12-14 23:49:22.422 INFO (MainThread) [ocpp] charger: send [2,"b3760749-f6ca-4653-8d23-056e237501b5","GetConfiguration",{"key":["ClockAlignedDataInterval"]}]
2023-12-14 23:49:22.430 INFO (MainThread) [ocpp] charger: receive message [3,"b3760749-f6ca-4653-8d23-056e237501b5",{"configurationKey":[{"key":"ClockAlignedDataInterval","readonly":false,"value":"300"}],"unknownKey":[]}]
2023-12-14 23:49:22.433 DEBUG (MainThread) [custom_components.ocpp] 'charger' post connection setup completed successfully
2023-12-14 23:49:22.445 INFO (MainThread) [ocpp] charger: send [2,"0977b55b-d905-436a-ad2e-296258552a87","GetConfiguration",{"key":["WebSocketPingInterval"]}]
[... cut ...]
2023-12-14 23:49:22.977 INFO (MainThread) [ocpp] charger: receive message [2,"2","StatusNotification",{"connectorId":1,"status":"Available","errorCode":"NoError","info":"TriggerMessage","timestamp":"2023-12-14T22:49:22.951Z","vendorId":"Etrel"}]
2023-12-14 23:49:22.979 INFO (MainThread) [ocpp] charger: send [3,"2",{}]
2023-12-14 23:49:22.999 INFO (MainThread) [ocpp] charger: receive message [2,"3","StatusNotification",{"connectorId":0,"status":"Available","errorCode":"NoError","info":"","timestamp":"2023-12-14T22:49:22.986Z","vendorId":"Etrel","vendorErrorCode":"ConnectionWithCentralSystemRestored"}]
2023-12-14 23:49:23.001 INFO (MainThread) [ocpp] charger: send [3,"3",{}]
```

mazzy89, 2023-12-14:
What is not clear to me is why the measurands are all `Rejected`.

mazzy89, 2023-12-14:
According to the OCPP 1.6 standard the `Rejected` response in the `ConfigurationStatus` means 

> Configuration key supported, but setting could not be changed. 

Any inputs why is not possible to be changed?

drc38, 2023-12-14:
Try a version before 0.5 where you could manually select the measurands to match your charger menu and see if that works. The integration will try setting the selected measurands in 1 message rather than individually.

mazzy89, 2023-12-15:
I've changed the measurands from `.storage` config entries defining the one supported by my charger. I did not get `Rejected` for any the measurands set but still in the GUI they do not appear selected. I've rebooted both HA and my charger.

drc38, 2023-12-15:
Ok, what is the issue the integration? It seems the issue is with the charger GUI from your description above.

mazzy89, 2023-12-15:
I set the meter values in the GUI, reboot the station, and the values are still there checked. But when I reboot HA and the integration reboot as well, the values are not checked anymore. A reboot of integration wipes out the set meter values in the charger. I do not get how it is possible.

drc38, 2023-12-15:
Sounds like a question for the charger manufacturer.

mazzy89, 2023-12-15:
I will reach out to the vendor then. Thanks

mazzy89, 2023-12-15:
Not sure if this helps but I get continuously in my HA the following notification:

```
"A reboot is required to apply MeterValuesSampledData="
```

drc38, 2023-12-15:
That will be coming from your charger, when you reboot it only what happens?

mazzy89, 2023-12-15:
I've run a charging session and I think despite the GUI not aligned, meter values get correctly at the set internal to HA. But the problem is now on the HA that it does not set them.

I set them in the OCPP integration

```
Current.L1,Current.L2,Current.L3,Current.Offered,Energy.Active.Import.Register,Frequency,Power.Active.Total,Power.Active.L1,Power.Active.L2,Power.Active.L3,Power.Factor,Temperature,Voltage.L1,Voltage.L2,Voltage.L3
```

new sensors are created in HA but they remain `Unknown`.

I get these sampled data

```
{
  "sampledValue": [
    {
      "value": "15.728",
      "context": "Sample.Periodic",
      "measurand": "Current.Import",
      "phase": "L2",
      "unit": "A"
    },
    {
      "value": "16.256",
      "context": "Sample.Periodic",
      "measurand": "Current.Import",
      "phase": "L1",
      "unit": "A"
    },
    {
      "value": "16.016",
      "context": "Sample.Periodic",
      "measurand": "Current.Import",
      "phase": "L3",
      "unit": "A"
    },
    {
      "value": "17",
      "context": "Sample.Periodic",
      "measurand": "Current.Offered",
      "unit": "A"
    },
    {
      "value": "74.627432",
      "context": "Sample.Periodic",
      "measurand": "Energy.Active.Import.Register",
      "unit": "kWh"
    },
    {
      "value": "50.03",
      "context": "Sample.Periodic",
      "measurand": "Frequency"
    },
    {
      "value": "11.072",
      "context": "Sample.Periodic",
      "measurand": "Power.Active.Import",
      "unit": "kW"
    },
    {
      "value": "3.744",
      "context": "Sample.Periodic",
      "measurand": "Power.Active.Import",
      "phase": "L1",
      "unit": "kW"
    },
    {
      "value": "3.648",
      "context": "Sample.Periodic",
      "measurand": "Power.Active.Import",
      "phase": "L2",
      "unit": "kW"
    },
    {
      "value": "3.664",
      "context": "Sample.Periodic",
      "measurand": "Power.Active.Import",
      "phase": "L3",
      "unit": "kW"
    },
    {
      "value": "0.975",
      "context": "Sample.Periodic",
      "measurand": "Power.Factor"
    },
    {
      "value": "13",
      "context": "Sample.Periodic",
      "measurand": "Temperature",
      "unit": "Celsius"
    },
    {
      "value": "236.53",
      "context": "Sample.Periodic",
      "measurand": "Voltage",
      "phase": "L1",
      "unit": "V"
    },
    {
      "value": "234.98",
      "context": "Sample.Periodic",
      "measurand": "Voltage",
      "phase": "L3",
      "unit": "V"
    },
    {
      "value": "237.46",
      "context": "Sample.Periodic",
      "measurand": "Voltage",
      "phase": "L2",
      "unit": "V"
    }
  ]
}
```

drc38, 2023-12-15:
Not sure how you're setting them but they need to match the OCPP standard ie `Current.Import` not `Current.L1`

mazzy89, 2023-12-15:
@drc38 they are not called as the OCPP standards. as you from my logs they are rejected https://github.com/lbbrhzn/ocpp/issues/1015#issue-2042630684

when I set them in the OCPP integration 

```
Current.L1,Current.L2,Current.L3,Current.Offered,Energy.Active.Import.Register,Frequency,Power.Active.Total,Power.Active.L1,Power.Active.L2,Power.Active.L3,Power.Factor,Temperature,Voltage.L1,Voltage.L2,Voltage.L3
```

then they are not rejected.

mazzy89, 2023-12-15:
let me make a recap and connect all the dots.

Initially when I set up the OCPP integration leaving the measured the default one, almost them are rejected. As your suggestion I need to match the measurands of my charger.

The measurands have the following name

```
Current.L1,Current.L2,Current.L3,Current.Offered,Energy.Active.Import.Register,Frequency,Power.Active.Total,Power.Active.L1,Power.Active.L2,Power.Active.L3,Power.Factor,Temperature,Voltage.L1,Voltage.L2,Voltage.L3
```

Once I set them up in the OCPP config, they are not anymore rejected. Futher OCPP create new sensors. However some of these sensors as shown remain in `Unknown` state

The meter values that I get from the charger seems to follow the OCPP standards but they are not written because I use different measurands.

mazzy89, 2023-12-15:
Also to add I've noticed that the OCPP integration process correctly the metric values

```
2023-12-15 10:47:58.842 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Current.Import, phase_info: {'unit': 'A', 'L2': 15.936, 'L1': 16.416, 'L3': 16.208} value: 48.560000 unit :A
2023-12-15 10:47:58.842 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Power.Active.Import, phase_info: {'unit': 'kW', 'L1': 3.712, 'L2': 3.696, 'L3': 3.728} value: 11.136000 unit :kW
2023-12-15 10:47:58.842 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Voltage, phase_info: {'unit': 'V', 'L1': 233.24, 'L3': 236.11, 'L2': 237.51} value: 235.620000 unit :V
```

So now the question is which measurands should I define in my case? if i define the default one then they are `Rejected`. If I define the one from the vendor, they are `Accepted` but the sensor remain `Unknown`.

mazzy89, 2023-12-15:
Looking at this past issue https://github.com/lbbrhzn/ocpp/issues/19 I have found that L1,L2,L3 values are reported as attributes regardless of the name of the measurands.

I have then found the right combination of measurands to use 

```
Current.Import,Current.Offered,Energy.Active.Import.Register,Frequency,Power.Active.Import,Power.Factor,Temperature,Voltage
```

for Etrel stations.

mazzy89, 2023-12-15:
I think we can close this because the GUI persistent issue should be an issue of the the vendor. Thanks @drc38

closed 2023-12-15 by mazzy89