lbbrhzn/ocpp #1019: Autel MaxiCharger - Can't modify the amp value

Nidrachp, 2023-12-18:
**Describe the bug**
Type charger : Autel Maxicharger
OCPP information:
Security profile: WS
URL : homeassistant.local
Server path: /ws/webSocket?=sn
Server port: 9000

I can connect my home assistant to the charger. It is possible to switch the charger ON/OFF. Not possible to edit the amp value

I have the message:  [547608280000] PropertyConstraintViolationError: When chargingProfilePurpose =ChargingStationMaxProfile, chargingProfileKind SHALL NOT be Relative, {}

Logs:

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:238
Integration: Home Assistant WebSocket API ([documentation](https://www.home-assistant.io/integrations/websocket_api), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+websocket_api%22))
First occurred: 16:44:32 (2 occurrences)
Last logged: 18:02:50

[547608280000] PropertyConstraintViolationError: When chargingProfilePurpose =ChargingStationMaxProfile, chargingProfileKind SHALL NOT be Relative, {}
[547371048768] PropertyConstraintViolationError: When chargingProfilePurpose =ChargingStationMaxProfile, chargingProfileKind SHALL NOT be Relative, {}
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 238, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2067, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2104, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 272, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 878, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 106, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/ocpp/number.py", line 127, in async_set_native_value
    resp = await self.central_system.set_max_charge_rate_amps(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 287, in set_max_charge_rate_amps
    return await self.charge_points[cp_id].set_charge_rate(limit_amps=value)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 639, in set_charge_rate
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 313, in call
    response = await self._get_specific_response(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 894, in _get_specific_response
    raise resp.to_exception()
ocpp.exceptions.PropertyConstraintViolationError: PropertyConstraintViolationError: When chargingProfilePurpose =ChargingStationMaxProfile, chargingProfileKind SHALL NOT be Relative, {}

Thank you for your help.

Arnaud.

david-j-duran, 2024-08-03:
I have the same issue

shauncraill, 2024-12-12:
I'm still busy with a lot of back and forth emails with Autel, it seems like the issue is they aren't complying to OCPP 100%. They now want to send me a custom OCPP profile for the specific charge current I want to charge with, then I need to apply a different profile each time I want to adjust charge current.

Obviously updating firmware each time you want to adjust current does not make any sense. I will keep this thread updated in the next week about a possible solution. Otherwise I am returning the Autec and installing a Wallox Pulsar

PeterMikeSmooth, 2024-12-15:
Hi there, 

I'm getting the same issue on my Autel MaxiCharger :
`Failed to perform the action number/set_value. PropertyConstraintViolationError: When chargingProfilePurpose =ChargingStationMaxProfile, chargingProfileKind SHALL NOT be Relative, {}`

@lbbrhzn would it possible that Autel MaxiCharger jsut does not support "Relative" chargingProfileKind ? 
I would like to try with "Absolute" chargingProfileKind but I don't see that definition anywhere in the HA OCPP code : 

`petermike@petermike:~/projects/HA/ocpp$ grep -ri ChargingProfileKindType
custom_components/ocpp/ocppv16.py:    ChargingProfileKindType,
custom_components/ocpp/ocppv16.py:                    om.charging_profile_kind.value: ChargingProfileKindType.relative.value,
custom_components/ocpp/ocppv16.py:                    om.charging_profile_kind.value: ChargingProfileKindType.relative.value,
custom_components/ocpp/ocppv201.py:    ChargingProfileKindType,
custom_components/ocpp/ocppv201.py:                    "charging_profile_kind": ChargingProfileKindType.relative.value,
tests/test_charge_point_v201.py:    ChargingProfileKindType,
tests/test_charge_point_v201.py:        "charging_profile_kind": ChargingProfileKindType.relative.value,
tests/test_charge_point_v201.py:        "charging_profile_kind": ChargingProfileKindType.relative.value,
tests/test_charge_point_v201.py:        "charging_profile_kind": ChargingProfileKindType.relative.value,
tests/test_charge_point_v201.py:        "charging_profile_kind": ChargingProfileKindType.relative.value,
`

Any recommandation to make a test with "Absolute" ?

Thank you !

PeterMikeSmooth, 2024-12-15:
> However, the Autel range supports ModbusTCP. 

I tried ModbusTCP, MaxiCharger does answer to just a few registers ...

`:mbpoll -m tcp -r 1 -c 5 192.168.1.33
[1]: 0
[2]: 1
[3]: 100
[4]: 1
[5]: 1

mbpoll -m tcp -r 20001 -c 10 192.168.1.33
[20001]: 0
[20002]: 22000
[20003]: 0
[20004]: 3200
[20005]: 0
[20006]: 22000
[20007]: 0
[20008]: 3200
[20009]: 0
[20010]: 0

mbpoll -m tcp -r 21001 -c 10 192.168.1.33
[21001]: 0
[21002]: 22000
[21003]: 0
[21004]: 3200
[21005]: 0
[21006]: 22000
[21007]: 0
[21008]: 3200
[21009]: 0
[21010]: 0`

And not the one to tune max charging amps. Let us know if you succeed doing something ...

npirtube, 2025-02-15:
Hi y friend, i have two Autel MaxiCharger and i resolve more or less the first issue, but i found a new issue.
I can change max amper but the code dont sent a new value, always send 22000W 

i am trying to resolve this

shauncraill, 2025-02-24:
Hi, See attached respose from Autel

> Hi,
We have noticed that this issue is occurring due to the OCPP coding on your side. Kindly ask your OCPP team to review and send the correct code.
For your reference, please find the logs attached. Please ensure your team updates the code correctly for this function.

```
""", ,"event"
AE0007A1GRAC003591,"2025-02-23 15:48:19.111"        , ,"""[2025-02-23T07:47:58.000] OCPPMgr:EVENT:2123:recv:[2,""384eb62b-d44e-4469-981b-e87a734faac5"",""SetChargingProfile"",{""connectorId"":0,""csChargingProfiles"":{""chargingProfileId"":8,""stackLevel"":20,""chargingProfileKind"":""Relative"",""chargingProfilePurpose"":""ChargePointMaxProfile"",""chargingSchedule"":{""chargingRateUnit"":""A"",""chargingSchedulePeriod"":[{""startPeriod"":0,""limit"":15.0}]}}}]

```

> The charging profile purpose (chargingProfilePurpose) is set to ChargePointMaxProfile, but the charging profile kind (chargingProfileKind) is set to Relative. According to the system's rules, when the purpose is ChargePointMaxProfile, the kind must be Absolute, not Relative.

```
AE0007A1GRAC003591,"2025-02-23 15:48:19.497"        , ,"""[2025-02-23T07:47:58.000] OCPPMgr:EVENT:1936:3send:[4,""384eb62b-d44e-4469-981b-e87a734faac5"",""PropertyConstraintViolation"",""When chargingProfilePurpose =ChargingStationMaxProfile, chargingProfileKind SHALL NOT be Relative"",{}]
""", ,"event"
AE0007A1GRAC003591,"2025-02-23 15:48:19.111"        , ,"""[2025-02-23T07:47:58.000] OCPPMgr:EVENT:2123:recv:[2,""384eb62b-d44e-4469-981b-e87a734faac5"",""SetChargingProfile"",{""connectorId"":0,""csChargingProfiles"":{""chargingProfileId"":8,""stackLevel"":20,""chargingProfileKind"":""Relative"",""chargingProfilePurpose"":""ChargePointMaxProfile"",""chargingSchedule"":{""chargingRateUnit"":""A"",""chargingSchedulePeriod"":[{""startPeriod"":0,""limit"":15.0}]}}}]

```

shauncraill01, 2025-04-26:
I’m still struggling to fix it in code every now and then, unfortunately to no avail so far

nickdos, 2025-05-11:
I've hit this issue too. Interestingly, I was using EVCC and it was able to set the max amp value there, via OCPP. Suggesting it is possible with the current firmware. I'll revert to using EVCC until this is fixed.

I'm not able to charge at all with this integration. As soon as I plug my connector into the car (MG4) the availability entity changes to unavailable. I suspect it might be due to my car only supporting 6.6 kW charging and not the full 7 kW (32A).

french-sponge, 2025-05-11:
@nickdos, I have the same behavior as what you describe here, but that looks normal.
When I plug the car, then availability turns unavailable, and "Status Connector" goes to the "Preparing" state. Then I activate the charging by triggering "Charge Control" and the charge starts.
I have added a little automation that triggers the "Charge Control" 5 seconds after the "Status Connector" enters the "Preparing" state.
Which version is the "current firmware" you mention?

nickdos, 2025-05-11:
> [@nickdos](https://github.com/nickdos), I have the same behavior as what you describe here, but that looks normal. When I plug the car, then availability turns unavailable, and "Status Connector" [...]

I'm unable to initiate charging via the "Charge Control" entity. Car never charges. I think this is because the amp is set to 32A and I can't change it, as per this issue. My car will not accept 7kW - it's max AC rate is 6.6kW (MG4 64 essence).

My charger is stuck on firmware v1.19.00 but there's an update available (v1.22.00) but update fails every time I try. I've contacted the place I bought it from about this. Not sure if the update contains OCPP fixes or not.

french-sponge, 2025-05-12:
I also have 1.19.00
(when I upgrade the firmware, it goes until the end of the upload but nothing happens, it's still 1.19.0)

I did send an email to the support team regarding the OCPP, they replied that currently they don't take in charge the integration with Home Assistant.

nickdos, 2025-05-12:
> I also have 1.19.00
> (when I upgrade the firmware, it goes until the end of the upload but nothing happens, it's still 1.19.0)

Apparently this is a known issue and reading between the lines, it seemed 1.22.0 was intended for users who got the "bad" firmware (1.21.0) that bricked chargers. So I'm guessing 1.22 is a regression release identical to 1.19.

cjueden, 2025-05-19:
@PeterMikeSmooth can you elaborate on how you got this connection?
I'm have the same problems as others with the amps not responding, I tried to get an esp32 to communicate with the rs485 port but I read that this is not Modbus RTU and can't be used, I also tried but failed to get Modbus tcp working as your example above. 

My versions are PFA0102|V0.00.00|V1.36.00|V1.21.00|2.4.3.0

shauncraill, 2025-05-19:
Something I noticed, is that you have to connect via bluetooth to the charger and under load balancing choose external EMS, only then does the Modbus TCP start working (also not that this requires an ethernet connection not WiFi)

The RS485 port for what Autel calls meter connection would work the same, they disable the port unless you change to Load Balancing profile where the meter is required.

I'm still working on the modbus solution, hopefully almost there

cjueden, 2025-05-19:
> Just to help guide you to it (there are a couple of places in the app that
> look the same)
>
> Open app (where you would start charging)
>
> Tap settings cog left top
> Load balancing (should be there then)
>
> IMG_7524.jpeg (view on web)

I seem to be missing something. Thank you for that!

french-sponge, 2025-05-19:
hello, I managed to make it work with the OCPP integration.
As recommended elsewhere, I use the version 0.7.0 of OCPP integration from HACS (not the latest)

I can modify the current during the charging, using exactly this script in an automation :

```
action: ocpp.set_charge_rate
data:
  custom_profile:
    transactionId: "{{ states('sensor.ev_charger_transaction_id') | int }}"
    chargingProfileId: 1
    stackLevel: 2
    chargingProfilePurpose: TxProfile
    chargingProfileKind: Relative
    chargingSchedule:
      chargingRateUnit: A
      chargingSchedulePeriod:
        - startPeriod: 0
          limit: "{{ states('input_number.ev_current_limit') | int }}"
  conn_id: 1
```

This sets the current limitation on the transaction. The transaction must be started, i.e. the charge has started.
For stackLevel, I have arbitrarily chosen 2
"ev_current_limit" is the value in amps that I want to set (only tested with integers)
If this value is below 6A, the charger suspends the charge. It gets resumed when I set a higher value.

Here is the actual current-chart :

[image]

cjueden, 2025-05-19:
I'm extremely happy to say that I was able to implement your solution and it worked as expected. Including anything under 6 amps going to a suspended state then restarting when it goes above 6

[image]

dmatyas, 2025-05-20:
It's working with 0.8.3 too.

mavnz72, 2025-07-16:
@french-sponge. Thanks for your post.  Has enabled me to build a full solar charging automation which just works as it should.

shauncraill01, 2025-07-16:
I still need to do that 😅
Not completely sure where to start

cjueden, 2025-07-16:
Yea it does. One of the cars seemed to mind but not the others. It messes with the accuracy of my consumption monitors more than anything else I think so
I set a toggle for the automation on my EV dashboard and just turn it off when the clouds are intermittent. 

[image]

closed 2025-10-29 by github-actions[bot]