lbbrhzn/ocpp #1025: I am having trouble getting the sensors working  - fine for 10 months but has not worked for about 4 weeks now.  I am unable to track it to anything I have done: Controls and some information is coming through okay

MikeatLorne, 2023-12-26:
I am having trouble getting the sensors working  - fine for 10 months but has not worked for about 4 weeks now.  I am unable to track it to anything I have done: Controls and some information is coming through okay

I am getting this in debug files (sample): 

2023-12-26 15:39:02.005 DEBUG (MainThread) [custom_components.ocpp] 'A14223215317' trying measurand 'Energy.Reactive.Import.Register'
2023-12-26 15:39:02.005 INFO (MainThread) [ocpp] A14223215317: send [2,"4da1b115-d5b1-42a1-9b18-7a9a1c67aa6f","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Import.Register"}]
2023-12-26 15:39:02.105 INFO (MainThread) [ocpp] A14223215317: receive message [3,"4da1b115-d5b1-42a1-9b18-7a9a1c67aa6f",{"status":"RebootRequired"}]

any ideas?

_Originally posted by @MikeatLorne in https://github.com/lbbrhzn/ocpp/discussions/789#discussioncomment-7946643_

stefan73, 2024-01-21:
FYI: I see the same with an Entratek Charger, which is a rebranded IOCharger. My model is a IOCAH10-22T. From the log attached you have the same model. 22103101-22060401-22060401 is your firmware version. 22121701-21111301-21111301 is my firmware version. Another problem is that the working sensors are super slow. I guess it is an IOCharger firmware issue. Hope they will fix this.

MikeatLorne, 2024-01-25:
agreed - I hope they fix it.  It is a pain and stuffs my metering up big
time.

I am not sure how to get it moving??? I was hoping an upgrade would have
fixed it by now.

M

MikeatLorne, 2024-03-26:
It turns out that when fiddling, I managed to leave our the sensors in my charger config page - no problem with the OCPP integration - thanks for those that helped me

MikeatLorne, 2024-03-26:
It turns out that when fiddling, I managed to leave out the sensors in my charger config page - no problem with the OCPP integration - thanks to those that helped me

closed 2024-03-26 by MikeatLorne