lbbrhzn/ocpp #1035: Support charging in the full 1.4 - 11kW range

munterkalmsteiner, 2024-01-05:
**Is your feature request related to a problem? Please describe.**
There exist chargers that can switch between 1, (2?) and 3-phase charging. Switching between 1, 2, and 3-phase charging would allow optimal use of photovoltaic facilities as the whole range between 1.4kW (1 phase, 230V, 6A) and 11kW (3 phases, 230V, 16A) could be used. This allows charging when there is less than 4.1kW (3 phase, 230V, 6A) solar output. Furthermore, it would allow more optimal use of energy when using a load balancer as the charging would not be stopped when less than 4.1kW is available but continue to charge down to 1.4kW.

**Describe the solution you'd like**
OCPP 1.6 specifies smart charging profiles. It allows one to choose between 1, 2, and 3-phase charging if the charger supports it (configuration key `ConnectorSwitch3to1PhaseSupported`). Furthermore, the protocol allows to specify `chargingRateUnit` (which could be Watts or Ampere). I **assume** when choosing Watts, and the charger supports automatic phase switching, one can simply specify the wanted amount of power and the charger would then use the corresponding amount of phases. This would be a better solution than letting the management system decide on the number of phases.

**Describe alternatives you've considered**
With Tesla's unofficial API, one can choose the charging current between 1-16 A. This allows low-power charging (1 x 3A x 230V=690W). However, this is limited to Teslas and dependent on the future availability of the API.

**Additional context**
I've a [Garo Entity Pro](https://www.garo.se/en/professional/products/e-mobility/wallbox/entity-pro/wallbox-entity-pro-14-sigi-5) charger that allows 1 and 3-phase charging. I've used [evcc](https://evcc.io/en/) with the [OCPP integration](https://docs.evcc.io/en/docs/devices/chargers#ocpp-16j-compatible-charger-with-smart-charging-profile) and I can change with 1 and 3 phases. Hence, if this is implemented, I can test it.

drc38, 2024-01-05:
You can change the `ConnectorSwitch3to1PhaseSupported` using the `ocpp.configure` service. Give it a try and see how you get on.

munterkalmsteiner, 2024-01-05:
@drc38 according to the OCPP 1.6 spec, `ConnectorSwitch3to1PhaseSupported` is read-only. I've tried to use the `ocpp.get_configuration` service on this OCPP key to check what my charger responds, but I don't get any return value (also not for other config keys). Am I using the service wrong?

```
service: ocpp.get_configuration
data: {
  ocpp_key: ConnectorSwitch3to1PhaseSupported
}
```

drc38, 2024-01-06:
You need to look in the log for the returned value.

munterkalmsteiner, 2024-01-06:
Ah, of course. Phase switching is supported by the charger:
```
2024-01-06 09:03:09.629 INFO (MainThread) [ocpp] GaroCS-49152E2A66280: send [2,"1f95438f-49c9-4a27-baf9-76001e8d5d9d","GetConfiguration",{"key":["ConnectorSwitch3to1PhaseSupported"]}]
2024-01-06 09:03:09.658 INFO (MainThread) [ocpp] GaroCS-49152E2A66280: receive message [3,"1f95438f-49c9-4a27-baf9-76001e8d5d9d",{"configurationKey":[{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"True"}]}]
```

As well as smart charging:
```
2024-01-06 09:05:38.126 INFO (MainThread) [ocpp] GaroCS-49152E2A66280: send [2,"3412425c-433b-461b-b144-f3e840d766ba","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2024-01-06 09:05:38.171 INFO (MainThread) [ocpp] GaroCS-49152E2A66280: receive message [3,"3412425c-433b-461b-b144-f3e840d766ba",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger"}]}]
```

Is there a way to call the `set_charge_rate` service? It does not show up in the list of available services in my HA instance, even though it is listed in services.yaml. I could then try to call it with `limit_watts` as a parameter, specifying for example `2000`. This would allow me to check if automated phase switching works.

drc38, 2024-01-06:
Have you checked your unit supports setting `chargingRateUnit` in Watts? Why would Watts need to be specified to go between 1 and 3 phase charging? I would assume the charger selects 1 or 3 phase based on the car's inverter/cable compatibility?

munterkalmsteiner, 2024-01-06:
I don't find any config key that would answer whether the unit supports `chargingRateUnit` in Watts. However, my unit has a dynamic phase management system, see the [manual](https://technical-documentation.garo-next-gen.com/quick-guides/v1/en/EN_Manual_QG_Entity_Pro_InstallOperate_v1_3_4.pdf) page 32:

"GARO Entity PRO is available with a built-in phase management system, that when using DLM, automatically selects the phase(s) that can be used."

And then page 46:

"If 3-phase power is not available for minimum charging current of a charging station, but there is still current capacity on any single phase (1-phase), then charging can be continued on this/those individual phase(s) available on some single phase. Some models of charging stations have a built-in phase management system, that can select the 1–phase system fully dynamically to select the correct phase(s)."

I assume I could provide the available power in Watts, e.g. 2000, and then the charger would switch to 1-phase charging, since that is not enough for 3-phase charging (min. is 4100 W for 3-phase charging).

munterkalmsteiner, 2024-01-06:
> Have you checked your unit supports setting `chargingRateUnit` in Watts? 

The unit supports the smart charging profile (see output [here](https://github.com/lbbrhzn/ocpp/issues/1035#issuecomment-1879593397)) which specifies the parameter `chargingRateUnit`. Hence it would be at least possible to set it to Watts, as well as explicitly specifying how many phases should be used for charging.

drc38, 2024-01-06:
`ChargingScheduleAllowedChargingRateUnit` will specify what your unit supports

munterkalmsteiner, 2024-01-06:
Ok, only current is supported. Seems that setting the phases in the smart charging profile is the way to go:

```
2024-01-06 19:23:54.395 INFO (MainThread) [ocpp] GaroCS-49152E2A66280: receive message [3,"852010f9-a8e0-4d25-908b-fd56bf704be0",{"configurationKey":[{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current"}]}]
```

drc38, 2024-01-06:
Try manually adding the extra parameter to lines 631 and 657

munterkalmsteiner, 2024-01-06:
Yep, that worked!

[image]

Charging at 6A, 1 phase, 230V. I had also to add the `numberPhases` key to enums.py.

drc38, 2024-01-21:
> Ok, only current is supported. Seems that setting the phases in the smart charging profile is the way to go:
>
> ```
> 2024-01-06 19:23:54.395 INFO (MainThread) [ocpp] GaroCS-49152E2A66280: receive [...]

Try the service added in #1050 to define your custom charging profile for 1 or 3 phase charging. It can also be used for much more complex charging arrangements if needed.

mazzy89, 2024-01-25:
@drc38 can the `set_charge_rate` service be used to achieve the following use case?

I would need to start my charging session with pretty low amperage (around 9A) and then (automating it with HA) after some minutes ramp it up to the max allowed amperage. At the moment I have tried the `number.charger_max_current` slider but it does not work. I'm waiting for your PR to be merged and trying to achieve that. However I'm not sure if then once the charging session has started the amperage can be changed.

drc38, 2024-01-25:
@mazzy89 if the slider does not work that suggests your charger requires a more complex charging profile. You'll need to figure that out first. 

OCPP documentation is available [here](https://github.com/mobilityhouse/ocpp/tree/master/docs/v16)

drc38, 2024-01-25:
If you want to test the new service before it's merged just copy and replace the two altered files from `new_service` branch

mazzy89, 2024-01-25:
> If you want to test the new service before it's merged just copy and replace the two altered files from `new_service` branch

Yeah I know. At the end of the day is git. It just that I did not want to mess up with `custom_components` files directly but get a clean release.

> if the slider does not work that suggests your charger requires a more complex charging profile. 

I start the charging session with `FAST` profile. Does it mean I need to use `SMART` as profile?

mazzy89, 2024-01-25:
@drc38 once I set up a new Charging Profile, how can I start a session remotely using that profile? Currently I'm starting a session remotely toggling the `switch.charger_charge_control`

drc38, 2024-01-25:
> @drc38 once I set up a new Charging Profile, how can I start a session remotely using that profile? Currently I'm starting a session remotely toggling the `switch.charger_charge_control`

Yes, it should be the same as you do currently. Note it would be sensible to issue a clear profile request first to remove any existing profiles.

mazzy89, 2024-01-25:
But I do not want to remove existing profiles. I want to have the possibility to choice among charging profiles.

mazzy89, 2024-01-25:
@drc38 I have read the OCPP and I've noticed that unfortunately this integration misses some of the features. For instance according OCPP 1.6 is possible to remote start a transaction defining a charging profile. In our case we can start a transaction remotely but we cannot setup a charging profile along with the request :-(

drc38, 2024-01-25:
@mazzy89 in what circumstance would you use this feature? Applying the charge profile to all transactions and using HA to control the profile dynamically should be equivalent.

mazzy89, 2024-01-25:
It's hard to explain but I'll try to do my best.

Many chargers out there are not able to recognize the number of phases during the bootstrap of the charging session causing the charging session to be set with wrong number of phases and naturally causing issues at load balancing.

The way to go is to start the charging session with low amperage and then ramp up.

I have plenty of documented material about this issue with different chargers.

mazzy89, 2024-01-25:
Besides that what would be the current way to control the profile dynamically?

drc38, 2024-01-25:
If you need to specify the number of phases the new service should work along with a simple script that sets profile at the start of the transaction and say increments charge rate every x seconds until you reach your set point. If phases don't need to be specified the same can be done with the slider.

mazzy89, 2024-01-25:
@drc38 looking deeply at my logs I have noticed that the slider indeed sends out a set charging profile but in my case the ampere do not change. Note that the request is accepted by the charger. I do that before the transaction starts. Does it matter? Should I do after?

drc38, 2024-01-25:
You can change the slider before, during and after a transaction. The charger if it accepts the request should adjust its charging rate accordingly. If it doesn't you need to figure out why not with the manufacturer. The get diagnostics service may assist.

munterkalmsteiner, 2024-01-26:
> Try the service added in #1050 to define your custom charging profile for 1 or 3 phase charging. It can also be used for much more complex charging arrangements if needed.

@drc38 I can confirm that this works. I can switch with the custom charging profile between 1, 2, and 3 phase charging on the GARO Entity Pro. This fits my use case, i.e. calling the service in an automation when sufficient solar power is here for 1 phase charging and then switch, based on production, to 2 or 3 phase charging.

A word of warning for those who plan to do something similar. Switching phases causes the charging to stop and start again. Doing that too frequently (e.g. every couple of seconds when a cloud comes/disappears) seems like a bad idea. There are likely physical relays in the charging port of the car and switching them too often can wear them out prematurely. The automation should probably use some delays before phase switching. 
Here is some further information from [cFos](https://www.cfos-emobility.de/en/cfos-charging-manager/documentation/phase-switching.htm) and [SMA](https://my.sma-service.com/s/article/Automatic-Phase-Switching-by-SMA-EV-Charger?language=en_US). My GARO Entity does **not** pause 120 seconds, as recommended, when switching phases, so I need to manage this in the automation until GARO supports solar charging officially.

mazzy89, 2024-01-27:
@drc38 I have actually found out that the slider works for me. When I do manually from the UI works. The problem is that when I do via the automation. I use it 

```yaml
    - service: input_number.set_value
      data:
        value: 9
        entity_id: number.charger_maximum_current
```

but seems that the entity is not find by the `input_number.set_value`. the `input_number` service expects only `input_number` entity and not `number` entity created programmatically via the integration.

closed 2024-04-05 by github-actions[bot]

Vengman, 2024-09-16:
[image]

Hi, I can't figure out how to switch from 3 to 1 phase in home assistant. has a Keba p30 charger that is connected to ocpp and keba integration. thought it should support it. see image from web interface. Can someone try to help me get it up and running? preferably very elaborate