lbbrhzn/ocpp #1036: Frequent reboots of ocpp

pzucchel, 2024-01-07:
i experience frequent reboots with my wattpilot charger from fronius. 

## Version of the custom_component

 current version ocpp, version 40.5 fronius firmware

## Configuration

```yaml
This error originated from a custom integration.

Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:519
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 09:52:58 (3 occurrences)
Last logged: 10:02:56

Configuration of the charger failed: NotImplementedError: , {} 

This error originated from a custom integration.

Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:877
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 09:52:57 (3 occurrences)
Last logged: 10:02:53

NotSupported while setting MeterValuesSampledData to
This error originated from a custom integration.

Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:866
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 09:52:57 (3 occurrences)
Last logged: 10:02:53

MeterValuesSampledData is a read only setting

```

## Describe the bug

I get hundreds of notifications per day, and some interruptions in the logging. Said this, the charger still works correctly. I feel I should disable some measurands, but I couldn't find any instructions on how to do it. 

## Debug log

2024-01-07 09:52:49.468 INFO (MainThread) [ocpp] : receive message [2,"531531716","BootNotification",{"chargePointVendor":"fronius","chargePointModel":"wattpilot_V2","chargePointSerialNumber":"91022316","firmwareVersion":"40.7"}]
2024-01-07 09:52:49.476 INFO (MainThread) [ocpp] : send [3,"531531716",{"currentTime":"2024-01-07T08:52:49Z","interval":3600,"status":"Accepted"}]
2024-01-07 09:52:49.640 INFO (MainThread) [ocpp] : receive message [2,"531531717","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2024-01-07T08:52:48.477Z"}]
2024-01-07 09:52:49.645 INFO (MainThread) [ocpp] : send [3,"531531717",{}]
2024-01-07 09:52:49.791 INFO (MainThread) [ocpp] : receive message [2,"531531718","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"SuspendedEVSE","timestamp":"2024-01-07T08:52:48.477Z"}]
2024-01-07 09:52:49.793 INFO (MainThread) [ocpp] : send [3,"531531718",{}]
2024-01-07 09:52:49.932 INFO (MainThread) [ocpp] : receive message [2,"531531719","MeterValues",{"connectorId":1,"transactionId":1704534842,"meterValue":[{"timestamp":"2024-01-07T08:52:48.477Z","sampledValue":[{"value":"0.00","measurand":"Current.Import","unit":"A","phase":"L1"},{"value":"0.00","measurand":"Current.Import","unit":"A","phase":"L2"},{"value":"0.00","measurand":"Current.Import","unit":"A","phase":"L3"},{"value":"275450.00","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0.00","measurand":"Power.Active.Import","unit":"W","phase":"L1"},{"value":"0.00","measurand":"Power.Active.Import","unit":"W","phase":"L2"},{"value":"0.00","measurand":"Power.Active.Import","unit":"W","phase":"L3"},{"value":"0.00","measurand":"Power.Active.Import","unit":"W"}]}]}]
2024-01-07 09:52:49.940 INFO (MainThread) [ocpp] : send [3,"531531719",{}]
2024-01-07 09:52:51.314 INFO (MainThread) [ocpp] : send [2,"7bf0685d-3305-4fb1-8028-42ffac764a02","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2024-01-07 09:52:51.582 INFO (MainThread) [ocpp] : receive message [3,"7bf0685d-3305-4fb1-8028-42ffac764a02",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,SmartCharging,RemoteTrigger"}],"unknownKey":[]}]
2024-01-07 09:52:51.587 INFO (MainThread) [ocpp] : send [2,"9bccf4fe-23ca-4b6c-8dff-47069e9db8cd","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2024-01-07 09:52:51.677 INFO (MainThread) [ocpp] : receive message [3,"9bccf4fe-23ca-4b6c-8dff-47069e9db8cd",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,SmartCharging,RemoteTrigger"}],"unknownKey":[]}]
2024-01-07 09:52:51.680 INFO (MainThread) [ocpp] : send [2,"2a7a4d6b-b539-4825-ae08-537e54b31967","GetConfiguration",{"key":["NumberOfConnectors"]}]
2024-01-07 09:52:51.777 INFO (MainThread) [ocpp] : receive message [3,"2a7a4d6b-b539-4825-ae08-537e54b31967",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}],"unknownKey":[]}]
2024-01-07 09:52:51.780 INFO (MainThread) [ocpp] : send [2,"3a391966-767b-452b-99f2-bc6a34a3f307","GetConfiguration",{"key":["NumberOfConnectors"]}]
2024-01-07 09:52:51.903 INFO (MainThread) [ocpp] : receive message [3,"3a391966-767b-452b-99f2-bc6a34a3f307",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}],"unknownKey":[]}]
2024-01-07 09:52:51.905 INFO (MainThread) [ocpp] : send [2,"66eff552-252c-4d1f-b1df-f3b9a9a3a786","GetConfiguration",{"key":["HeartbeatInterval"]}]
2024-01-07 09:52:52.038 INFO (MainThread) [ocpp] : receive message [3,"66eff552-252c-4d1f-b1df-f3b9a9a3a786",{"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"3600"}],"unknownKey":[]}]
2024-01-07 09:52:52.043 INFO (MainThread) [ocpp] : send [2,"5637b369-8a1b-4b53-856d-5cdd2935f5eb","GetConfiguration",{"key":["HeartbeatInterval"]}]
2024-01-07 09:52:52.137 INFO (MainThread) [ocpp] : receive message [3,"5637b369-8a1b-4b53-856d-5cdd2935f5eb",{"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"3600"}],"unknownKey":[]}]
2024-01-07 09:52:52.139 INFO (MainThread) [ocpp] : send [2,"cf27a1ab-c656-4b1f-b767-e4ffed0e5daf","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Export"}]
2024-01-07 09:52:52.319 INFO (MainThread) [ocpp] : receive message [3,"cf27a1ab-c656-4b1f-b767-e4ffed0e5daf",{"status":"NotSupported"}]
2024-01-07 09:52:52.323 INFO (MainThread) [ocpp] : send [2,"b830dd4f-f476-450d-9c51-cc749941e91d","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Export"}]
2024-01-07 09:52:52.437 INFO (MainThread) [ocpp] : receive message [3,"b830dd4f-f476-450d-9c51-cc749941e91d",{"status":"NotSupported"}]
[... 98 lines cut ...]
2024-01-07 09:52:57.973 INFO (MainThread) [ocpp] : send [2,"2f7e5e6d-b07c-4051-b9c9-e4d02409f972","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2024-01-07 09:52:58.076 INFO (MainThread) [ocpp] : receive message [3,"2f7e5e6d-b07c-4051-b9c9-e4d02409f972",{"configurationKey":[{"key":"MeterValueSampleInterval","readonly":false,"value":"60"}],"unknownKey":[]}]
2024-01-07 09:52:58.079 INFO (MainThread) [ocpp] : send [2,"188667c6-fcd1-4d19-91b2-b9be2e480e31","GetConfiguration",{"key":["ClockAlignedDataInterval"]}]
2024-01-07 09:52:58.197 INFO (MainThread) [ocpp] : receive message [3,"188667c6-fcd1-4d19-91b2-b9be2e480e31",{"configurationKey":[{"key":"ClockAlignedDataInterval","readonly":false,"value":"900"}],"unknownKey":[]}]
2024-01-07 09:52:58.200 INFO (MainThread) [ocpp] : send [2,"800f6de0-5187-4c16-a257-4773a2ac0f00","GetConfiguration",{"key":["ClockAlignedDataInterval"]}]
2024-01-07 09:52:58.321 INFO (MainThread) [ocpp] : receive message [3,"800f6de0-5187-4c16-a257-4773a2ac0f00",{"configurationKey":[{"key":"ClockAlignedDataInterval","readonly":false,"value":"900"}],"unknownKey":[]}]
2024-01-07 09:52:58.324 INFO (MainThread) [ocpp] : send [2,"ec830e8e-ad78-40de-be3c-ec7bbbb2950c","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2024-01-07 09:52:58.446 INFO (MainThread) [ocpp] : receive message [3,"ec830e8e-ad78-40de-be3c-ec7bbbb2950c",{"configurationKey":[{"key":"WebSocketPingInterval","readonly":false,"value":"60"}],"unknownKey":[]}]
2024-01-07 09:52:58.450 INFO (MainThread) [ocpp] : send [2,"d7053917-ea75-41a4-84a0-4486ccc73093","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2024-01-07 09:52:58.560 INFO (MainThread) [ocpp] : receive message [3,"d7053917-ea75-41a4-84a0-4486ccc73093",{"configurationKey":[{"key":"WebSocketPingInterval","readonly":false,"value":"60"}],"unknownKey":[]}]
2024-01-07 09:52:58.561 INFO (MainThread) [ocpp] : send [2,"c6316c0b-5e10-4ab5-ab16-727305a71726","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2024-01-07 09:52:58.658 INFO (MainThread) [ocpp] : receive message [4,"c6316c0b-5e10-4ab5-ab16-727305a71726","NotImplemented","",{}]
2024-01-07 09:52:58.659 ERROR (MainThread) [custom_components.ocpp] Configuration of the charger failed: NotImplementedError: , {}
[... cut ...]
2024-01-07 09:52:58.720 INFO (MainThread) [ocpp] : receive message [4,"e8b68c36-f26c-4472-9da3-0caf1e01c934","NotImplemented","",{}]
2024-01-07 09:52:58.720 ERROR (MainThread) [custom_components.ocpp] Configuration of the charger failed: NotImplementedError: , {}
2024-01-07 09:53:05.747 ERROR (MainThread) [custom_components.wattpilot] 5d5b19ce1c181ad84d95ca38857bdec5 - async_setup_entry: Timeout - charger not connected: False (60 sec)
2024-01-07 09:53:06.770 ERROR (MainThread)

```text

Add your logs here.

```

drc38, 2024-01-07:
It looks like the Fronius unit has some way to go to be OCPP compliant. I suggest raising a support ticket with them.

drc38, 2024-01-07:
You can also look at how others have customised the integration to workaround charger issues in the interim eg Grizzl-E chargers

closed 2024-03-16 by github-actions[bot]