lbbrhzn/ocpp #1062: Support for pauze/resume of the charger via OCPP commands

scauwe, 2024-01-26:
**Is your feature request related to a problem? Please describe.**

We are confronted with peak consumption pricing. In order to lower the peak, I want to be able to pauze charging when needed. OCPP (and the used library) support this (RemoteStopTransaction & RemoteStartTransaction) and the functions seem to be implemented in the HA layer (async def start_transaction(self) & async def stop_transaction(self)), but these are not exposed as controls (switches).

**Describe the solution you'd like**
Expose a switch. 
 * When set to off, it will pauze the charger (RemoteStopTransaction), with the running transactionID as parameter
 * When set to on, it will resume the transaction (RemoteStartTransaction), with the tagid of the last transaction as parameter

**Describe alternatives you've considered**
 * The wallbox app has a pauze/resume, but that is disabled as soon as OCPP is connected.
 * The wallbox HA integration also has pauze/resume, but that is disabled as soon as OCPP is connected.

mazzy89, 2024-01-27:
@scauwe I think this is already present. I start/stop remotely the transaction. See my comment with my automation https://github.com/lbbrhzn/ocpp/discussions/456#discussioncomment-8266382

scauwe, 2024-01-28:
Hi,
Indeed: that is not what I am looking for. I want to be able to continue the transaction without the need to unplug & replug the cable.
From what I understood, the RemoteStopTransaction and RemoteStartTransaction are supposed to be able to work without the need the unplug and replug the cable.
I know that my charger supports it: when no ocpp is connected, I get a pauze/resume functionality in the app.

What I just noticed in the logs: the HA ocpp call RemoteStartTransaction is not using the tag of the previous session, but it uses the serial number of the charger. As a response, the ocpp sends an "Invalid" back. I'll try adding that serial number as a valid tagid and see if that makes this work (I configured my ocpp to authenticate tagids since it's accesible from the street).

scauwe, 2024-01-28:
So, I added the serial number of the charger to the tagid configuration as a valid tag id, and now indeed the StartRemoteTransaction seems to work. without the need to unplug and replug my cable It's not exactly the same as the pauze & resume though: it actually stops and starts a new transaction (the pauze/resume keeps the same transaction going). The result is that I now have like 15 transactions in a day (but I assume I can live with that).
What I was yet unable to test if I can unlock the charger with my bage: since the transaction was not started with my badge, I assume I will not be able to stop it.
But still: it does sound like a bug to start the transaction using the serial number of the charger as the tagid. Either no tag id, either the last known tagid I would say (if you know that the charger status never became "Available", I would opt for the second: last known tag id.

Stefaan

mazzy89, 2024-01-28:
> So, I added the serial number of the charger to the tagid configuration as a valid tag id, and now indeed the StartRemoteTransaction seems to work. without the need to unplug and replug my cable [...]

Exactly. However, I'm wondering how we put the charger in the `SuspendedEVSE` state. That is the only state that signals that the charger is in pause. Certainly, it is not done with Start/Stop Transaction.
I would need to look at the OCPP RFC and see how to send a request to pause the charger putting the charger in `SuspendedEVSE`.

scauwe, 2024-01-28:
I think I found it. I will try this tomorrow:
From https://chargehq.net/kb/ocpp-charger-testing-procedure
"Zero power limit
Set power limit to 0.
Wait one minute.
Status should change to SuspendedEVSE."

mazzy89, 2024-01-28:
Very good idea. I think it is exactly that. You can set the limit to 0 using the new service set charge rate merged few days ago

scauwe, 2024-01-30:
I tried it and it works as it should: the wallbox connector status goes into "SuspendedEVSE" when current is set to 0 amps and back to "Charging" when it is set to 6 amps.
Thanks for this!

closed 2024-01-30 by scauwe