lbbrhzn/ocpp #1068: Rounding of power and energy measurand

scauwe, 2024-01-30:
**Describe the bug**
Various received measurand seem to be rounded up/down. Example: my wallbox sends every minute the MeterValues. One of them is 
,{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Interval","unit": "Wh","value": "67.5"} 
When looking in the sensors, this is rounded 0.1kWh. I seem unable to find in the code where this rounding occures. I assumed it was a displaying issue, but changing the display precision did not change anything.
I see in the code that the Wh is converted to kWh, where a float is devided by 1000 (resulting in a new float), but nowhere is it rounded to a precision of 0.1

**To Reproduce**
View any numeric sensor (kW or kWh). They are all rounded to a precision of 0.1, removing precision details given by the meter.

**Expected behavior**
No rounding or rounding like most other power/energy meter: to 0.01

drc38, 2024-01-31:
Ultimately the precision is set by what your charger reports and its base unit. You can amend the display units and precision to Wh and 0.1 in HA by changing the settings on each sensor.

scauwe, 2024-01-31:
I doubt that:

- the HA extention converts all values to 'k' by deviding it to 1000 (Wh => kWh and  W => kW). Since this is a float, precision is not yet lost.
- I just found a scale defined in sensors.py, that is set to 1. The native value is then rounded in the following lines:
```
if isinstance(value, float):
            value = round(value, self.entity_description.scale)
```
I assume this rounding is what ends up in HA (in the display and the states dB table). The precision is lost after that. This is hardcoded so cannot be configured by any HA display setting.

drc38, 2024-01-31:
Good spot, that looks to be a legacy from when HA did not support setting the display precision for sensors. Comment it out and see how you get on.

scauwe, 2024-02-01:
I can confirm. I changed this to "2", and now I have 2 digits in my sensors.

PR #1071 (2024-03-23): Remove scaling from sensors
fixes #1068
closed 2024-03-23 by lbbrhzn