lbbrhzn/ocpp #1069: Negative value at the start for charger Energy Session

scauwe, 2024-01-30:
**Describe the bug**
When a session is started, the value for "charger Energy Session" always starts with a slightly negative value. I **assume** the reason is that the energy at the end of the session is not correctly captured (see context below).

**To Reproduce**
With my wallbox copper sb, start a session.

**Expected behavior**
Non-negative start values and better capture of energy session value.

**Screenshots**
[image]

**Additional context**
Log when stopping a transaction, shows (in the StopTransaction) a metervalue of 2259988.
The first MeterValues after that (without a transaction id), shows a slightly higher meter value of 2259992.1. I asume that the time between the stop transaction and the actual termination of the charging is responsible for this difference, resulting in a negative value when a new session starts: difference between the value "meterStart" and the value between "meterStop" of the previous transaction.

```
2024-01-30 15:29:16.767 INFO (MainThread) [ocpp] YYYY: receive message [2,"585855959","StopTransaction",{"meterStop": 2259988,"timestamp": "2024-01-30T14:29:16Z","idTag": "XXXXXX","transactionData": [],"reason": "Local","transactionId": 1706601047}]
2024-01-30 15:29:16.771 INFO (MainThread) [ocpp] YYYY: send [3,"585855959",{"idTagInfo":{"status":"Accepted"}}]
2024-01-30 15:29:38.671 INFO (MainThread) [ocpp] YYYY: receive message [2,"659854644","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Available","timestamp": "2024-01-30T14:29:38Z"}]
2024-01-30 15:29:38.674 INFO (MainThread) [ocpp] YYYY: send [3,"659854644",{}]
2024-01-30 15:30:00.369 INFO (MainThread) [ocpp] YYYY: receive message [2,"543368638","MeterValues",{"meterValue": [{"timestamp": "2024-01-30T14:15:00Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Clock","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "2259992.1"} ]} ],"connectorId": 0}]

```

drc38, 2024-01-31:
Seems your charger is unusual in reporting to 0.1Wh. Try removing int from lines 1424 and 1451, although this doesn't explain a 0.6kWh difference. It may be to do with out of sequence messages from the charger ie to get a negative value HA must receive a higher first then lower meter value.

scauwe, 2024-01-31:
Seems that I have a different version (downloaded via HACS, manifest.json reports v0.5.1) than what is in git.
I changed corresponding lines 1371 and 1398, replacing int with float and restarted HA. 
I'll report an update.

Stefaan

scauwe, 2024-01-31:
Strange: the manifest.json in github, main branch, reports v 0.4.42.

drc38, 2024-01-31:
> Strange: the manifest.json in github, main branch, reports v 0.4.42.

The value in github `manifest.json` is arbitrary, it gets updated via workflow when a release is issued.

scauwe, 2024-02-04:
I validated the problem. The meter details submitted on the StopRequest (meterStop) are not the details when actually stopping. Those are details at the time of submitting the stop request. During the time of the stop request and actual stopping, the charger continues charging for a second or so.
The next MeterValues received from the charge point contain the actual meter readings (Energy.Active.Import.Register). Given that I charged on 3 phases, I quickly get 0.6kWh difference between these two values.
The subsequent startTransaction then contains the actual meter reading at that time. For the HA OCCP, these are 0.6kWh higher than it should, resulting in a negative value.
As a test, I changed to code to aded the first meter values after a session was finished to the "charger Energy Session" sensor. This solved the negative values.
The time between stopping a session and getting the next meter values can however be up to 15 minutes (when no session is present, the wallbox apparently submits data every 15 minutes). So, it can take a while to have this reflected in the HA data.
Idealy, the charger should submit a MeterValues immidiatly after stopping a session. I did find the message "TriggerMessage", that can request the charge point to submit a message to the OCCP. One of the messages that can be triggered is MeterValues. Would be very interested to add that (if the used library supports that).

Roos-AID, 2024-03-17:
Polling Metervalues might also help with this issue? https://github.com/lbbrhzn/ocpp/issues/1031

closed 2024-05-25 by github-actions[bot]